meta-flutter/meta-flutter #266: Compilation error with flutter-engine 3.7.6

witekio-kwe, 2023-03-15:
```
| [46/8852] CXX obj/flutter/flutter_vma/flutter_vma.flutter_vma.o
| FAILED: obj/flutter/flutter_vma/flutter_vma.flutter_vma.o
| /build-path/build/tmp/work/cortexa53-crypto-simplecoredistro-linux/flutter-engine/git-r0/src/buildtools/linux-x64/clang/bin/clang++ -MD -MF obj/flutter/flutter_vma/flutter_vma.flutter_vma.o.d --target=aarch64-unknown-linux-gnu --sysroot /build-path/build/tmp/work/cortexa53-crypto-simplecoredistro-linux/flutter-engine/git-r0/recipe-sysroot -DUSE_OPENSSL=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_LIBCPP_DISABLE_AVAILABILITY=1 -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DVK_USE_PLATFORM_XCB_KHR -DVK_USE_PLATFORM_WAYLAND_KHR -DVMA_DYNAMIC_VULKAN_FUNCTIONS=0 -DVMA_STATIC_VULKAN_FUNCTIONS=0 -I../.. -Igen -I../../third_party/libcxx/include -I../../third_party/libcxxabi/include -I../../build/secondary/third_party/libcxx/config -I../../third_party/vulkan-deps/vulkan-headers/src/include -I../../third_party/vulkan_memory_allocator/include -fno-strict-aliasing -fstack-protector --param=ssp-buffer-size=4 -fPIC -pipe -pthread --target=aarch64-linux-gnu -DBORINGSSL_CLANG_SUPPORTS_DOT_ARCH -fcolor-diagnostics -Wall -Wextra -Wendif-labels -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-implicit-int-float-conversion -Wno-c99-designator -Wno-deprecated-copy -Wno-psabi -Wno-unqualified-std-cast-call -Wno-non-c-typedef-for-linkage -Wno-range-loop-construct -fvisibility=hidden --sysroot=/build-path/build/tmp/work/cortexa53-crypto-simplecoredistro-linux/flutter-engine/git-r0/recipe-sysroot -Wstring-conversion -Wnewline-eof -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -fvisibility-inlines-hidden -std=c++17 -fno-rtti -nostdinc++ -nostdinc++ -fvisibility=hidden -fno-exceptions -Wno-c++98-compat-extra-semi -Wno-deprecated-copy -Wno-implicit-fallthrough -Wno-nullability-completeness -Wno-suggest-destructor-override -Wno-suggest-override -Wno-unused-private-field -Wno-unused-variable -c ../../flutter/flutter_vma/flutter_vma.cc -o obj/flutter/flutter_vma/flutter_vma.flutter_vma.o
| In file included from ../../flutter/flutter_vma/flutter_vma.cc:19:
| In file included from ../../flutter/flutter_vma/flutter_vma.h:7:
| In file included from ../../third_party/vulkan_memory_allocator/include/vk_mem_alloc.h:130:
| ../../third_party/vulkan-deps/vulkan-headers/src/include/vulkan/vulkan.h:53:10: fatal error: 'xcb/xcb.h' file not found
| #include <xcb/xcb.h>
|          ^~~~~~~~~~~
| 1 error generated.
| [47/8852] CXX obj/flutter/third_party/tonic/tonic.dart_class_provider.o
| [48/8852] CXX obj/flutter/third_party/tonic/tonic.dart_wrappable.o
```
is what I get with the latest proposed 3.7.6 release.
We only support wayland, not x11 in our builds, therefore even adding libxcb doesn’t change anything

jwinarske, 2023-03-16:
@witekio-kwe Yes known issue.  Workaround is to include x11 in DISTRO_FEATURES (yes I know).  I'll get to a formal patch it in the next day or so.

jwinarske, 2023-03-17:
https://github.com/meta-flutter/meta-flutter/commit/dc90eaef0e72d7e207866e1e73906a71b7ffb027

closed 2023-03-17 by jwinarske