meta-flutter/meta-flutter #268: eSDK install

jwinarske, 2023-03-17:
While installing eSDK, I faced on the error related to this issue.

```
ERROR: ExpansionError during parsing /workdir/eSDK/layers/meta-flutter/recipes-graphics/flutter-apps/flutter-test-plugins_git.bb
Traceback (most recent call last):
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 525, in OpenerDirector.open(fullurl='https://storage.googleapis.com/flutter_infra_release/releases/releases_linux.json', data=None, timeout=<object object at 0x7f7cabe04f00>):
             sys.audit('urllib.Request', req.full_url, req.data, req.headers, req.get_method())
    >        response = self._open(req, data)

  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 542, in OpenerDirector._open(req=<urllib.request.Request object at 0x7f7ca9313f40>, data=None):
             protocol = req.type
    >        result = self._call_chain(self.handle_open, protocol, protocol +
                                       '_open', req)
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 502, in OpenerDirector._call_chain(chain={'unknown': [<urllib.request.UnknownHandler object at 0x7f7ca9313fd0>], 'http': [<urllib.request.HTTPHandler object at 0x7f7ca930ddc0>], 'ftp': [<urllib.request.FTPHandler object at 0x7f7ca930df70>], 'file': [<urllib.request.FileHandler object at 0x7f7ca930d4c0>], 'data': [<urllib.request.DataHandler object at 0x7f7ca930d790>], 'https': [<urllib.request.HTTPSHandler object at 0x7f7ca930d130>]}, kind='https', meth_name='https_open', *args=(<urllib.request.Request object at 0x7f7ca9313f40>,)):
                 func = getattr(handler, meth_name)
    >            result = func(*args)
                 if result is not None:
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 1397, in HTTPSHandler.https_open(req=<urllib.request.Request object at 0x7f7ca9313f40>):
             def https_open(self, req):
    >            return self.do_open(http.client.HTTPSConnection, req,
                     context=self._context, check_hostname=self._check_hostname)
  File "/workdir/eSDK/buildtools/sysroots/x86_64-oesdk-linux/usr/lib/python3.8/urllib/request.py", line 1357, in HTTPSHandler.do_open(http_class=<class 'http.client.HTTPSConnection'>, req=<urllib.request.Request object at 0x7f7ca9313f40>, **http_conn_args={'context': None, 'check_hostname': None}):
                 except OSError as err: # timeout error
    >                raise URLError(err)
                 r = h.getresponse()
bb.data_smart.ExpansionError: Failure expanding variable do_install, expression was
    install -d /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/data/flutter_assets
    cp -r /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/git/plugins/build/flutter_assets/* /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/data/flutter_assets/

    if ${@bb.utils.contains('FLUTTER_RUNTIME', 'release', 'true', 'false', d)} || \
        ${@bb.utils.contains('FLUTTER_RUNTIME', 'profile', 'true', 'false', d)}; then

        bbnote "Flutter Application: Installing debug"

        install -d /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/lib
        cp /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/git/plugins/libapp.so /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/share/flutter/linux_plugins/lib/
    fi

    if [[ "bundle" =~ .*"linux".* ]]; then

        bbnote "Flutter Application: Installing GTK bundle"

        if [ -n "${FLUTTER_REMOVE_LINUX_BUILD_ARTIFACTS}" ]; then
            rm /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/linux_plugins | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/lib/*.so | true
        else
            # expecting default "release" build
            mv /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/linux_plugins /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/bin/ | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/bin/linux_plugins | true
            rm -rf /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/usr/lib/*.so | true
        fi
    fi
    install -D -m0644 /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/lsb-release \
        /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/etc/lsb-release
    sed -i "s|@FLUTTER_SDK_TAG@|${@get_flutter_sdk_version(d)}|g" /workdir/eSDK/tmp/work/core2-64-oe-linux/flutter-test-plugins/git-r0/image/etc/lsb-release
 which triggered exception URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1131)>
```

Since the network access happens while parsing recipes but somehow HTTPS access cannot be executed in eSDK installation, the above error is caused.

I think https://github.com/meta-flutter/meta-flutter/pull/229 may solve the above error also.

At least, we need to add something like the following before urllib.request.urlopen in order to support Yocto eSDK if using urllib.request.urlopen.

    import ssl
    ssl._create_default_https_context = ssl._create_unverified_context

cc @WAKazuyoshiAkiyama

WAKazuyoshiAkiyama, 2023-04-05:
@jwinarske 
I created the backport PR for the dunfell branch.
https://github.com/meta-flutter/meta-flutter/pull/276

jwinarske, 2023-04-05:
@WAKazuyoshiAkiyama Note all dunfell based BSPs that support the override specifier introduced in kirkstone (it was back ported to dunfell), will work as-is with the kirkstone branch.  If you use tip of tree meta-rust, then meta-flutter kirkstone just works.

WAKazuyoshiAkiyama, 2023-04-06:
@jwinarske 
However, in the old dunfell, the new override syntax is not supported, so, this PR is useful for the users using such revision of dunfell.

jwinarske, 2023-11-16:
@WAKazuyoshiAkiyama can this be closed?

WAKazuyoshiAkiyama, 2023-11-22:
Yes, this can be closed.

closed 2023-11-22 by jwinarske