meta-flutter/meta-flutter #274: Inquiry about support for Sony's example video player app

tinhembedded, 2023-03-28:
Hello developers,

I recently want to create an example video player app to run on AGL, but I have a question about its support.

Could you please clarify whether this example video player app is developed by Sony( https://github.com/sony/flutter-elinux-plugins/tree/main/packages/video_player/example ) is supported to build and run by flutter-auto?

Thank you again for your contribution to the developer community and I look forward to your response.

tinhembedded, 2023-03-28:
If it worked, please describe how to to build and run that app

jwinarske, 2023-03-30:
@tinhembedded At one point the Flutter 1P video player example worked with no change with flutter-auto/ivi-homescreen when this was added to build config `-DBUILD_PLUGIN_GSTREAMER_EGL=ON`:
https://github.com/flutter/packages/tree/main/packages/video_player/video_player/example

I haven't ran it in a while, and the example may have moved to FFI.  If you can provide a test source (https, or other) that represents your desired format, I can take a look.

It may be as simple as moving to a specific commit in flutter/packages and enabling the build flag.  The build flag is off by default, as it requires a commercial license.  If it's just for personal use, then no-worries.

closed 2023-07-03 by jwinarske