meta-flutter/meta-flutter #277: desktop-auto flickering and QEMU image missing

bernardcraddock, 2023-04-10:
Hi @jwinarske 
long time not talk trust all is well with you.

I recently had to rebuild Flutter Workspace Automation clean on my Ubuntu/desktop 22.04 jammy using 
$ git clone –b kirkstone https://github.com/meta-flutter/meta-flutter.git
$ cd meta-flutter/tools
$ ./setup_flutter_workspace.py --clean

workspace install seemed to run just fine but encountered 2 issues that hopefully you can advise me on

1. when running ped using desktop-auto ped launches just fine but then the screen starts flickering like crazy, I thought it might be something i'd done in PED so tried flutter_homescreen and same thing happening there, And If i configure and run using default linux desktop both working just fine.

$ source ./setup_env.sh  
$ cd meta-flutter/tools/app/ped
$ flutter run -d desktop-auto

[ 2023-04-10 16:40:07 ped-vm /home/parallels/Documents/GitHub/meta-flutter/tools/app/ped gitb:(agl) flutterc: ]
parallels > flutter run -d desktop-auto
Launching lib/main.dart on Toyota flutter-auto in debug mode...
[INFO:main.cc(85)] agl @ 2109ff0
[INFO:main.cc(102)] Json Configuration: /home/parallels/Documents/GitHub/meta-flutter/tools/.flutter-auto/default_config.json
[INFO:main.cc(132)] Bundle Path: /tmp/pumped_end_device
[INFO:configuration.cc(230)] View Count: 1
[INFO:configuration.cc(284)] **********
[INFO:configuration.cc(285)] * Global *
[INFO:configuration.cc(286)] **********
[INFO:configuration.cc(287)] Application Id: .......... flutter-auto
[INFO:configuration.cc(289)] JSON Configuration: ...... /home/parallels/Documents/GitHub/meta-flutter/tools/.flutter-auto/default_config.json
[INFO:configuration.cc(291)] Cursor Theme: ............ DMZ-White
[INFO:configuration.cc(292)] Disable Cursor: .......... false
[INFO:configuration.cc(294)] Debug Backend: ........... false
[INFO:configuration.cc(296)] ********
[INFO:configuration.cc(297)] * View *
[INFO:configuration.cc(298)] ********
[INFO:configuration.cc(305)] Bundle Path: .............. /tmp/pumped_end_device
[INFO:configuration.cc(306)] Window Type: .............. NORMAL
[INFO:configuration.cc(307)] Output Index: ............. 0
[INFO:configuration.cc(309)] Size: ..................... 1920 x 1080
[INFO:configuration.cc(314)] Fullscreen: ............... false
[INFO:configuration.cc(316)] Accessibility Features: ... 0
[INFO:display.cc(37)] + Display()
[INFO:display.cc(139)] Wayland: wl_compositor version 5
[INFO:display.cc(146)] 	Buffer Scale Enabled
[INFO:display.cc(139)] Wayland: wl_drm version 2
[INFO:display.cc(139)] Wayland: wl_shm version 1
[INFO:display.cc(139)] Wayland: wl_output version 2
[INFO:display.cc(182)] Wayland: Output [0]
[INFO:display.cc(139)] Wayland: zxdg_output_manager_v1 version 3
[INFO:display.cc(139)] Wayland: wl_data_device_manager version 3
[INFO:display.cc(139)] Wayland: zwp_primary_selection_device_manager_v1 version 1
[INFO:display.cc(139)] Wayland: gtk_primary_selection_device_manager version 1
[INFO:display.cc(139)] Wayland: wl_subcompositor version 1
[INFO:display.cc(139)] Wayland: xdg_wm_base version 4
[INFO:display.cc(139)] Wayland: zxdg_shell_v6 version 1
[INFO:display.cc(139)] Wayland: gtk_shell1 version 5
[INFO:display.cc(139)] Wayland: wp_viewporter version 1
[INFO:display.cc(139)] Wayland: zwp_pointer_gestures_v1 version 3
[INFO:display.cc(139)] Wayland: zwp_tablet_manager_v2 version 1
[INFO:display.cc(139)] Wayland: wl_seat version 5
[INFO:display.cc(139)] Wayland: zwp_relative_pointer_manager_v1 version 1
[INFO:display.cc(139)] Wayland: zwp_pointer_constraints_v1 version 1
[INFO:display.cc(139)] Wayland: zxdg_exporter_v1 version 1
[INFO:display.cc(139)] Wayland: zxdg_importer_v1 version 1
[INFO:display.cc(139)] Wayland: zwp_linux_dmabuf_v1 version 4
[INFO:display.cc(139)] Wayland: zwp_keyboard_shortcuts_inhibit_manager_v1 version 1
[INFO:display.cc(139)] Wayland: zwp_text_input_manager_v3 version 1
[INFO:display.cc(139)] Wayland: wp_presentation version 1
[INFO:display.cc(139)] Wayland: xdg_activation_v1 version 1
[INFO:display.cc(78)] - Display()
[INFO:app.cc(29)] +App::App
libEGL warning: pci id for fd 5: 1ab8:0010, driver (null)

pci id for fd 8: 1ab8:0010, driver (null)
[INFO:egl.cc(33)] EGL major: 1, minor 4
[INFO:egl.cc(606)] EGL_KHR_debug initialized
[INFO:egl.cc(43)] EGL has 140 configs
[INFO:egl.cc(60)] Buffer size for config 0 is 32
[INFO:window.cc(47)] (0) + WaylandWindow()
[INFO:display.cc(247)] Physical width: 0 mm x 0 mm
[INFO:display.cc(264)] Video mode: 2152 x 1365 @ 59.987 Hz
[INFO:display.cc(276)] Display Scale Factor: 1
[INFO:display.cc(303)] Pointer Present
[INFO:display.cc(312)] Keyboard Present
[INFO:display.cc(336)] Seat: seat0
[INFO:display.cc(673)] [keyboard repeat info] rate: 33, delay: 500
[INFO:window.cc(111)] (0) - WaylandWindow()
[INFO:engine.cc(418)] (0) PersistentCachePath: /home/parallels/Documents/GitHub/meta-flutter/tools/.config/flutter/.config/flutter-auto
[INFO:engine.cc(87)] (0) +Engine::Engine
[INFO:engine.cc(100)] (0) libflutter_engine.so: /tmp/pumped_end_device/lib/libflutter_engine.so
[INFO:engine.cc(133)] (0) flutter_assets: /tmp/pumped_end_device/data/flutter_assets
[INFO:engine.cc(148)] (0) icudtl.dat: /tmp/pumped_end_device/data/icudtl.dat
[INFO:engine.cc(161)] (0) Runtime=debug
[INFO:engine.cc(205)] (0) -Engine::Engine
[INFO:engine.cc(242)] (0) +Engine::Run
[INFO:gl_process_resolver.cc(39)] dlopen: libEGL.so
[INFO:gl_process_resolver.cc(39)] dlopen: 
[INFO:engine.cc(188)] (0) Running Task
[INFO:engine.cc(258)] (0) Engine::m_running = 1
[INFO:engine.cc(283)] (0) -Engine::Run
[INFO:engine.cc(334)] (0) SetWindowSize: width=1920, height=1080, pixel_ratio=1
[INFO:flutter_view.cc(97)] (0) Engine running...
[INFO:app.cc(50)] -App::App
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:21.828151]: DEBUG: UiSettingsDao : Retrieving instance of UiSettings
[INFO:secure_storage.cc(102)] secure_storage: [Read] key: ped_ui_settings_coll-ped_ui_settings
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.115985]: DEBUG: PumpedApp : Registering instance of geoLocationWrapper
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.117952]: DEBUG: PumpedApp : Registering instance of locationDataSource
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.118773]: DEBUG: PumpedApp : Registering instance of underMaintenanceService
[INFO:platform.cc(41)] (0) Platform: ApplicationSwitcherDescription
	label: ""
	primaryColor: 4282602937
[INFO:navigation.cc(32)] (0) Navigation: Select Single Entry History
[INFO:navigation.cc(55)] (0) Navigation: Route Information Updated
	location: /
	state: 
	replace: 0
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.402972]: DEBUG: SplashScreen : Pumped Backend not scheduled for maintenance, ped is up and available for business
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.403719]: DEBUG: UiSettingsDao : Retrieving instance of UiSettings
[INFO:secure_storage.cc(102)] secure_storage: [Read] key: ped_ui_settings_coll-ped_ui_settings
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.544912]: DEBUG: SplashScreen : Pumped Backend not scheduled for maintenance, ped is up and available for business
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.575543]: DEBUG: MockLocationDao : getting the pinned mock location
[INFO:engine.cc(334)] (0) SetWindowSize: width=1920, height=1080, pixel_ratio=1
[INFO:secure_storage.cc(102)] secure_storage: [Read] key: ped_pinned_mock_loc_collection-ped_pinned_mock_loc_doc_id
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.587188]: DEBUG: LocationDataSource : No pinned location found
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.587831]: DEBUG: LocationDataSource : Overriding users actual location with static value, temporary dev workaround for AGL Web and Linux
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.588729]: DEBUG: SplashScreen : latitude : -33.865107, longitude : 151.205282
[INFO:engine.cc(83)] (0) flutter: [2023-04-10 16:40:22.603176]: DEBUG: SplashScreen : Pumped Backend not scheduled for maintenance, ped is up and available for business
Syncing files to device Toyota flutter-auto...                     123ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

5. setup is no-longer bringing down QEMU_IMAGE
when running ./setup_flutter_workspace.py --clean it is no-longer downloading the agl image flutter ?
QEMU_IMAGE=${FLUTTER_WORKSPACE}/.agl/agl-image-flutter-runtimedebug-qemux86-64.wic.vmdk

If you can advise on either of these appreciated

regards

jwinarske, 2023-04-12:
@bernardcraddock Hey.  Tip of tree 3.7.x has issues that I haven't resolved yet. AGL and Toyota are still on 3.3.7.  You are safe to switch to 3.3.10 or earlier.  3.7.x is not a high priority right now.  I'm figuring a couple of weeks.

Also heads up work for workspace automation is moving.  It has been refactored for easier configs, pulls AGL images from AGL server, etc:
https://github.com/meta-flutter/workspace-automation

bernardcraddock, 2023-04-12:
@jwinarske - Yes I'm working on 3.7.x and super busy with android, ios and linux (desktop) so i can wait a couple of weeks no problem.

thanks for the heads up i'll switch to new repo shortly

jwinarske, 2023-07-03:
@bernardcraddock heads up I have a fix in place for EGL impeller (flickering).  Vulkan works fine as-is.  Hoping to have it released as OSS release by Friday.  This supports 3.10.5, and will allow me to roll AGL the following week.

bernardcraddock, 2023-07-04:
@jwinarske - Super thank you
I've just now pulled and reviewed your latest changes both in meta-flutter/kirkstone and workspace-automation/main
and will test it out over coming days and advise if i encounter any issues

regards
                                                                                                                                                                    
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                                         
                                                                                                                                                                                        your changes

jwinarske, 2023-11-16:
This has been resolved some time back

closed 2023-11-16 by jwinarske