meta-flutter/meta-flutter #278: wayland-client.h file not found

Krunalb11, 2023-04-14:
I am building flutter-pi with kirkstone branch and by default, it is using FLUTTER_SDK_TAG = "3.7.7". While building flutter-pi, I see flutter-engine fail with the following error:
| In file included from ../../flutter/flutter_vma/flutter_vma.cc:19:
| In file included from ../../flutter/flutter_vma/flutter_vma.h:7:
| In file included from ../../third_party/vulkan_memory_allocator/include/vk_mem_alloc.h:130:
| ../../third_party/vulkan-deps/vulkan-headers/src/include/vulkan/vulkan.h:41:10: fatal error: 'wayland-client.h' file not found
| #include <wayland-client.h>
|          ^~~~~~~~~~~~~~~~~~
| 1 error generated.

I am wondering if there are hard dependencies on Wayland.

jwinarske, 2023-04-17:
@Krunalb11 what are your DISTRO_FEATURES set to?

Krunalb11, 2023-04-17:
@jwinarske , here is the output:

DISTRO_FEATURES="acl alsa argp bluetooth debuginfod ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc  vfat seccomp \${DISTRO_FEATURES_LIBC} pam wayland opengl opencv openmp systemd pulseaudio sysvinit gobject-introspection-data ldconfig"
DISTRO_FEATURES_BACKFILL="pulseaudio sysvinit gobject-introspection-data ldconfig"
DISTRO_FEATURES_DEFAULT="acl alsa argp bluetooth debuginfod ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc x11 vfat seccomp"
DISTRO_FEATURES_FILTER_NATIVE="api-documentation debuginfod opengl wayland opencl opencv openmp"
DISTRO_FEATURES_FILTER_NATIVESDK="api-documentation debuginfod opengl wayland opencl opencv openmp"
DISTRO_FEATURES_NATIVE="x11 ipv6 xattr"
DISTRO_FEATURES_NATIVESDK="x11"

If Flutter has dependency on Wayland, it should be defined in the packgroups or depends of the flutter recipe?

jwinarske, 2023-04-17:
@Krunalb11 what branch and commit are you using?

jwinarske, 2023-04-18:
@Krunalb11 I see.  I am suspecting the TI BSP requires an explicit recipe to pull in the wayland header.  This is would be issue specific to TI BSP.  Is it public?  flutter-pi is DRM only, so no compositor/wayland.  It's most likely solved with a bbappend adding the dependency that pulls in Wayland header.  If it's public I can stage a build and take a look.