meta-flutter/meta-flutter #281: flutter-engine fails while it is compiling

Gmatarrubia, 2023-04-23:
My project is based on a raspberry pi 3b machine.
I had successfully been compiling it before when It was based on honister branch. Now, I'm trying to port my work to the kirkstone branch. And now, I'm facing a very weir error that I am not able to diagnostic. Every hint is welcome.

Here the error message of flutter-engine compiling task:

```
| [3130/5162] STAMP clang_x64/obj/third_party/dart/runtime/platform/libdart_platform_precompiler.stamp
| [3131/5162] ACTION //third_party/dart/runtime/observatory:build_observatory(//build/toolchain/custom:custom)
| FAILED: gen/third_party/dart/runtime/observatory/observatory/web/main.dart.js
| vpython3 ../../third_party/dart/build/gn_run_binary.py compiled_action /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/dart --dfe=/yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot compile js -o /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/out/linux_debug_arm/gen/third_party/dart/runtime/observatory/observatory/web/main.dart.js --packages=/yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/.dart_tool/package_config.json /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/runtime/observatory/web/main.dart --minify
| Command failed: /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/dart --dfe=/yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot compile js -o /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/out/linux_debug_arm/gen/third_party/dart/runtime/observatory/observatory/web/main.dart.js --packages=/yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/.dart_tool/package_config.json /yocto/raspberrypi3/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/flutter-engine/git-r0/src/third_party/dart/runtime/observatory/web/main.dart --minify
| output: Info: Compiling with sound null safety.
| 
| [3132/5162] CXX clang_x64/obj/third_party/dart/runtime/bin/libdart_builtin.file_linux.o
| [3133/5162] CXX clang_x64/obj/third_party/dart/runtime/bin/libdart_builtin.file.o
```

jwinarske, 2023-05-11:
your best bet is to follow the pattern from here:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-rpi4-64-vulkan.yml

jwinarske, 2023-07-03:
@Gmatarrubia any update on this?

Gmatarrubia, 2023-07-03:
I took your configuration as reference. But it has a lot of automotive stuff I don't need. I still facing the same issue for the kirkstone branch. I noticed currently the github pipelines are not running. Looks like the dark SDK is built with null-safety and some files are not prepared for null-safety feature. 

Honister branch is keeping working for me, even for a clean build.

jwinarske, 2023-07-05:
@Gmatarrubia Are you targeting 32-bit or 64-bit for RPI 3B?  

I think the honister branch is the most complete RPI 3B image I've done.  I'll consider adding one to kirkstone.

CI has been down for a while.  Hit a hardware issue with my raid array.  Since fixed that.  Updated the self hosted runner to red hat.  Was hoping to get podman working with action-runner.  Should have some time to finish in a couple of weeks

Gmatarrubia, 2023-07-06:
I'm targeting 32-bit Rbi Pi 3B. I use flutter-pi and flutter-engine with vulkan graphics.

jwinarske, 2023-07-06:
Any reason for 32-bit?

jwinarske, 2023-10-13:
closing due to inactivity

closed 2023-10-13 by jwinarske