meta-flutter/meta-flutter #283: Copyright statements don't go well with the layer license

kwavnet, 2023-05-09:
```
Copyright (c) 2020-2023 Joel Winarske. All rights reserved.
```

is the copyright statement in most (if not all) of the files in the layer, but the layer license has the following clause in it

```
including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software
```

so this doesn't go well with each other. INAL but one could read it as if non of the files containing that statement are licensed under MIT.

I think these CR statements should be more like this

```
#
# Copyright (c) 2020-2023 Joel Winarske
#
# SPDX-License-Identifier: MIT
#
```

and in addition I would say it makes sense to even turn that into something machine readable

```
# SPDX-FileCopyrightText: (C) 2020-2023 Joel Winarske
# SPDX-License-Identifier: MIT
```

so one could use [reuse](https://github.com/fsfe/reuse-tool) to automatically verify things.

It might be very nitpicky, but (and again INAL) I think every file that has seen contributions from more than one person needs to name everyone.
An alternative would be the way upstream YP does it, with a generic term like

```
# SPDX-FileCopyrightText: (C) 2020-2023 meta-flutter contributors
```

or something

jwinarske, 2023-05-11:
@kwavnet Yeah that makes.  I like your suggestion of using:

    # SPDX-FileCopyrightText: (C) 2020-2023 meta-flutter contributors
    # SPDX-License-Identifier: MIT

jwinarske, 2026-08-25:
Revisiting this. It is still unimplemented — 277 of 319 tracked files carry `All rights reserved` — and surveying the layer to scope the work turned up something more serious than what was originally reported.

## The layer currently makes two contradictory license claims

Five files under `tools/` declare `SPDX-License-Identifier: Apache-2.0`:

```
tools/common.py  tools/create_recipes.py  tools/pubspec.py
tools/roll_meta_flutter.py  tools/update_version_files.py
```

The layer's only `LICENSE` is MIT. There is no `LICENSES/` directory, no `REUSE.toml`, and nothing in `README.md` or `conf/layer.conf` that scopes `tools/` differently. So the repository asserts MIT at the root and Apache-2.0 in five files, with nothing reconciling the two.

That is a genuine licensing defect rather than a cosmetic one, and it is separable from the header sweep, so it is worth fixing first.

## The reported contradiction is now self-contained

Three files carry both `All rights reserved` and an SPDX identifier in the same header:

```
conf/include/app-utils.inc   conf/include/common.inc   tools/create_recipes.py
```

`tools/create_recipes.py` manages all three claims simultaneously: MIT by repository, `Apache-2.0` by header, `All rights reserved` by copyright line. That is exactly the ambiguity described in the original report, now visible within a single file.

## A sweep on its own will not hold

```
tools/create_recipes.py:294   f.write('# Copyright (c) 2020-2025 Joel Winarske. All rights reserved.\n')
tools/create_recipes.py:414   f.write('# Copyright (c) 2020-2025 Joel Winarske. All rights reserved.\n')
```

Generated recipes get the header stamped into them, so any cleanup that does not also patch those two emit sites reintroduces the problem on the next roll. `lib/gn.py` carries the header as well. This is what makes it a code change rather than a one-shot `sed`.

## Year ranges are pure churn

Eight distinct ranges currently exist for the same author — `2020-2025` (212 files), `2020-2024` (32), `2020-2026` (14), `2026` (6), `2024` (5), plus singletons — while `LICENSE` itself still reads `2018-2021`. Nothing keeps any of it in sync. SPDX and REUSE do not require years, so dropping the ranges removes the entire category of drift.

## Care needed when mechanising

A blanket substitution would erase real third-party attribution:

```
Copyright (c) 2020-2022 Woven Alpha, Inc        (2 files)
Copyright (c) 2026 Ahmed Wafdy
Copyright (c) 2013 The Chromium Authors          (vendored)
```

Those have to survive. The sweep needs to target one specific copyright line and leave other parties untouched.

## On naming every contributor

History now has 45 distinct contributors. Reconstructing per-file attribution across 277 files retroactively is not practical, so @kwavnet's own fallback — the Yocto-style `meta-flutter contributors` — is the right call.

## Plan

1. Resolve MIT vs Apache-2.0 under `tools/`. Small and self-contained; PR to follow.
2. Sweep headers to the SPDX pair, preserving other parties' copyright lines.
3. Patch the two `create_recipes.py` emit sites in the same PR as (2), or it undoes itself.
4. Optionally add `reuse lint` to CI, which is what keeps it from drifting again.

One practical note on (2): a 277-file header sweep touches nearly every recipe and will conflict with anything in flight across the release branches, so it is best landed when the backport queue is quiet — or applied to master only, accepting that the release branches keep the older headers rather than replaying the churn four times.

jwinarske, 2026-08-28:
Done across all five branches.

Headers now carry the copyright line without the reservation, plus an explicit identifier — the form suggested in the issue:

```
#
# Copyright (c) 2020-2025 Joel Winarske
#
# SPDX-License-Identifier: MIT
#
```

## Generators first, then the files

`tools/create_recipes.py` writes that header into every recipe it produces, at two sites, and `lib/gn.py` carried it in its own module docstring. Those were fixed **before** the sweep — otherwise the tree would have regressed the moment anyone regenerated a recipe. Three lines protecting ~275 files.

| branch | files swept | `All rights reserved` remaining | files with SPDX |
|---|---|---|---|
| master | 271 | 4 | 280 |
| wrynose | 271 | 4 | 280 |
| scarthgap | 275 | 4 | 284 |
| kirkstone | 276 | 4 | 285 |
| dunfell | 279 | 4 | 288 |

Before this, ten files in the layer carried an SPDX identifier.

## Four files deliberately keep their headers

They carry a copyright holder other than the layer's:

| file | holder |
|---|---|
| `conf/include/gn-utils.inc` | The Flutter Authors |
| `recipes-graphics/flutter-engine/files/BUILD.gn.in` | The Flutter Authors |
| `…/flatpak-minimal-appstream-dart-flathub-catalog_0.4.2.bb` | Ahmed Wafdy |
| `recipes-devtools/dart/files/gcc_toolchain.gni.in` | The Chromium Authors |

Declaring a file MIT while a second holder reserves all rights over it is a claim about their work, not the layer's, and wants their agreement rather than a search and replace. `lib/gn.py`'s second holder (Woven Alpha) is likewise untouched — that line claims no reservation.

So four `All rights reserved` remain in each tree on purpose. If you would like those converted too, that is a conversation with the respective holders rather than a follow-up commit here.

Three spellings were in use — `(c)`, `(C)`, and no parenthesis at all — all normalised. Files that already had an identifier keep the one they had.

## Verification

`bitbake -p` over the full layer: 3162 recipes on wrynose and 3526 on master, 0 errors in both. wrynose additionally went green on all four machines with the ten integration tests and three FFI applications building, which is the check that catches a header edit a parse would tolerate.

PRs: #810, #811 (wrynose), #812 (master), #813 (scarthgap), #814 (kirkstone), #815 (dunfell).

closed 2026-08-28 by jwinarske