meta-flutter/meta-flutter #285: Testing Flutter on am62xx meta-ti kirkstone latest

munoz0raul, 2023-05-09:
Hi 

meta-ti recently add a lot of patches to enable the graphics on kirkstone. I tested weston-simple-egl today and works well so I decided to test Flutter.

First build complained about Vulkun libs, So I decided to remove Vulkan from the build config:

Before add `DISTRO_FEATURES:remove = "vulkan "`

```
root@am62xx-evm:/var/rootdirs/home/fio# flutter-auto --b=/usr/share/flutter/gallery/3.7.7/release/ 
[INFO:main.cc(85)] agl @ 4dbc64f
[INFO:configuration.cc(284)] **********
[INFO:configuration.cc(285)] * Global *
[INFO:configuration.cc(286)] **********
[INFO:configuration.cc(287)] Application Id: .......... flutter-auto
[INFO:configuration.cc(291)] Cursor Theme: ............ 
[INFO:configuration.cc(292)] Disable Cursor: .......... false
[INFO:configuration.cc(294)] Debug Backend: ........... false
[INFO:configuration.cc(296)] ********
[INFO:configuration.cc(297)] * View *
[INFO:configuration.cc(298)] ********
[INFO:configuration.cc(305)] Bundle Path: .............. /usr/share/flutter/gallery/3.7.7/release/
[INFO:configuration.cc(306)] Window Type: .............. NORMAL
[INFO:configuration.cc(307)] Output Index: ............. 0
[INFO:configuration.cc(309)] Size: ..................... 1920 x 720
[INFO:configuration.cc(314)] Fullscreen: ............... false
[INFO:configuration.cc(316)] Accessibility Features: ... 0
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/third_party/filament/libs/bluevk/src/BlueVKLinuxAndroid.cpp(54)] Unable to load Vulkan from libvulkan.so.1
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/wayland_vulkan.cc(38)] BlueVK is unable to load entry points.
```

After add `DISTRO_FEATURES:remove = "vulkan "` I got this error:

```
root@am62xx-evm:/var/rootdirs/home/fio# flutter-auto --b=/usr/share/flutter/gallery/3.7.7/release/
[INFO:main.cc(85)] agl @ 4dbc64f
[INFO:configuration.cc(284)] **********
[INFO:configuration.cc(285)] * Global *
[INFO:configuration.cc(286)] **********
[INFO:configuration.cc(287)] Application Id: .......... flutter-auto
[INFO:configuration.cc(291)] Cursor Theme: ............ 
[INFO:configuration.cc(292)] Disable Cursor: .......... false
[INFO:configuration.cc(294)] Debug Backend: ........... false
[INFO:configuration.cc(296)] ********
[INFO:configuration.cc(297)] * View *
[INFO:configuration.cc(298)] ********
[INFO:configuration.cc(305)] Bundle Path: .............. /usr/share/flutter/gallery/3.7.7/release/
[INFO:configuration.cc(306)] Window Type: .............. NORMAL
[INFO:configuration.cc(307)] Output Index: ............. 0
[INFO:configuration.cc(309)] Size: ..................... 1920 x 720
[INFO:configuration.cc(314)] Fullscreen: ............... false
[INFO:configuration.cc(316)] Accessibility Features: ... 0
[INFO:engine.cc(577)] (0) Loading AOT: /usr/share/flutter/gallery/3.7.7/release//lib/libapp.so
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(552)] **** EGL Error
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(553)] 	error: 12291
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(554)] 	command: eglMakeCurrent
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(560)] 	error: EGL_BAD_ALLOC
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(596)] 	messageType: 13241
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(597)] 	threadLabel: (nil)
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(598)] 	objectLabel: (nil)
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(599)] 	message: failed to allocate color buffer
libEGL warning: DRI2: failed to rebind the previous context
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(552)] **** EGL Error
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(553)] 	error: 12297
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(554)] 	command: eglMakeCurrent
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(578)] 	error: EGL_BAD_MATCH
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(596)] 	messageType: 13242
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(597)] 	threadLabel: (nil)
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(598)] 	objectLabel: (nil)
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(599)] 	message: eglMakeCurrent
[ERROR:/usr/src/debug/flutter-auto/git-r0/git/shell/backend/egl.cc(98)] Make current failed: 12297
[FATAL:flutter/shell/gpu/gpu_surface_gl_delegate.cc(57)] Check failed: gl_version_string. The GL proc resolver's glGetString(GL_VERSION) failed
Aborted
```

So any suggestion on what could be happening?

munoz0raul, 2023-06-01:
Hi @jwinarske I made some progress on the am62xx.

I add some patches that are still on the list and did some adjusts to have all libs in place.

Now I'm getting a different error. I do have vulkan enalbed!

```
flutter-auto --b=/usr/share/flutter/gallery/3.7.7/release/ --w=800 --h=600
[INFO:main.cc(85)] agl @ 4dbc64f
[INFO:configuration.cc(284)] **********
[INFO:configuration.cc(285)] * Global *
[INFO:configuration.cc(286)] **********
[INFO:configuration.cc(287)] Application Id: .......... flutter-auto
[INFO:configuration.cc(291)] Cursor Theme: ............ 
[INFO:configuration.cc(292)] Disable Cursor: .......... false
[INFO:configuration.cc(294)] Debug Backend: ........... false
[INFO:configuration.cc(296)] ********
[INFO:configuration.cc(297)] * View *
[INFO:configuration.cc(298)] ********
[INFO:configuration.cc(305)] Bundle Path: .............. /usr/share/flutter/gallery/3.7.7/release/
[INFO:configuration.cc(306)] Window Type: .............. NORMAL
[INFO:configuration.cc(307)] Output Index: ............. 0
[INFO:configuration.cc(309)] Size: ..................... 800 x 600
[INFO:configuration.cc(314)] Fullscreen: ............... false
[INFO:configuration.cc(316)] Accessibility Features: ... 0
[INFO:wayland_vulkan.cc(135)] Enabling 2 instance extensions:
[INFO:wayland_vulkan.cc(138)]   - VK_KHR_surface
[INFO:wayland_vulkan.cc(138)]   - VK_KHR_wayland_surface
[INFO:wayland_vulkan.cc(384)] vendor 1010, device 33011003, driver 61b995, api 1.3
[INFO:engine.cc(577)] (0) Loading AOT: /usr/share/flutter/gallery/3.7.7/release//lib/libapp.so
[ERROR:flutter/shell/common/platform_view.cc(74)] Failed to create platform view rendering surface
```

On the strace, the error seems to be on memory allocation: 

```
munmap(0x7fbc7a9000, 608)               = 0
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\26\0\0\0\0\0\10\0\24\0\0\0\3\0\20\0\27\0\0\0\v\0\0\0\1\0\0\0\0\0\f\0"..., iov_len=36}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_DONTWAIT|MSG_NOSIGNAL) = 36
ppoll([{fd=3, events=POLLIN}], 1, NULL, NULL, 0) = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\0\1\0\f\0\26\0\0\0\27\0\0\0\1\0\f\0`\2\0\0\27\0\0\0\2\0\24\0"..., iov_len=2956}, {iov_base="", iov_len=1140}], msg_iovlen=2, msg_control=[{cmsg_len=20, cmsg_level=SOL_SOCKET, cmsg_type=SCM_RIGHTS, cmsg_data=[15]}], msg_controllen=24, msg_flags=MSG_CMSG_CLOEXEC}, MSG_DONTWAIT|MSG_CMSG_CLOEXEC) = 204
mmap(NULL, 608, PROT_READ, MAP_PRIVATE, 15, 0) = 0x7fbc7a9000
close(15)                               = 0
openat(AT_FDCWD, "powervr.ini", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/powervr.ini", O_RDONLY) = -1 ENOENT (No such file or directory)
ioctl(5, DRM_IOCTL_AMDGPU_GEM_CREATE or DRM_IOCTL_VIA_ALLOCMEM, 0x7fef587198) = 0
ioctl(17, DRM_IOCTL_VERSION, 0x558f62af50) = 0
ioctl(17, DRM_IOCTL_VERSION, 0x558f62af50) = 0
ioctl(17, DRM_IOCTL_MODE_CREATE_DUMB, 0x7fef587438) = -1 ENOMEM (Cannot allocate memory)
ioctl(5, DRM_IOCTL_AMDGPU_GEM_CREATE or DRM_IOCTL_VIA_ALLOCMEM, 0x7fef587278) = 0
write(2, "[ERROR:/usr/src/debug/flutter-au"..., 115) = 115
write(2, "[ERROR:/usr/src/debug/flutter-au"..., 112) = 112
exit_group(1)                           = ?
+++ exited with 1 +++
```

`ioctl(17, DRM_IOCTL_MODE_CREATE_DUMB, 0x7fef587438) = -1 ENOMEM (Cannot allocate memory)
`

Any suggestions?

jwinarske, 2023-06-02:
Could be a missing dt entry.  Does egl behave the same way?  vulkaninfo returns without error?  It could also be a Compositor path issue.  I would start with Vulkaninfo, then when stack is properly loading try other Vulkan apps, small then bigger.  I merged some updates from a TI dev into meta-vulkan.  He was adding x11 support...  It might be they only have x11 flushed out.

https://github.com/jwinarske/meta-vulkan

jwinarske, 2023-06-02:
@StaticRocket 
Tested targets images for meta-vulkan prior to your recent changes were Wayland (no X11) for:  
* Raspberry Pi 4
* Qualcomm SA8155

jwinarske, 2023-10-11:
@munoz0raul Heads up.  I'm back on AGL work as we locked for production cycle.  I just verified ivi-homescreen running GLES2/EGL backend on the SK-TDA4VM (PowerVR GE???).  Same SoC as on the BeagleBone AI64.  This is with X11 removed from the image.  For Vulkan there is a hard dependency on X11; which needs to be addressed.  I did just order a SK-AM69.  Automotive Grade Linux (AGL) is planning to pick up support for the BeagleBone AI64.  I may be demoing future tech at CES with the SK-AM69.  X11 is out of the question across the board.

@StaticRocket why are the X11 deps required, and what is a timeline to factor out the dependencies?

StaticRocket, 2023-10-11:
@jwinarske WSI extensions. IMG has their own implementations instead of relying on the dynamic window system extensions provided by mesa. This introduces a hard runtime dep. We were experimenting with Zink on beagle so we've been enabling the X extensions but after some debate this may not be reasonable moving forward. I'll drop them at some point. Right now I'm still reanimating the corpse of SGX for legacy devices.

jwinarske, 2023-10-11:
@StaticRocket ah bummer.  I see a CPU load decrease of ~48% with ivi-homescreen running the Vulkan backend vs GLES2/EGL.  I am raising the question on their forum to see what I get.  Thanks.

@munoz0raul  SK-TDA4VM
```
Apr 28 10:42:34 tda4vm-sk weston[551]: [10:42:34.985] GL version: OpenGL ES 3.2 build 23.1@6404501
Apr 28 10:42:34 tda4vm-sk weston[551]: [10:42:34.986] GLSL version: OpenGL ES GLSL ES 3.20 build 23.1@6404501
Apr 28 10:42:34 tda4vm-sk weston[551]: [10:42:34.986] GL vendor: Imagination Technologies
Apr 28 10:42:34 tda4vm-sk weston[551]: [10:42:34.986] GL renderer: PowerVR Rogue GE8430
```

jwinarske, 2023-10-17:
@StaticRocket I think the solution to remove link time dependencies is extremely easy.  Function pointers.  Dlopen, check the version, assign the function pointers.  Is that something imagination would need to provide?

StaticRocket, 2023-10-17:
As it stands currently, yes. Though I would much rather they just use the mesa standard WSI extensions at that point... We'll bring it up with them. Bigger fish to fry at the moment though.

jwinarske, 2023-10-23:
@StaticRocket response from Imagination support:
```
After checking with our DDK linux team on this, I was told that “For Vulkan, the various supported window systems are specified when the DDK is built, using the SUPPORT_VK_PLATFORMS environment variable” which is set according to the platforms supported in Yocto e.g. which may not be the case with TI. They think the problem here could be that the IMG DDK Vulkan library is closed source, and so TI build it with all window systems enabled, to cover all potential use cases, and distribute it in binary form to customers" To confirm which we would need to discuss with TI if that is the case here.
``` 
It's clearly an implementation issue on their end.

jwinarske, 2026-08-30:
Closing this as stale — it ran its course in 2023 and the remaining blocker was never ours.

Where it landed, for anyone finding this later: Flutter came up on am62xx once Vulkan was taken out of the build config, and the thread then narrowed to the Imagination DDK's Vulkan WSI. Per Imagination support, the supported window systems are fixed when the DDK is built (`SUPPORT_VK_PLATFORMS`), so a binary DDK built without Wayland WSI cannot be made to provide it downstream. The suggestion of resolving the entry points at runtime via `dlopen` and function pointers would remove the link-time dependency, but it needs the DDK to expose them in the first place.

Nothing actionable in meta-flutter, and three years on the platform has moved. Happy to reopen if this is still live on a current meta-ti — but a fresh issue against a current release would probably be more useful than reviving this one.

Thanks @munoz0raul and @StaticRocket for working it through at the time.

closed 2026-08-30 by jwinarske