meta-flutter/meta-flutter #286: Flutter app build: `engine_sdk/sdk/clang_x64/gen_snapshot` not found

edrose, 2023-05-16:
I'm having a weird issue when trying to build a flutter app using 3.7.7 - just tested this morning with the latest commit but it's been around for a few commits at least.

Flutter engine compiles successfully, but during the app compile it fails. Here is the output from a build of `flutter-gallery`:

```
| NOTE: [profile] flutter build bundle: Completed
| 
| NOTE: kernel_snapshot_profile: Starting
| NOTE: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/dart-sdk/bin/dart                 --verbose                 --disable-analytics                 --disable-dart-dev /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/linux-x64/frontend_server.dart.snapshot                 --sdk-root /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/common/flutter_patched_sdk/                 --target=flutter                 --no-print-incremental-dependencies                 -Ddart.vm.profile=true                 -Ddart.vm.product=false                                   --track-widget-creation --aot --tfa                 --packages .dart_tool/package_config.json                 --output-dill .dart_tool/flutter_build/*/app.dill                 --depfile .dart_tool/flutter_build/*/kernel_snapshot.d                 package:gallery/main.dart
| result 7272741e-e30e-4450-8a05-1f5f98cd8769
| ../pub_cache/hosted/pub.dev/flutter_staggered_grid_view-0.6.2/lib/src/widgets/sliver_aligned_grid.dart:191:49: Warning: Operand of null-aware operation '!' has type 'ScrollableState' which excludes null.
|  - 'ScrollableState' is from 'package:flutter/src/widgets/scrollable.dart' ('../recipe-sysroot-native/usr/share/flutter/sdk/packages/flutter/lib/src/widgets/scrollable.dart').
|     final axis = axisDirectionToAxis(Scrollable.of(context)!.axisDirection);
|                                                 ^
| ../pub_cache/hosted/pub.dev/flutter_staggered_grid_view-0.6.2/lib/src/widgets/staggered_grid.dart:85:22: Warning: Operand of null-aware operation '?.' has type 'ScrollableState' which excludes null.
|  - 'ScrollableState' is from 'package:flutter/src/widgets/scrollable.dart' ('../recipe-sysroot-native/usr/share/flutter/sdk/packages/flutter/lib/src/widgets/scrollable.dart').
|           Scrollable.of(context)?.axisDirection ??
|                      ^
| ../pub_cache/hosted/pub.dev/flutter_staggered_grid_view-0.6.2/lib/src/widgets/staggered_grid.dart:101:22: Warning: Operand of null-aware operation '?.' has type 'ScrollableState' which excludes null.
|  - 'ScrollableState' is from 'package:flutter/src/widgets/scrollable.dart' ('../recipe-sysroot-native/usr/share/flutter/sdk/packages/flutter/lib/src/widgets/scrollable.dart').
|           Scrollable.of(context)?.axisDirection ??
|                      ^
| 7272741e-e30e-4450-8a05-1f5f98cd8769
| 7272741e-e30e-4450-8a05-1f5f98cd8769 .dart_tool/flutter_build/80bea4e67a96401f252d3d9eb2a71d32/app.dill 0
| NOTE: kernel_snapshot_profile: Complete
| NOTE: aot_elf_profile: Started
| Archive:  /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/recipe-sysroot/usr/share/flutter/3.7.7/profile/engine_sdk.zip
|    creating: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/
|  extracting: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/flutter_sdk.version
|    creating: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/
|   inflating: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/flatc
|  extracting: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/engine.version
|  extracting: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/flutter.runtime
| /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/temp/run.do_compile.125091: 239: /home/ed/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-tewke-linux-gnueabi/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot: not found
| WARNING: exit code 127 from a shell command.
ERROR: Task (/home/ed/yocto/build/../meta-flutter/recipes-graphics/flutter-apps/flutter-gallery_git.bb:do_compile) failed with exit code '1'
```

So I can see that we are extracting the engine sdk from a zip file, and exepecting `gen_snapshot` to be in that zip file. However the output of `unzip -l` shows this for all versions (release/profile/debug):

```
  Length      Date    Time    Name
---------  ---------- -----   ----
        0  2023-05-15 17:12   sdk/
        1  2023-05-15 17:12   sdk/flutter_sdk.version
        0  2023-05-15 17:12   sdk/clang_x64/
  3005864  2023-05-15 17:12   sdk/clang_x64/flatc
        2  2023-05-15 17:12   sdk/engine.version
        1  2023-05-15 17:12   sdk/flutter.runtime
---------                     -------
  3005868                     6 files
```

I seem to have fixed the issue in my own build system by applying the following patch:

```
diff --git a/recipes-graphics/flutter-engine/flutter-engine_git.bb b/recipes-graphics/flutter-engine/flutter-engine_git.bb
index 5950918..0d8f1ef 100644
--- a/recipes-graphics/flutter-engine/flutter-engine_git.bb
+++ b/recipes-graphics/flutter-engine/flutter-engine_git.bb
@@ -184,6 +184,8 @@ do_install() {
         # create SDK
         install -D -m 0755 ${S}/${BUILD_DIR}/clang_x64/exe.unstripped/analyze_snapshot \
             ${D}${datadir}/flutter/${FLUTTER_SDK_VERSION}/${MODE}/sdk/clang_x64/analyze_snapshot || true
+        install -D -m 0755 ${S}/${BUILD_DIR}/clang_x64/exe.unstripped/gen_snapshot \
+            ${D}${datadir}/flutter/${FLUTTER_SDK_VERSION}/${MODE}/sdk/clang_x64/gen_snapshot || true
         install -D -m 0755 ${S}/${BUILD_DIR}/clang_x64/exe.unstripped/dart \
             ${D}${datadir}/flutter/${FLUTTER_SDK_VERSION}/${MODE}/sdk/clang_x64/dart || true
         install -D -m 0755 ${S}/${BUILD_DIR}/clang_x64/exe.unstripped/flatc \
-- 
```

Have I just worked-around a deeper problem with my build system, or is this a valid fix? Happy to open a pull request if it is the latter.

jwinarske, 2023-05-16:
@edrose what branch/commit?

edrose, 2023-05-16:
Kirkstone branch, `eac8f882bab485f499622b8ed687bbe71476ab19`

kusstas, 2023-05-16:
I have the same issue

jwinarske, 2023-05-17:
@kwavnet   Looks like your PR removed gen_snapshot.  Can submit a fix please?

kwavnet, 2023-05-17:
#287 - sorry for this

jwinarske, 2023-05-17:
@edrose @kusstas tip of tree has fix.

edrose, 2023-05-17:
Thank you, will test later today!

edrose, 2023-05-18:
Thats fixed my issue. Thank you very much.

Is there any scope for testing that flutter apps build within the CI/CD pipeline? It seems like quite an easy thing to catch in the future.

jwinarske, 2023-05-18:
@edrose yes.  My runner has been offline due to btrfs issue.  I'm the meanwhile I've been upgrading the storage HW, and switched to ext4.  Should be back up tomorrow.  Essentially the prior PR would never have passed CI jobs.

closed 2023-05-18 by edrose