meta-flutter/meta-flutter #288: Uninative Support

edrose-tewke, 2023-05-19:
This is by no means a high priority issue, but I've noticed that `meta-flutter` doesn't seem to use uninative.

The issue presents like this: If I perform a build on our build server (Ubuntu 22.04 LTS), the `SSTATE` cache is generated during the build and shared over the network with our developer workstations. If a developer is using an older version of Ubuntu (20.04 LTS in our case), building the app fails because of a `glibc` mismatch. We've fixed the issue internally by keeping everyone on the same Ubuntu version.

I don't know enough about how uninative works to understand this fully, but my understanding was that it provided common versions of things like `glibc` so that the SSTATE cache was compatible across different build hosts - that doesn't seem to be the case when building flutter engine. However I also don't understand this dependency on `glibc`, since flutter is built using `clang`... So this could just be a side-effect of meta-clang.

This may also be something that needs fixing in order to resolve #271

jwinarske, 2023-05-19:
@edrose-tewke 

Yes matching developer versions is a common approach.  The other approach is to create a "simulator" environment (implemented via docker compose) that has runtime and SDK container options.

In my Kirkstone CI jobs I am using a Ubuntu 20 container:
https://github.com/meta-flutter/meta-flutter/tree/kirkstone/.github/workflows
https://github.com/meta-flutter/docker-containers/blob/main/ubuntu-20-dev/Dockerfile

I haven't moved to 22 as last I checked it was not listed as supported.  That may have changed.

uni-native effectively pulls native tools (pre-built) that are expected to match/resolve with the build OS.  If you run a newer OS image than what is available for un-inative you will have problems, and the only resolution is to turn un-inative off in your local.conf.  If you are on a newer host OS than what the Yocto project officially supports for a given release, then your best bet is to use one of the containers from [docker-containers](https://github.com/meta-flutter/docker-containers).

The dependency on glibc is for the RT, and is required for linking.  It's not present/resolved in/by the Clang pre-built toolchain, so it gets manually copied in, and needs to match the runtime environment.

The reproducible build issue is around the flutter build implementation will modify things in the tree on each run.  This requires upstream changes.  Likely to get push back, as it's not a required scenario for the other platforms.

jwinarske, 2023-05-22:
@edrose-tewke What specifically were you hitting with glibc and building flutter application?  What variants of the engine are you including in image?

edrose-tewke, 2023-05-24:
I've just recreated it in a Ubuntu 20.04 docker container.

To recreate it, I didn't build an image. I just ran `bitbake flutter-gallery`. Bitbake is using the SSTATE from our 22.04 build server. At the beginning of the build, `NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/3.9/x...` is displayed, which I believe indicates that uninative is enabled. I believe that it is building all variants of the engine and app since there is nothing overriding the defaults in this configuration.

Here is the output - the full log is 2000 lines long so I've cut out the irrelevant warnings around locale in the middle:

```
DEBUG: Executing shell function do_compile
NOTE: BUILD_LDFLAGS=-L/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -L/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,--enable-new-dtags -Wl,-rpath-link,/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,-rpath,/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/builder/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 mandir=/usr/share/man systemd_system_unitdir=/lib/systemd/system BUILD_CFLAGS=-isystem/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/include -O2 -pipe CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot= -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native= -fvisibility-inlines-hidden BUILD_CCLD=gcc PYTHONHASHSEED=0 GIT_CEILING_DIRECTORIES=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0 systemd_user_unitdir=/usr/lib/systemd/user OBJCOPY=aarch64-tewke-linux-llvm-objcopy BUILD_CPPFLAGS=-isystem/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/include docdir=/usr/share/doc HOME=/tmp/tmp5foojpvp OLDPWD=/builder/yocto/build sysconfdir=/etc BUILD_CXX=g++ systemd_unitdir=/lib/systemd LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot= -fdebug-prefix-map=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native= -Wl,-z,relro,-z,now includedir=/usr/include infodir=/usr/share/info BUILD_RANLIB=ranlib -D FC=aarch64-tewke-linux-gfortran -target aarch64-tewke-linux -march=armv8-a+crc -mlittle-endian -Qunused-arguments -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot BUILD_STRIP=strip libexecdir=/usr/libexec STRINGS=aarch64-tewke-linux-llvm-strings CPP=aarch64-tewke-linux-clang -target aarch64-tewke-linux -march=armv8-a+crc -mlittle-endian -Qunused-arguments -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot -E base_libdir=/lib base_sbindir=/sbin sbindir=/usr/sbin STRIP=aarch64-tewke-linux-llvm-strip base_bindir=/bin PKG_CONFIG_DISABLE_UNINSTALLED=yes BUILD_CC=gcc bindir=/usr/bin PKG_CONFIG_SYSTEM_LIBRARY_PATH=/lib:/usr/lib PKG_CONFIG_DIR=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig PKG_CONFIG_PATH=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig:/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/usr/share/pkgconfig:/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig:/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/usr/share/pkgconfig exec_prefix=/usr AR=aarch64-tewke-linux-llvm-ar READELF=aarch64-tewke-linux-llvm-readelf PERL_HASH_SEED=0 BUILD_FC=gfortran AS=aarch64-tewke-linux-as libdir=/usr/lib sharedstatedir=/com PATH=/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin:/builder/yocto/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/builder/yocto/oe-core/scripts:/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/bin/aarch64-tewke-linux:/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/recipe-sysroot/us [... cut ...] linux/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot)
/builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /builder/yocto/build/tmp/work/armv8a-tewke-linux/flutter-gallery/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot)
WARNING: exit code 1 from a shell command.
```

The last 4 lines contain the messages about the `glibc` version mismatch.

jwinarske, 2023-06-13:
@edrose-tewke I see.

A couple of possibilities
1.  Disable uninative in local.conf, and build on same version that devs are running on.  That would prevent the runtime resolution problem.
2. Isolate glibc and deps to a portable build root, and chroot when running gensnaphot.  Basically all the items when doing an ldd on gensnapshot.

In the case of glibc dep tree we can add optional flag that would install glibc dep tree to the engine SDK.  Then runtime consumption of SDK would use chroot prior to executing on devs machine.

Another option is to require devs to use a container with yocto SDK + flutter SDK (supported) installed for app builds.

jwinarske, 2026-08-25:
Revisiting this against current master. It still reproduces, and the diagnosis in this thread stopped one step short. There are two independent reasons uninative cannot help here, and neither is "meta-flutter doesn't use uninative".

## uninative is already enabled

`INHERIT_DISTRO` in `openembedded-core/meta/conf/distro/defaultsetup.conf` includes `uninative`, and every distro requires that file. meta-flutter never disables it. uninative is active in these builds — it just never touches the binary that fails.

## Reason 1: the relocation hook skips target recipes

`uninative_changeinterp` is an `SSTATEPOSTUNPACKFUNC`, and its first act is to bail out:

```python
# uninative.bbclass
if not (bb.data.inherits_class('native', d) or bb.data.inherits_class('crosssdk', d)
        or bb.data.inherits_class('cross', d)):
    return
```

`flutter-engine` is a plain target recipe — `inherit gn-fetcher features_check pkgconfig`, no `BBCLASSEXTEND`. uninative returns before walking a single file of its sstate output.

The other half of uninative, the link-time `-Wl,--dynamic-linker=${UNINATIVE_LOADER}` appended to `BUILD_LDFLAGS`, is equally inert: the engine's gn/ninja build supplies its own link flags and never reads `BUILD_LDFLAGS`.

## Reason 2: even for a native recipe, the zip hides it

`uninative_changeinterp` walks `SSTATE_INSTDIR` and patchelfs each dynamic ELF it finds. `gen_snapshot` is not a file there — it is inside an archive:

```
flutter-engine_git.bb:530   zip -r engine_sdk.zip sdk ; rm -rf sdk
flutter-engine_git.bb:621   installs .../engine_sdk.zip
common.inc:562              unzip ${STAGING_DATADIR}/flutter/.../engine_sdk.zip -d .../engine_sdk
common.inc:570              exec engine_sdk/sdk/clang_x64/gen_snapshot
```

`os.walk` sees one opaque blob, `oe.qa.ELFFile` raises `NotELFFileError`, and it moves on. The unzip happens in the *application's* `WORKDIR` at build time, long after any relocation window has closed. The binary keeps whatever interpreter and glibc symbol versions the build host gave it — which is exactly the `GLIBC_2.32/2.33/2.34 not found` in the report above.

Note that it is `STAGING_DATADIR`, not `STAGING_DATADIR_NATIVE`: a build-host executable is being delivered through the *target* sysroot. That is the underlying design issue, and it is why neither uninative path was ever going to engage.

## On the clang question

> I also don't understand this dependency on `glibc`, since flutter is built using `clang`... So this could just be a side-effect of meta-clang.

It is not. Compiler choice does not determine libc — clang links against the host glibc just as gcc does. This would happen identically with a gcc-built engine, so meta-clang is not implicated.

## What a fix looks like

Split the host-side tools — `gen_snapshot`, `dart`, `flatc`, `analyze_snapshot`, `impellerc` — into a recipe that inherits `native`, staged **unzipped** into `recipe-sysroot-native`. Both uninative mechanisms then apply for free: `BUILD_LDFLAGS` at link time, `uninative_changeinterp` at sstate unpack. No meta-flutter-specific machinery, no chroot, and no container requirement — the three workarounds floated earlier in this thread all become unnecessary.

That is a real restructuring of `flutter-engine` rather than a config knob, so it is not a quick fix. The interim answer stays what it was: build sstate on the oldest glibc you intend to consume it on.

## Relation to #271

The suggestion that this blocks #271 is partly right, for a reason worth separating out. Shipping a zip makes the engine's host tooling opaque to every sstate-level mechanism, reproducibility checks included, and the mtimes and entry ordering stored inside the archive are their own reproducibility hazard on top of the binaries'. Un-zipping serves both issues; the native split is what serves this one specifically.

Keeping this open — it is a genuine unfixed design issue, not stale noise.

edrose-tewke, 2026-09-18:
I'm happy to close this since I think I just didn't understand it fully when I made the issue. I think a big part of our uninative problems came down to an interaction between meta-clang and a workaround for rust bindgen. I was wrong when I believed that it was meta-flutter that caused it.

Also sorry for the swapping of accounts. I'm using @edrose from now on.

jwinarske, 2026-09-18:
No worries, and thanks for coming back to it.  Account swap noted.

Closing this one, but the digging wasn't wasted -- it turned up a real defect that just isn't the one in the title.  uninative is enabled and does nothing for us: gen_snapshot ships inside engine_sdk.zip, staged through the target sysroot, so the relocation hook never sees a file to patch.  Still true on master today.

Filed as #1009 with the diagnosis and what the fix looks like.  The title here ("Uninative Support") sends people looking for a config knob, which is why it's better off in its own issue.

closed 2026-09-18 by jwinarske