meta-flutter/meta-flutter #294: FLUTTER_SDK_TAG > 3.7.7 not working

Markus43, 2023-06-13:
Hi,

Thanks for your work on this repository! There have not been updates to the readme file for some time - may I ask about your vague plans  for Flutter 3.7 and 3.10 with meta-flutter and flutter-auto?

After seeing [this ](https://github.com/meta-flutter/meta-flutter/commit/8afaf07f26286ff086ee6ca4b2667b3322cd4fe8) commit that rolls JSON files I expected flutter 3.7.12 to be buildable via FLUTTER_SDK_TAG. However, this doesn't work: The scripts parsing the version fail because [releases_linux.json](https://github.com/meta-flutter/meta-flutter/blob/kirkstone/conf/include/releases_linux.json) doesn't contain versions newer than 3.7.7.

Would be nice to get latest 3.7 (I assume many people that stick with 3.7 will just want to use the latest bugfix release 3.7.12) and a current 3.10 version of flutter to build with this repo here - more and more pub.dev packages stop supporting flutter 3.3 or even 3.7 ...

Is there some script to update the files so I could create a PR for this? Or how are these files usually updated?

Best regards
Markus

jwinarske, 2023-06-13:
@Markus43 hello

Kirkstone is at 3.7.7.  I haven't bumped it in a little while:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/conf/include/flutter-version.inc

AGL is locked at 3.3.7 for the moment, which matches what Toyota is currently at for now:
https://gerrit.automotivelinux.org/gerrit/gitweb?p=AGL/meta-agl-devel.git;a=tree;f=meta-agl-flutter/recipes-graphics/flutter-engine;h=1e632762c1b84c19784f0175960994dff1d03331;hb=HEAD

Once we wrap up on our current milestone I will be working out moving to impeller for flutter-auto.

While impeller is in stable, that is technically only for Android and iOS.  The embedder was stated by Google when impeller started coming in to be the lowest priority platform, in regards to support.

Updates are all done with CI builds.  I change the inc file, spin the build matrix, work out the patches.  Each CI job represents a certain combo, so each one needs to pass.

That said you are welcome to create a WIP PR, and I will use it the next time I roll it.  If you have tested with adequate coverage, or the change is low risk then I will merge it before then.

At Toyota we don't use any public pubdev packages, and I'm currently in Japan for a few more weeks.

Markus43, 2023-06-29:
> Updates are all done with CI builds. I change the inc file, spin the build matrix, work out the patches. Each CI job represents a certain combo, so each one needs to pass.

So the CI tests are only done for the "default" FLUTTER_SDK_TAG in flutter_version.inc ? Did I understand this correctly?

Hmm... But how do you fill the files engine-revision.json , dart-revision.json and releases_linux.json when new versions of flutter are released? Is this done by hand, or are you using some kind of script here?

jwinarske, 2023-06-29:
https://github.com/meta-flutter/workspace-automation
The CLI option is `--version-files`

jwinarske, 2023-06-30:
@Markus43 CI tests are done for the "default" FLUTTER_SDK_TAG in flutter_version.inc.  There is no regression done in CI.  This would be a very large support matrix

jwinarske, 2023-07-01:
Bumped to 3.10.5 yesterday for meta-flutter, workspace-automation just now.  Working on moving to impeller for AGL late next week.

jwinarske, 2023-07-18:
Bumped to 3.10.6 yesterday.  I will update AGL to 3.10.x when Scott is back from holidays

closed 2023-07-18 by jwinarske