meta-flutter/meta-flutter #304: Flutter engine compilation issue (using flutter-app.bbclass)

fleroyener, 2023-07-02:
Hi,

I'm using the repo with the last commit (3.10.5) and i have this issue during compilation : 

```
| FAILED: gen/third_party/dart/utils/dartdev/generate_dartdev_snapshot.dart.dill
| vpython3 ../../third_party/dart/build/gn_run_binary.py compiled_action ../../third_party/dart/tools/sdks/dart-sdk/bin/dart -Dsdk_hash=9830d0d6cb --dfe=../../third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot gen/third_party/dart/utils/gen_kernel/bootstrap_gen_kernel.dill --packages=../../third_party/dart/.dart_tool/package_config.json --platform=vm_platform_strong.dill --no-aot --no-embed-sources --no-link-platform --output=gen/third_party/dart/utils/dartdev/generate_dartdev_snapshot.dart.dill --depfile=gen/third_party/dart/utils/dartdev/generate_dartdev_snapshot.dart.dill.d -Dsdk_hash=9830d0d6cb -Ddart.vm.product=false ../../third_party/dart/pkg/dartdev/bin/dartdev.dart
| Command failed: /home/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/dart -Dsdk_hash=9830d0d6cb --dfe=../../third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot gen/third_party/dart/utils/gen_kernel/bootstrap_gen_kernel.dill --packages=../../third_party/dart/.dart_tool/package_config.json --platform=vm_platform_strong.dill --no-aot --no-embed-sources --no-link-platform --output=gen/third_party/dart/utils/dartdev/generate_dartdev_snapshot.dart.dill --depfile=gen/third_party/dart/utils/dartdev/generate_dartdev_snapshot.dart.dill.d -Dsdk_hash=9830d0d6cb -Ddart.vm.product=false ../../third_party/dart/pkg/dartdev/bin/dartdev.dart
| output: ../../third_party/dart/pkg/analysis_server/lib/src/analytics/noop_analytics.dart:18:15: Error: Type 'ToolInfo' not found.
|   Map<String, ToolInfo> get parsedTools => throw UnimplementedError();
|               ^^^^^^^^
| ../../third_party/dart/pkg/analysis_server/lib/src/server/driver.dart:241:41: Error: The getter 'okToSend' isn't defined for the class 'Analytics'.
|  - 'Analytics' is from 'package:unified_analytics/src/analytics.dart' ('../../third_party/dart/third_party/pkg/tools/pkgs/unified_analytics/lib/src/analytics.dart').
| Try correcting the name to the name of an existing getter, or defining a getter or field named 'okToSend'.
|       return analyticsManager.analytics.okToSend;
|                                         ^^^^^^^^
| 
```

I have tried to add bbappend with this inside but no changes : 
`GN_ARGS:append = " --no-prebuilt-dart-sdk"`

Did you have anything to propose for fixing this issue ? 
Distro is `distro: fsl-imx-xwayland`

jwinarske, 2023-07-02:
@fleroyener I would expect to see this if you had same engine commit archived before yesterdays commit.  The fix is simply to:
* `bitbake flutter-engine -cdo_cleansstate`
* delete the archive from your download directory
* re-run `bitbake flutter-engine`

fleroyener, 2023-07-03:
Thanks !  It solved the issue.
I have another issue now with rendering but I will create another topic because it's different issue.

closed 2023-07-03 by fleroyener