meta-flutter/meta-flutter #307: Error while building the recipe depot-tools-native

ratzdi, 2023-07-27:
Hi,

I need support for my strange problem on building the recipe "dopot-tools-native". 
Does anyone know the reason? The problem seems to be related with  virtualenv  and  compile build-step of the recipe.  
```
gclient --version
[E2023-07-27T14:46:55.961819+02:00 2719995 0 annotate.go:273] original error: permission denied

goroutine 1:
#0 go.chromium.org/luci/vpython/venv/venv.go:310 - venv.(*Env).ensure()
  reason: failed to ensure VirtualEnv

#1 go.chromium.org/luci/vpython/venv/venv.go:154 - venv.With()
  reason: failed to create empty probe environment

#2 go.chromium.org/luci/vpython/run.go:62 - vpython.Run()
#3 go.chromium.org/luci/vpython/application/application.go:320 - application.(*application).mainImpl()
#4 go.chromium.org/luci/vpython/application/application.go:408 - application.(*Config).Main.func1()
#5 go.chromium.org/luci/vpython/application/support.go:46 - application.run()
#6 go.chromium.org/luci/vpython/application/application.go:407 - application.(*Config).Main()
#7 vpython/main.go:110 - main.mainImpl()
#8 vpython/main.go:116 - main.main()
#9 runtime/proc.go:225 - runtime.main()
#10 runtime/asm_amd64.s:1371 - runtime.goexit()
```

jwinarske, 2023-07-27:
@ratzdi Can you share more info on how you are using depot-tools-native? 

* Build environment?
* Yocto SDK or Yocto Build?
* Using as part of a flutter recipe build, or attempting to use standalone?
* Branch/commit

ratzdi, 2023-07-28:
Hi,
build-env: debian 11 within docker container (tried also on native ubuntu 22) and I try to build an image not SDK. 

The commit is e327d2369e0db6de2fe0a25b73e0b6910be2c05c (kirkstone). 
I added the recipe [flutter-gallery](https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-apps/flutter-gallery_git.bb) to my own image.

ratzdi, 2023-08-03:
Problem solved. The issue was the **vpython_cpid_cache** directory in my home directory respectively the access rights to this directory. This directory is needed by virtualenv during building of depot-tools-native recipe. It still was there but for any reason it was created by root and not by default user.

closed 2023-08-03 by ratzdi