meta-flutter/meta-flutter #315: flutter-sdk creates files in ~/

aduskett, 2023-08-18:
The .dart directory, .dart-tool directory, and .flutter file are dumped in ~/ when running flutter-configure.
The only way I have figured out how to stop flutter from doing this is to set the HOME environment variable.

Just thought you would like to know!

jwinarske, 2023-08-19:
@aduskett These are the magic variables to sandbox:
https://github.com/meta-flutter/meta-flutter/blob/6a2f191dc57067cfe87b1a1505b622295bf66fe7/classes/flutter-app.inc#L79-L80

I use them in workspace-automation to sandbox the environment:
https://github.com/meta-flutter/workspace-automation/blob/079c2f52ed977d7fbbafebbd925628d1477980a3/flutter_workspace.py#L259-L269

aduskett, 2023-08-21:
Thank you, @jwinarske ! That's very helpful!

aduskett, 2023-08-21:
@jwinarske After setting the above, the same directories and files are made in ~/. Setting the HOME variable is the only way I have found so far to prevent the files and directories from every being put in ~/ in the first place.

jwinarske, 2023-08-21:
@aduskett Are you setting both variables to desired path?
```
export PUB_CACHE=<my sandbox pub cache>
export XDG_CONFIG_HOME=<my sandbox .config>
```
dart looks for PUB_CACHE, among others.
flutter tooling looks for XDG_CONFIG_HOME, before using HOME

With above environmental variable being set what are you seeing?

jwinarske, 2023-11-15:
Closing as there is no clear repro

closed 2023-11-15 by jwinarske