meta-flutter/meta-flutter #318: compilation error when flutter-engine >= 3.13

r3flow, 2023-08-29:
related issue :-(
https://github.com/flutter/flutter/issues/132765

JesvinJoseph, 2023-08-29:
I am also getting the below error with 3.13.1.

Just updated the engine version to 3.13.1 and run the bitbake command to build the image.

`| [6965/9221] ACTION //third_party/dart/utils/analysis_server:analysis_server_dill(//build/toolchain/custom:custom)
| FAILED: gen/third_party/dart/utils/analysis_server/analysis_server.dart.dill
| vpython3 ../../third_party/dart/build/gn_run_binary.py compiled_action ../../third_party/dart/tools/sdks/dart-sdk/bin/dart -Dsdk_hash=1b9c3b35ce --dfe=../../third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot gen/third_party/dart/utils/gen_kernel/bootstrap_gen_kernel.dill --packages=../../third_party/dart/.dart_tool/package_config.json --platform=vm_platform_strong.dill --no-aot --no-embed-sources --no-link-platform --output=gen/third_party/dart/utils/analysis_server/analysis_server.dart.dill --depfile=gen/third_party/dart/utils/analysis_server/analysis_server.dart.dill.d -Dsdk_hash=1b9c3b35ce -Ddart.vm.product=false ../../third_party/dart/pkg/analysis_server/bin/server.dart
| Command failed: /home/jesvin/yocto/build/tmp/work/aarch64-poky-linux/flutter-engine/git-r0/src/third_party/dart/tools/sdks/dart-sdk/bin/dart -Dsdk_hash=1b9c3b35ce --dfe=../../third_party/dart/tools/sdks/dart-sdk/bin/snapshots/kernel-service.dart.snapshot gen/third_party/dart/utils/gen_kernel/bootstrap_gen_kernel.dill --packages=../../third_party/dart/.dart_tool/package_config.json --platform=vm_platform_strong.dill --no-aot --no-embed-sources --no-link-platform --output=gen/third_party/dart/utils/analysis_server/analysis_server.dart.dill --depfile=gen/third_party/dart/utils/analysis_server/analysis_server.dart.dill.d -Dsdk_hash=1b9c3b35ce -Ddart.vm.product=false ../../third_party/dart/pkg/analysis_server/bin/server.dart
| output: ../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart:224:23: Error: The method 'send' isn't defined for the class 'Analytics'.
|  - 'Analytics' is from 'package:unified_analytics/src/analytics.dart' ('../../third_party/dart/third_party/pkg/tools/pkgs/unified_analytics/lib/src/analytics.dart').
| Try correcting the name to the name of an existing method, or defining a method named 'send'.
|       await analytics.send(Event.memoryInfo(
|                       ^^^^
| ../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart:224:28: Error: The getter 'Event' isn't defined for the class 'AnalyticsManager'.
|  - 'AnalyticsManager' is from 'package:analysis_server/src/analytics/analytics_manager.dart' ('../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart').
| Try correcting the name to the name of an existing getter, or defining a getter or field named 'Event'.
|       await analytics.send(Event.memoryInfo(
`

jwinarske, 2023-08-29:
Note that each engine version or ranges of versions usually requires a unique change.  It could be a patch, or updated DEPS inclusion, or changes to the build arguments.  The current state of this repo is to build for 3.10.6.  When you move forward it's is 90% likelihood a change will be required beyond changing the version number.

I'm in the middle of a change to better manage LTS.  So I am holding off moving to 3.13 until I have this work done.  I need to implement cipd fetch gn, and then it will switch to new scheme (pre-process gn to be Yocto compatible).

JesvinJoseph, 2023-08-30:
Thanks for the update. It would be great if you can release it quickly. I was able to build the meta-flutter with 3.13.1. Now my flutter application is working smoothly with Toradex board. But with 3.10.6, I was getting an error of "Check filed: disaply_.empty()".

jwinarske, 2023-08-30:
https://github.com/meta-flutter/meta-flutter/pull/323

closed 2023-08-30 by jwinarske

JesvinJoseph, 2023-08-31:
Thanks @jwinarske