meta-flutter/meta-flutter #327: Please get rid of SRCREV = "${AUTOREV}"

angelobt, 2023-09-01:
It's impossible to have a reproducible setup.

jwinarske, 2023-09-01:
@ardera

jwinarske, 2023-09-04:
@angelobt Yes I completely agree.  Only ivi-homescreen and flutter-auto are locked to a commit.

To lock for either flutter-pi or Sony you simply use a bbappend file to override the SRCREV.

ardera, 2023-09-05:
@jwinarske should we have some kind of auto-roller for flutter-pi then?

jwinarske, 2023-09-05:
@ardera The common pattern is to only move the commit when there is a logical release, but that is 100% relative to the upstream repo and it's release patterns.

I'm not against a nightly CI job to auto-roll deps.  Anything to cut down on the maintenance overhead of this layer, I'm all for it.

jwinarske, 2024-01-16:
https://github.com/meta-flutter/meta-flutter/pull/377