meta-flutter/meta-flutter #341: ivi-homescreen vs. flutter-engine - libflutter_engine.so install location and search paths

emanuel-th, 2023-10-27:
Hi, I'm building flutter-engine and ivi-homescreen for a im6 based board. Using kirkstone branch.

When trying to run homescreen I have to setup symlinks to libflutter_engine.so and icudtl.dat.
Similar as it is done in https://github.com/meta-flutter/meta-flutter/issues/146#issuecomment-1220911726

Homescreen looks for libflutter_engine.so under /usr/lib and in the bundle-directory.
The flutter-engine files are installed under 
`${D}${datadir}/flutter/${FLUTTER_SDK_VERSION}/${MODE}/lib/libflutter_engine.so`
see https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-engine/flutter-engine_git.bb#L165

The only chance homescreen will find it is to link it from the bundle directory?
If I switch between the flutter runtimes (MODE) I have to change the links each time?

jwinarske, 2023-10-27:
Depending on package config for flutter-engine it will install each specified runtime mode.  Default is to install all three engine variants.

For application install it will create a bundle folder.  The bundle folder is documented in a number of places.  Start with the ICU-homescreen readme.  Bundle creation for debug is handled by host side using a properly configured workspace automation config, via custom device config.  

The creation of a bundle folder should use symlinks.  This creates a single point of authority, and reduces disk space.

For a release build you would adjust flutter-engine package config for just release runtime.

Home screen will check local bundle folder for overrides first, of not present will look in system paths.

So you need a valid bundle folder structure.

The bundle folder model comes from the Linux GTK embedder.

jwinarske, 2023-10-27:
Debug bundle folder is created by the debug host.  See workspace automation configs in AGL.

Selecting release or profile depends on Yocto variables at build time.  Meaning you can have both installed side by side.  It's just a matter of picking a different directory for the bundle folder.

jwinarske, 2023-10-31:
Closing issue due to inactivity

closed 2023-10-31 by jwinarske