meta-flutter/meta-flutter #346: Applications fail to build after latest update

perezmeyer, 2023-11-06:
The latest AGL-repo update brought in:

meta-flutter ccacfb9640fd46312ad6184883dbd4a0ca9119ad
workspace-automation 767298cdbfeef43015d7041947bb4ad6e8ce1b02

Starting from that commit trying to build a Flutter app shields:

```
ERROR: flutter-ics-homescreen-git-r0 do_compile: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_compile(d)
     0003:
File: '/home/lisandro/AGL/master/external/meta-flutter/conf/include/flutter-app.inc', lineno: 178, function: do_compile
     0174:do_compile[network] = "1"
     0175:python do_compile() {
     0176:    import shutil
     0177:
 *** 0178:    pubspec_yaml = get_pubspec_yaml(d)
     0179:    pubspec_appname = d.getVar("PUBSPEC_APPNAME")
     0180:    if pubspec_appname != pubspec_yaml["name"]:
     0181:        bb.error("Set PUBSPEC_APPNAME to match name value in pubspec.yaml")
     0182:
File: '/home/lisandro/AGL/master/external/meta-flutter/conf/include/app-utils.inc', lineno: 33, function: get_pubspec_yaml
     0029:
     0030:
     0031:def get_pubspec_yaml(d):
     0032:    """ Returns python object of pubspec.yaml """
 *** 0033:    import yaml
     0034:
     0035:    filepath = get_pubspec_yaml_filepath(d)
     0036:
     0037:    data_loaded = None
Exception: ModuleNotFoundError: No module named 'yaml'

ERROR: Logfile of failure stored in: /home/lisandro/AGL/master/build-rpi4/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git-r0/temp/log.do_compile.121711
ERROR: Task (/home/lisandro/AGL/master/meta-agl-ics/recipes-demo/flutter-ics-homescreen/flutter-ics-homescreen_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2104 tasks of which 2094 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds

```

This seems to also be present in other builds by Scott Murray as per IRC.

jwinarske, 2023-11-06:
Does the build machine have pyyaml installed?

perezmeyer, 2023-11-06:
No.

jwinarske, 2023-11-15:
Closing since pyyaml dep was removed

closed 2023-11-15 by jwinarske