meta-flutter/meta-flutter #356: Help with the explanation of some dart arguments in flutter-app.inc

aduskett, 2023-11-19:
Hello, I'm the guy who is maintaining the flutter packages for Buildroot and I noticed I was missing a few dart arguments. I have made patches found here: https://patchwork.ozlabs.org/project/buildroot/list/?series=382808 and I had a few questions specifically about:

https://patchwork.ozlabs.org/project/buildroot/patch/20231118221414.1369987-3-adam.duskett@amarulasolutions.com/

I don't think my explanation of the dart arguments is correct, and the git log for flutter-app.inc is quite light. Specifically "-dart plugin registrant called in all cases with --obfuscate enabled"

I did some searching around the internet and source code of flutter-engine and I really don't understand why the following arguments are needed/what they actually do, and I am hoping you will be so kind as to share your knowledge with me, as it is clear that, at least for flutter-gallery, that the following dart arguments are mandatory:

```
f'--source file://{dart_plugin_registrant_file}'
'--source package:flutter/src/dart_plugin_registrant.dart'
f'-Dflutter.dart_plugin_registrant=file://{dart_plugin_registrant_file}'
```
If I try to build flutter-gallery with any of those three missing it fails to start.  Also, I couldn't find a good explanation as to what the native_assets.yaml file does/why some applications may need it to compile.

Any help would greatly be appreciated! Thank you so much!

Adam

jwinarske, 2023-11-20:
Hi Adam,

The dart_plugin_registrant.dart is generated during flutter build bundle.

Try searching flutter and dart SDK issues.  It illustrates the how the --source flag came into flutter.  There is nothing more I have found.

I raised a flutter issue where in some cases vm:entry does not work correctly.  It's still unresolved.  The work around I discovered was to enable obfuscated.

aduskett, 2023-11-24:
Thanks for the tips! I really appreciate it!

closed 2023-11-24 by aduskett