meta-flutter/meta-flutter #360: Random error in flutter-engine compile`: scene_flatbuffers.h file not found

kwavnet, 2023-11-22:
```
| In file included from ../../flutter/display_list/benchmarking/dl_complexity.cc:6:
| In file included from ../../flutter/display_list/benchmarking/dl_complexity_gl.h:8:
| In file included from ../../flutter/display_list/benchmarking/dl_complexity_helper.h:10:
| In file included from ../../flutter/display_list/dl_op_receiver.h:10:
| In file included from ../../flutter/display_list/dl_canvas.h:9:
| In file included from ../../flutter/display_list/dl_paint.h:12:
| In file included from ../../flutter/display_list/effects/dl_color_source.h:25:
| In file included from ../../flutter/impeller/scene/node.h:18:
| ../../flutter/impeller/scene/animation/animation.h:18:10: fatal error: 'impeller/scene/importer/scene_flatbuffers.h' file not found
| #include "impeller/scene/importer/scene_flatbuffers.h"
|          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| 1 error generated.
| [8078/9224] CXX clang_x64/obj/third_party/vulkan-deps/glslang/src/SPIRV/glslang_sources.GlslangToSpv.o
| [8079/9224] CXX obj/flutter/display_list/display_list.dl_builder.o
| FAILED: obj/flutter/display_list/display_list.dl_builder.o
```

We get this error randomly in flutter-engine compile.
Using poky kirkstone 4.0.14 (in before we never have seen that).
In most cases it works absolutely fine, but like in every 15-20th job this one errors out with above message.

Is that a know issue, like a race between fetching and building?
I admit I'm a bit lost where to look first here

jwinarske, 2023-11-23:
@kwavnet 
It might be a bug in the engine build itself.  If that file is being generated, it very well could be a race condition.

You try removing `impeller-3d` from the package config variable.  Then see if the problem goes away.  If that addresses it I'll remove it in the kirkstone branch.

kwavnet, 2023-11-23:
@jwinarske thanks, will try that

kwavnet, 2023-11-24:
@jwinarske it seems like it did the trick - builds are stable now with `impeller-3d`

jwinarske, 2023-11-24:
@kwavnet with our without?  The default is currently enabled by default

kwavnet, 2023-11-25:
> @kwavnet with our without? The default is currently enabled by default

without ofc, sorry

jwinarske, 2023-11-29:
Response from Brandon (Impeller 3D dev):
```
Some of the DL test targets don't directly depend on scene, so if one of those happen to be getting built before the impeller targets with the flag it would probably cause this issue.
we should need to add a more direct dependency for the header gen to the display list target when the flag is enabled: https://github.com/flutter/engine/blob/main/display_list/BUILD.gn#L87
gn parses header includes and tries to match them up with targets, so not sure why this isn't getting caught during the gn tool run
```

jwinarske, 2023-11-29:
The short term work around if impeller-3d is needed is to patch out unit tests from building.  If I have time in the next month, I'll come up with a patch

jwinarske, 2023-11-29:
See if this fixes it:
https://github.com/meta-flutter/meta-flutter/pull/366

If yes I'll click the merge button

closed 2024-01-04 by jwinarske