meta-flutter/meta-flutter #361: Please check for build errors.

Teddywonseokyoo, 2023-11-24:
I'm trying to build it with flutter-engine.
The latest #359 has been checked out.

I get an error like the following image:
[image]

Please help.

DarthLowen, 2023-11-26:
You're likely compiling the `kirstone` branch in an older yocto? You could just try removing references to the `debug2` method, or switch to a more recent yocto version (like `kirkstone`)

jwinarske, 2023-11-28:
@Teddywonseokyoo what yocto release are you working with?

Teddywonseokyoo, 2023-11-28:
Hi.
I already know that my yocto version is dufell.
However, if meta-flutter is checked out with dunfell, an error like #362 occurs.
Checkeout with d92cd07 and the build was successful. thank you.

jwinarske, 2023-12-01:
@Teddywonseokyoo okay you are unblocked, great.  After my current CES work I will get dunfell updated.

jwinarske, 2024-01-20:
https://github.com/meta-flutter/meta-flutter/pull/386

closed 2024-01-20 by jwinarske