meta-flutter/meta-flutter #362: Dunfell branch broken?

DarthLowen, 2023-11-25:
Hi,

apologies for filing this as a bug, I'm sure this is just me overlooking something, but I can't seem to be able to compile `flutter-engine` on the `dunfell` branch (I'm up-to-date, I checked).

I was able to compile and run flutter just fine on the `kirkstone` branch a few months back (which was at version `3.10.6` back then), however because I'm running into an issue in the device-trees/drivers on NXP's `kirkstone` branch, I had to fall back to `dunfell`.

The problem is very similar to some other issues that have already been reported, I'm getting a lot of these:
```
| ../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart:553:30: Error: The getter 'Event' isn't defined for the class 'AnalyticsManager'.
|  - 'AnalyticsManager' is from 'package:analysis_server/src/analytics/analytics_manager.dart' ('../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart').
| Try correcting the name to the name of an existing getter, or defining a getter or field named 'Event'.
|         await analytics.send(Event.clientRequest(
|                              ^^^^^
| ../../third_party/dart/pkg/analysis_server/lib/src/analytics/analytics_manager.dart:570:29: Error: The method 'send' isn't defined for the class 'Analytics'.
|  - 'Analytics' is from 'package:unified_analytics/src/analytics.dart' ('../../third_party/dart/third_party/pkg/tools/pkgs/unified_analytics/lib/src/analytics.dart').
| Try correcting the name to the name of an existing method, or defining a method named 'send'.
|             await analytics.send(Event.commandExecuted(
|                             ^^^^
```

I've tried to remove the `GN_CUSTOM_DEPS` variable as suggested in #319, but to no avail. Also, in #323 this is included (`kirkstone`), but not in #324 (`dunfell`)?

When I adjust the `FLUTTER_SDK_TAG` version to `3.10.6`, it compiles just fine, but then the `flutter-gallery` app doesn't want to compile:

```
Loading cache: 100% |################################################################################################################################################################################################################################################################################################################################################| Time: 0:00:02
Loaded 3909 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################################################################################################################################################################################################################################################################| Time: 0:00:01
Parsing of 2751 .bb files complete (2744 cached, 7 parsed). 3916 targets, 242 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-tdx-linux"
MACHINE              = "colibri-imx8x"
DISTRO               = "tdx-xwayland"
DISTRO_VERSION       = "5.7.3-devel-20231125084948+build.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta-toradex-nxp     = "HEAD:8797f291ca2d9c8e0b2efedf37bf3bd10709ee81"
meta-freescale       = "HEAD:f66fdef1c2dc735690a7ec7dd44a534d69ea3918"
meta-freescale-3rdparty = "HEAD:4c0a166fa2809f74c93bd4c56d9a0f7ccb7818c8"
nic                  = "master:579481efc8a92e22cb1ea33947b780a36c699c74"
meta-flutter         = "dunfell:1d2d1641a01d924f0bf4a4a25ada27b4f44fc360"
meta-toradex-bsp-common = "HEAD:ea047be150a35d5c9be5617f132c9c0dbf7c249b"
meta-oe              
meta-filesystems     
meta-gnome           
meta-xfce            
meta-networking      
meta-multimedia      
meta-python          = "dunfell:9135c7ea7350d5d241f4afc3b28087122ebe2d19"
meta-freescale-distro = "HEAD:5d882cdf079b3bde0bd9869ce3ca3db411acbf3b"
meta-toradex-demos   = "HEAD:8ffadb517be04a0b19ef1f5d5299a8c581f87383"
meta-qt5             = "HEAD:5ef3a0ffd3324937252790266e2b2e64d33ef34f"
meta-toradex-distro  = "HEAD:53f241006b8cf69c31c22ce27e837261156ddae8"
meta-poky            = "dunfell:455398490462728c44b9951af96c1aa61a5679e2"
meta                 = "HEAD:b3fc8ef9aba822b3d485242c8ebd0e0bff0ebfc8"

Initialising tasks: 100% |###########################################################################################################################################################################################################################################################################################################################################| Time: 0:00:14
Sstate summary: Wanted 716 Found 676 Missed 40 Current 1659 (94% match, 98% complete)
NOTE: Executing Tasks
ERROR: flutter-gallery-git-r0 do_compile: Execution of '/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/temp/run.do_compile.411088' failed with exit code 254
ERROR: Logfile of failure stored in: /home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/temp/log.do_compile.411088
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: sharedstatedir=/com base_libdir=/lib PKG_CONFIG_DIR=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig PKG_CONFIG_PATH=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot/usr/share/pkgconfig:/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot/usr/share/pkgconfig docdir=/usr/share/doc nonarch_base_libdir=/lib oldincludedir=/usr/include prefix=/usr PYTHONHASHSEED=0 BUILD_AS=as BUILD_AR=ar BUILD_LDFLAGS=-L/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -L/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,--enable-new-dtags -Wl,-rpath-link,/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/nic/Mustang/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 base_prefix= STRINGS=aarch64-tdx-linux-strings CPP=aarch64-tdx-linux-gcc -E --sysroot=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security BUILD_CXX=g++ BUILD_NM=nm datadir=/usr/share nonarch_libdir=/usr/lib PWD=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/git SOURCE_DATE_EPOCH=1678357822 LOGNAME=nic CXX=aarch64-tdx-linux-g++ -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot CXXFLAGS= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0=/usr/src/debug/flutter-gallery/git-r0 -fdebug-prefix-map=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot= -fdebug-prefix-map=/home/nic/Mustang/yocto/build/tmp/work/aarch64-tdx-linux/flutter-gallery/git-r0/recipe-sysroot-native= -fvisibility-inlines-hidden BUILD_CC=gcc TZ=UTC libexecdir=/usr/libexec BUILD_FC=gfortran includedir=/usr/include LDFLAGS=-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now HOME=/home/nic systemd_system_unitdir=/lib/systemd/system libdir=/usr/lib base_sbindir=/sbin CCLD=aarch64-tdx-linux-gcc -fstack-pro [... cut ...]
Summary: 1 task failed:
  /home/nic/Mustang/yocto/build/../layers/meta-flutter/recipes-graphics/flutter-apps/flutter-gallery_git.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

I'm on Ubuntu 22.04. I cleared all SDK paths, cleared all native flutter stuff (hidden folders in my home folder, uninstalled flutter on Ubuntu), even completely removed my yocto and started from a fresh checkout...always the same.

Would really appreciate some help! Thanks!

Cheers
Nic.

DarthLowen, 2023-11-26:
As a follow up, I've now successfully compiled the `kirkstone` branch in my `dunfell` yocto. All I had to do was remove the `recipes-devtools/flutter_rust_bridge` and `recipes-devtools/rust` and remove the references to `debug2` in `lib/cipd.py` and `lib/gn.py` (like #361 is experiencing) After that it compiles and runs like a charm!

I've cloned your repo and made the changes there, would be great if this could go into your mainline repo!

Excellent work by the way, I'm a fan!

Cheers,
Nic.

jwinarske, 2023-11-28:
@DarthLowen Hi there.  Yeah it's on my list to update dunfell.  Effectively the process is similar to what you did to get it to work :)

jwinarske, 2024-01-20:
https://github.com/meta-flutter/meta-flutter/pull/386

closed 2024-01-20 by jwinarske