meta-flutter/meta-flutter #363: 'Illegal instruction' error if I run the flutter gallery app on the toradex imx6 board.

JesvinJoseph, 2023-11-27:
Hi,

I am trying to run the flutter-gallery app on the tordex board. It was working fine before but with the latest meta-flutter I am getting the 'illegal instruction' error. 

[image]

weston-terminal also works fine. 

Am I missing any configuration?

jwinarske, 2023-11-29:
Potentially a repro of this: https://github.com/meta-flutter/meta-flutter/issues/204

Two experiments to try:
1. Try building and running with jit_release engine variant.
1. Next try setting APP_AOT_EXTRA in your app recipe to `--no-use-integer-division`

I would do these independent of each other, then together.  So three total attempts.

JesvinJoseph, 2023-11-29:
@jwinarske Thank you for the quick suggestion. I will try them and will let you know the result.

JesvinJoseph, 2023-12-01:
Hi @jwinarske ,

I was trying the first option. Below are the changes I made to use the jit_release but I am getting build error. Am I on the correct way?

1. In the flutter-engine.bbappend file I added `PACKAGECONFIG:append = " jit_release"`
2. On my app, I added `FLUTTER_APP_RUNTIME_MODES = "jit_release"`

The build error is 

> The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_compile(d)
     0003:
File: '/home/jesvin/kirkstone/build/../meta-flutter/conf/include/flutter-app.inc', lineno: 225, function: do_compile
     0221:        run_command(d, 'flutter clean', source_root, env)
     0222:
     0223:        if runtime_mode == 'jit_release':
     0224:            cmd = (f'flutter build {flutter_build_args} --local-engine')
 *** 0225:            run_command(cmd, source_root, env)
     0226:        else:
     0227:            cmd = f'flutter build {flutter_build_args}'
     0228:            run_command(d, cmd, source_root, env)
     0229:
Exception: TypeError: run_command() missing 1 required positional argument: 'env'

jwinarske, 2023-12-01:
@JesvinJoseph I'll take a look, looks like a regression

JesvinJoseph, 2023-12-01:
@jwinarske ,

Thanks,

Also for the second option, I got error while building with `APP_AOT_EXTRA `. It says there is no argument like `--no-use-integer-division`. But instead I tried with `APP_GEN_SNAPSHOT_FLAGS="--no-use-integer-division"`, then it worked without jit_release.

jwinarske, 2023-12-04:
@JesvinJoseph that's great.  Thanks for the update.

closed 2023-12-04 by jwinarske