meta-flutter/meta-flutter #364: How to cross compile the flutter app in Ubuntu using Yocto SDK?

JesvinJoseph, 2023-11-28:
Hi,

Can you explain how to cross compile the flutter app on Ubuntu 22 for the arm processor? I couldn't see the option to mention processor architecture in the flutter build command. May be I might have missed it. Do you have any document link?

jwinarske, 2023-11-29:
@JesvinJoseph There is a script here that will cross compile an AOT on a host machine:
https://github.com/meta-flutter/workspace-automation/blob/main/create_aot.py

I ported it from the meta-flutter flutter app compilation step.  So it does the same thing.  It does requires an artifact from the Yocto build which I call the `Engine SDK`.  It's a select set of engine build artifacts that are required for host side AOT generation.

jwinarske, 2023-11-29:
So you would want to add the package `flutter-engine-sdk-dev` to your OS image.  Then you should find it at /usr/share/flutter/engine_sdk.zip.  Easy to SCP from host.

JesvinJoseph, 2023-11-29:
@jwinarske  Thank you for the response. I will give a try with the script and will let you soon

jwinarske, 2023-12-04:
@JesvinJoseph it looks like you don't have your environment setup:

    /flutter/bin/cache/dart-sdk/bin/dart: not found

If you have a flutter workspace setup with workspace-automation, and setup the environment it will just work with engine SDK path set

JesvinJoseph, 2023-12-05:
Hi @jwinarske ,

I run the `./flutter_workspace.py ` and it run successfully. Initially there were one error regarding `vulkan-utils`, I fixed it by changing it to `vulkan-tools`, that is the package for ubuntu 22. To get the dart sdk, is there any argument I need to pass to the script?

jwinarske, 2023-12-11:
@JesvinJoseph this package is required:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-engine/flutter-engine_git.bb#L216C20-L216C20

I had someone new use the process last week without any issues, both workspace automation and create_aot.py.  Only difference is it was on Ubuntu 20.  In that case the only difference would be package name changes; which you found.

jwinarske, 2023-12-11:
cd <flutter workspace>
source ./setup_env.sh
export GEN_SNAPSHOT=<absolute path to gen_snapshot located in the engine_sdk.zip extract>
create_aot.py <path to root of app to create AOT>

If you need to include dart defines then export the appropriate variable; reference the os variables used in the script.

You should not have to modify the script.

JesvinJoseph, 2023-12-14:
Hi @jwinarske,

I tried the same. Still I am getting the below error. Is there any version missmatch in the gen_snapshot?

```
***************************************
** kernel_snapshot_release: Complete **
***************************************
******************************
** aot_elf_release: Started **
******************************
***************
** linux_arm **
***************
Running [/home/jesvin/sdk/sdk/clang_x64/gen_snapshot --strip --obfuscate --deterministic /home/jesvin/build/sampleapp/.dart_tool/flutter_build/*/app.dill] in /home/jesvin/build/sampleapp
failed 255 (cmd was /home/jesvin/sdk/sdk/clang_x64/gen_snapshot --strip --obfuscate --deterministic /home/jesvin/build/sampleapp/.dart_tool/flutter_build/*/app.dill):
Building a core snapshot requires specifying output files for --vm_snapshot_data and --isolate_snapshot_data.

Usage: gen_snapshot [<vm-flags>] [<options>] <dart-kernel-file>             
                                                                            
Common options:                                                             
--help                                                                      
  Display this message (add --verbose for information about all VM options).
--version                                                                   
  Print the SDK version.                                                    
                                                                            
To create a core snapshot:                                                  
--snapshot_kind=core                                                        
--vm_snapshot_data=<output-file>                                            
--isolate_snapshot_data=<output-file>                                       
<dart-kernel-file>                                                          
                                                                            
To create an AOT application snapshot as assembly suitable for compilation  
as a static or dynamic library:                                             
--snapshot_kind=app-aot-assembly                                            
--assembly=<output-file>                                                    
[--strip]                                                                   
[--obfuscate]                                                               
[--save-debugging-info=<debug-filename>]                                    
[--save-obfuscation-map=<map-filename>]                                     
<dart-kernel-file>                                                          
                                                                            
To create an AOT application snapshot as an ELF shared library:             
--snapshot_kind=app-aot-elf                                                 
--elf=<output-file>                                                         
[--strip]                                                                   
[--obfuscate]                                                               
[--save-debugging-info=<debug-filename>]                                    
[--save-obfuscation-map=<map-filename>]                                     
<dart-kernel-file>                                                          
                                                                            
AOT snapshots can be obfuscated: that is all identifiers will be renamed    
during compilation. This mode is enabled with --obfuscate flag. Mapping     
between original and obfuscated names can be serialized as a JSON array     
using --save-obfuscation-map=<filename> option. See dartbug.com/30524       
for implementation details and limitations of the obfuscation pass.  
```

JesvinJoseph, 2023-12-14:
@jwinarske,

I was still getting the same error.  Since it was showing an error with the arguments of gen_snapshot, I took a new build and copied the engine_sdk. Now it is working. I am able to run the cross compiled app. Thanks for the help.

closed 2023-12-14 by JesvinJoseph