meta-flutter/meta-flutter #368: Error during build

prakaashg, 2023-12-08:
Hi ,

I get this error when built in openstlinux, meta-flutter kirkstone branch
gn_fetcher_bbclass
     0003:# SPDX-License-Identifier: MIT
     0004:#
     0005:
     0006:python () {
 *** 0007:    import gn
     0008:    bb.fetch2.methods.append(gn.GN())
     0009:}
     0010:
     0011:DEPENDS += " \
Exception: ModuleNotFoundError: No module named 'gn'

Please help

prakaashg, 2023-12-09:
> @prakaashg <https://github.com/prakaashg> can you build an OS image
> before adding the meta-flutter layer? Please confirm

Yes I could create the image i added the layer after that it fails

jwinarske, 2023-12-09:
Have you reviewed the container runtime packages used in this CI job?
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/.github/workflows/kirkstone-stm32mp15.yml
https://github.com/meta-flutter/docker-containers/blob/main/ubuntu-18-dev/Dockerfile

prakaashg, 2023-12-11:
I copied from workflow yml now,  I get a different error this time,  

 tools/clang/lib/CodeGen/CMakeFiles/obj.clangCodeGen.dir/CGBuiltin.cpp.o -c /home/rsa-key-20231118/kirkstone/build-openstlinuxweston-stm32mp1/tmp-glibc/work-shared/llvm-project-source-14.0.6-r0/git/clang/lib/CodeGen/CGBuiltin.cpp
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.

Summary: 1 task failed:
  virtual:native:/home/rsa-key-20231118/kirkstone/layers/layers/meta-clang/recipes-devtools/clang/clang_git.bb:do_compile
Summary: There was 1 ERROR message, returning a non-zero exit code.

jwinarske, 2023-12-12:
What distro/version are you building with?

prakaashg, 2023-12-18:
DISTRO=openstlinux-weston MACHINE=stm32mp1

prakaashg, 2023-12-21:
Hi, 

Looks the dart sdk build fails, it tries to download this commit (3470ab8451567dad237ddd54f992bfed066a2707) and it fails, Looks similar to this https://github.com/dart-lang/sdk/issues/54279 , Any idea please help

________ running 'git -c core.deltaBaseCacheLimit=2g fetch origin 3470ab8451567dad237ddd54f992bfed066a2707 --verbose --no-tags' in '/home/rsa-key-20231118/dl/gn/https__github.com_dart-lang_sdk.git/sdk/third_party/pkg/dart_style'

[0:02:41] POST git-upload-pack (1000 bytes)
[0:02:41] error: RPC failed; HTTP 400 curl 22 The requested URL returned error: 400
[0:02:41] fatal: the remote end hung up unexpectedly

prakaashg, 2024-01-08:
Hi,
After changing the dart version the build succeeds, but running flutter auto results in segentation fault,
sr/share/flutter/gallery/3.13.9
[14:38:33.182080] [I] agl @ 3114782
[14:38:33.182512] [I] **********
[14:38:33.182544] [I] * Global *
[14:38:33.182568] [I] **********
[14:38:33.182632] [I] Application Id: .......... flutter-auto
[14:38:33.182670] [I] Disable Cursor: .......... false
[14:38:33.182700] [I] Debug Backend: ........... false
[14:38:33.182723] [I] ********
[14:38:33.182743] [I] * View *
[14:38:33.182762] [I] ********
[14:38:33.182785] [I] Bundle Path: .............. /usr/share/flutter/gallery/3.13.9
[14:38:33.182810] [I] Window Type: .............. NORMAL
[14:38:33.182867] [I] Output Index: ............. 0
[14:38:33.182898] [I] Size: ..................... 1920 x 720
[14:38:33.182926] [I] Fullscreen: ............... false
[14:38:33.182970] [I] Accessibility Features: ... 0
Segmentation fault (core dumped)

Any idea?

jwinarske, 2024-01-08:
@prakaashg What hardware, and BSP?

jwinarske, 2024-01-09:
The other thing is there are a number of patches for the AGL client Marius recently made.  If you're not using the AGL compositor please use ivi-homescreen (upstream to flutter auto).  I will be releasing a major point release ~first week of February

prakaashg, 2024-01-11:
Hi ,
Board is STM32MP157c-Dk2, It was working fine with dunfell when we upgraded dart to 3.x and kirkstone I get this error.

When i enable weston logs we get this error

[1467645.098] wl_registry@10.global(17, "zwp_linux_explicit_synchronization_v1", 2)
[1467979.630]  -> wl_registry@10.bind(17, "zwp_linux_explicit_synchronization_v1", 1, new id [unknown]@13)
[1467979.781] wl_registry@10.global(18, "weston_content_protection", 1)
[1467979.874] wl_registry@10.global(19, "wl_output", 3)
[1467979.963] wl_registry@10.global(20, "zwp_input_panel_v1", 1)
[1467980.051] wl_registry@10.global(21, "zwp_input_method_v1", 1)
[1467980.141] wl_registry@10.global(22, "zwp_text_input_manager_v1", 1)
[1467980.230] wl_registry@10.global(23, "xdg_wm_base", 3)
[1467980.318] wl_registry@10.global(24, "wl_shell", 1)
[1467980.406] wl_registry@10.global(25, "weston_desktop_shell", 1)
[1467980.497] wl_registry@10.global(26, "weston_screenshooter", 1)
[1467980.587] wl_callback@11.done(75)
[1468014.600]  -> wl_compositor@3.create_surface(new id wl_surface@11)
[1468014.742]  -> wl_surface@11.frame(new id wl_callback@14)

Could this be related to https://github.com/meta-flutter/meta-flutter/issues/295 ?

jwinarske, 2024-01-11:
I see.  It could, try applying the patch I have for NXP.  I will be updating the dunfell branch after I get back from CES, so expect some changes there

jwinarske, 2024-01-20:
dunfell branch has been updated:
https://github.com/meta-flutter/meta-flutter/pull/386

This is the patch I was referring to.  It applies to both ivi-homescreen and flutter-auto (only different today is the executable name).
https://gerrit.automotivelinux.org/gerrit/gitweb?p=AGL/meta-agl-devel.git;a=blob;f=meta-agl-flutter/recipes-graphics/toyota/files/0001-Disable-on_frame_base_surface-wl_surface_commit.patch;h=a37233b50917f49ba561e6994c746d0a6373f4e5;hb=HEAD

jwinarske, 2024-02-12:
closing since I have not gotten a response back.  Feel free to open a new issue, or re-open this issue as needed.

closed 2024-02-12 by jwinarske