meta-flutter/meta-flutter #370: libstd-rs_%.bbappend caused libstd-rs build fail

onlyxuyang, 2023-12-20:
libstd-rs_%.bbappend change S to a strange folder in RUSTSRC, and it will caused build fail.

Why we need this bbappend?

jwinarske, 2023-12-21:
@onlyxuyang the history on that is located here:
https://github.com/meta-rust/meta-rust/issues/266

onlyxuyang, 2023-12-22:
@jwinarske 
> @onlyxuyang the history on that is located here: [meta-rust/meta-rust#266](https://github.com/meta-rust/meta-rust/issues/266)

Thank you for your information. I use meta-rust [f6a80656953678f17de7d41343107112d99492d9](https://github.com/meta-rust/meta-rust/commit/f6a80656953678f17de7d41343107112d99492d9)  with rust version 1.73.0.

Yes, this bbappend not compatible with latest rust in meta-rust.

And though no libproc_macro-xx.rlib under ${WORKDIR}/recipe-sysroot,  
but libproc_macro2-xx.rlib is under ${WORKDIR}/build/target/release/deps/  and my package is build pass.

Could we verify this bbappend is still necessary or apply it only excepted version?

Thank you

jwinarske, 2023-12-22:
@onlyxuyang yes that makes sense.  I will make it version specific.  Thanks for filling the issue

svarmeddev, 2024-01-03:
I have same issue

kusstas, 2024-01-03:
I have same issue

jwinarske, 2024-01-04:
I'll push a change in the next couple of days

jwinarske, 2024-01-04:
https://github.com/meta-flutter/meta-flutter/pull/371

closed 2024-01-04 by jwinarske