meta-flutter/meta-flutter #375: Readme should explain current variables for building flutter apps (recommends outdated FLUTTER_EXTRA_BUILD_ARGS ...)

Markus43, 2024-01-11:
I added a dart-define to my project today, and it took me some time to understand that the "Recommended development flow" from the project's readme is outdated: It recommends FLUTTER_EXTRA_BUILD_ARGS, and that variable doesn't exist anymore.

To use dart-defines, one currently has to modify FLUTTER_BUILD_ARGS and APP_AOT_EXTRA as in the vgv-super-dash recipe:
FLUTTER_BUILD_ARGS = "bundle --target lib/main_prod.dart --dart-define FLUTTER_APP_FLAVOR=prod"
APP_AOT_EXTRA = "-DFLUTTER_APP_FLAVOR=prod"

What might also be interesting is a short summary of FLUTTER_APP_RUNTIME_MODES . From the README and from the code I could not fully understand what values are available here. Since you are querying "what flutter-engine installed": Is it possible to change "what flutter-engine installed" somehow? Does the engine configuration need additional attention when not only building release builds?

jwinarske, 2024-01-12:
@Markus43 I'll work on the docs soon.  I agree they need some attention.  Once I'm through my major feature update with ivi-homescreen.

There is a dependency chain between engine variants and app build/install.  Engine gets built first based on package config flags.  The default runtime config is this line:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-graphics/flutter-engine/flutter-engine_git.bb#L63

Which means the engine is currently built three different ways.  This can be overloaded in the top level local.conf.

For app install I only install release builds.  Profile is a rare exception, and debug should originate from the host via debug connection (using custom devices).

There are flags to add installation of app profile and debug, if you so wish to.  In the case of multiple apps image size can rapidly increase with multiple variant installs.  So I take the conservative approach.

jwinarske, 2024-01-12:
As for example usage the best approach is to review the apps I have included.  There are a variety of use cases covered.

Markus43, 2024-01-15:
Thanks for your reply!

> There are flags to add installation of app profile and debug, if you so wish to. In the case of multiple apps image size can rapidly increase with multiple variant installs. So I take the conservative approach.

And I agree that's the best solution for a default here.

> As for example usage the best approach today is to review the apps I have included. There are a variety of use cases covered.

One things that might also be interesting from a user perspective: using build runner before the actual build. I believe that the melos example is the way to go here, but I still have to try it out.

By the way: I found out that packagegroup-flutter-test-apps.bb doesn't build anymore (the flutter-test-texture-egl rename to toyota-test-texture-egl is missing there) - I used that in my getting-started-image because I thought these were all the example apps that also run through your CI. Given that this is part of the github CI files, does this mean that the current tip of the tree did not go through your CI? Please don't understand me wrong: I don't mean to complain, I just want to understand how you organize the work on this layer and how you're using the CI here.

And just to add: I tried out homescreen last week, the first time since a few months. I was really impressed how flawlessly it works now! I'm on an i.MX8MP with etnativ drivers, these showed the same flickering as reported in some other issues here some months ago. Now I have e.g. flutter-gallery and an app of mine running perfectly smooth without any flickering or errors. Really nice, thanks and keep up the good work!

jwinarske, 2024-02-21:
https://github.com/meta-flutter/meta-flutter/pull/417

closed 2024-02-21 by jwinarske