meta-flutter/meta-flutter #384: Various build failures, starting with 'Missing "curl" tool'

Markus43, 2024-01-19:
I was trying to build the current tip of the tree ( d2f7dbcf70246cafb9baedcb5d4083b3dba4bc6e ) yesterday to try your latest updates. Before that, I was using 7ec6b53c62f9b55b187708fbaa80a9bb9ca4b2cc (one day earlier) which was working fine for me, and which still builds fine on the same build host. I'm building for an i.MX8MP based on https://github.com/dh-electronics/meta-dhsom-imx-bsp - not sure if that matters.

I now see various failures in the build, which seem to be not fully reproducible. What is reproducible for me is that failures start with this:

> ERROR: flutter-sdk-native-git-r0 do_unpack: failed 1 (cmd was flutter config --clear-features):
> Missing "curl" tool. Unable to download Dart SDK.
> Consider running "sudo apt-get install curl".

This error repeats for all run_command calls from here: https://github.com/meta-flutter/meta-flutter/blob/d2f7dbcf70246cafb9baedcb5d4083b3dba4bc6e/recipes-graphics/flutter-sdk/flutter-sdk_git.bb#L109

FWIW, I have curl installed on my build host.

What feels weird is that these failures do not cancel the yocto build, and when re-running the build, this part of the failure does not occur anymore (but the follow-up's do).

Possible follow-up errors:

> ERROR: flutter-gallery-git-r0 do_compile: failed 1 (cmd was flutter pub get --offline):
> Downloading Linux x64 Dart SDK from Flutter engine 4a585b79294e830fa89c24924d58a27cc8fbf406...
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: storage.googleapis.com
> Warning: Problem : timeout. Will retry in 1 seconds. 3 retries left.
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: storage.googleapis.com
> Warning: Problem : timeout. Will retry in 2 seconds. 2 retries left.
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: storage.googleapis.com
> Warning: Problem : timeout. Will retry in 4 seconds. 1 retries left.
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: storage.googleapis.com
> 

or this here:

> ERROR: flutter-gallery-git-r0 do_compile: failed 127 (cmd was /home/markus/projects/build_316_updated/build/tmp/work/cortexa53-crypto-dhlinux-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/dart-sdk/bin/dart --disable-analytics --disable-dart-dev /home/markus/projects/build_316_updated/build/tmp/work/cortexa53-crypto-dhlinux-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/linux-x64/frontend_server.dart.snapshot --sdk-root /home/markus/projects/build_316_updated/build/tmp/work/cortexa53-crypto-dhlinux-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ --target=flutter --no-print-incremental-dependencies -Ddart.vm.profile=false -Ddart.vm.product=true --aot --tfa --target-os linux --packages .dart_tool/package_config.json --output-dill .dart_tool/flutter_build/*/app.dill --depfile .dart_tool/flutter_build/*/kernel_snapshot.d --verbosity=error package:gallery/main.dart):
> /bin/sh: 1: /home/markus/projects/build_316_updated/build/tmp/work/cortexa53-crypto-dhlinux-linux/flutter-gallery/git-r0/recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/dart-sdk/bin/dart: not found

... or others, which I don't have in my current build output.

The follow-up-errors differ on each new build, and I believe that I even need a fresh SSTATE_CACHE to get the original first error back.

jwinarske, 2024-01-19:
@Markus43 Hey thanks for filing the issue.  I'm in the process of catching up on my meta-flutter todo list given CES dev work is over (~2 months).

flutter-sdk has a build time dependency on curl-native.  So it shouldn't be using the system's curl.

There have been some odd issues AGL+Toyota+myself have seen when updating versions, and no one has captured the exact set of variables yet.  I think I have a solution though.  It involves adding the flutter sdk version to the DL_DIR filenames, as of right now that's not present.  This change would re-trigger a pub-cache fetch, as of recently I just added `bitbake flutter-gallery -cdo_cleanall` which will delete the pub-cache archive, as it has stale SDK references.

What OS and version are you building with, and are you using either one?

    INHERIT:remove = "uninative"
    INHERIT += "rm_work"

Markus43, 2024-01-19:
> @Markus43 Hey thanks for filing the issue. I'm in the process of catching up on my meta-flutter todo list given CES dev work is over (~2 months).

And thanks for your quick reply!

> What OS and version are you building with, and are you using either one?
> 
> ```
> INHERIT:remove = "uninative"
> INHERIT += "rm_work"
> ```
I'm on Ubuntu 20.04. I have _INHERIT += "rm_work"_ in my local.conf - do you think that might cause the issue, and removing it might fix it? A colleague added it and commented that it is there to preserve disk space ...

jwinarske, 2024-01-19:
@Markus43 I would try commenting out `INHERIT += "rm_work"` in your local.conf, -cdo_cleanall for flutter-sdk-native, flutter-engine, and related apps.  Then pull my latest commit on kirkstone and rebuild your app.  If that works, then I can take a closer look at any possible interaction.

Between AGL and Toyota we have a large matrix of targeted platforms.  The commits in production are currently lagging behind meta-flutter, and when a commits get merged is when the matrix build `flush-out` happens.  If I have time I will re-enable my CI on gitlab.freedesktop.org as gitlab works flawless with podman rootless.  I have strong security concerns running anything non-rootless with github runners.

jwinarske, 2024-01-20:
@Markus43 any update?

Markus43, 2024-01-20:
> @Markus43 I would try commenting out `INHERIT += "rm_work"` in your local.conf, -cdo_cleanall for flutter-sdk-native, flutter-engine, and related apps. Then pull my latest commit on kirkstone and rebuild your app. If that works, then I can take a closer look at any possible interaction.

I followed your steps, and the build then finished without errors :-)

jwinarske, 2024-01-20:
Maybe now add `INHERIT += "rm_work"` back in and rebuild your image.  If that works you should be good to go.

Markus43, 2024-01-22:
> Maybe now add `INHERIT += "rm_work"` back in and rebuild your image. If that works you should be good to go.

I started a full rebuild (still without `INHERIT += "rm_work"`) and get a new error:

> ERROR: No recipes in default available for:
>   /home/markus/projects/build_316_latest/build/../layers/meta-flutter/dynamic-layers/clang-layer/recipes-graphics/toyota/ivi-homescreen-next_%.bbappend

jwinarske, 2024-01-22:
You can delete it for now.  I will remove from the repo

jwinarske, 2024-01-31:
https://github.com/meta-flutter/meta-flutter/pull/393

jwinarske, 2024-01-31:
Are we good, can we close this issue?

Markus43, 2024-01-31:
> Are we good, can we close this issue?

I'll do a few builds with today's tip of the tree and report back within a few days. 

> Maybe now add `INHERIT += "rm_work"` back in and rebuild your image. If that works you should be good to go.

So rm_work is  supposed to fully work now with meta-flutter?

Markus43, 2024-01-31:
> > Maybe now add `INHERIT += "rm_work"` back in and rebuild your image. If that works you should be good to go.
> 
> So rm_work is supposed to fully work now with meta-flutter?

It doesn't. I still see the curl error when building an image with `INHERIT += "rm_work"`  :

> ERROR: flutter-sdk-native-git-r0 do_unpack: failed 1 (cmd was flutter config --clear-features):
> 
> Missing "curl" tool. Unable to download Dart SDK.
> Consider running "sudo apt-get install curl".

This is on f4f58f952482793a3ddb1ea3685733e48da45b8a , building with a fresh DL_DIR and SSTATE_CACHE.

jwinarske, 2024-02-03:
If flutter-sdk version changed after recent pull and you see build errors.  Follow this sequence:
```
bitbake flutter-sdk-native -cdo_cleanall
bitbake flutter-engine -cdo_cleanall
bitbake <your app>
```

perezmeyer, 2024-02-04:
I have tried this with Ubuntu 20.04, 22.04 and Debian-12, same error. I am now trying without rm_work.

jwinarske, 2024-02-05:
In case of rm_work I suspect flutter-engine-sdk-dev needs to be added to do_compile as a dependency in flutter-app.inc.

I've been building in an AGL tree since CES.  No issues for me on Fedora (selinux, etc) without a container.  Followed the pre-requisites from Yocto project site.

Tested dunfell last week without issue using a container on redhat running rootless podman.  rm_work not set.

@perezmeyer Are either of your build scenarios using a container, and are you using rm_work too?

jwinarske, 2024-02-05:
What recipe is failing when rm_work is present?  Only flutter app?

Markus43, 2024-02-05:
> What recipe is failing when rm_work is present? Only flutter app?

flutter-sdk-native is the one that fails with the above error message, and after that all flutter apps fail.

perezmeyer, 2024-02-05:
@jwinarske Docker container, using rm_work. But I do not suspect the container here, but something not playing nicely with rm_work.

perezmeyer, 2024-02-05:
@jwinarske now that you mention it... I have always built the AGL current master with `rm_work` enabled. So the version of meta-flutter on it does not creates any issues. Or somehow the build order happens to paper over it?

Markus43, 2024-02-05:
I now also tested a fresh build with empty DL_DIR / SSTATE_CACHE and *without* `INHERIT += "rm_work"`, and in that case I also get the _Missing "curl" tool_ error (!). 

When I then clean flutter-sdk-native and flutter-engine and build my image again, the build succeeds. When I then build again in a new build folder, but from the same SSTATE_CACHE / DL_DIR, the build also succeeds.

Since I also see the issue without rm_work, I wonder if it has to do anything with the issue at all? Or if it's just a problem with dependency resolution of flutter-sdk-native - the root issue is that curl is not found, and when building the second time, flutter_engine has been built in-between (which takes very long, so all other recipes with no dependencies have been built in-between) ?

Another thing that I noticed when testing this with several builds with both fresh and old cache: I only get the "missing curl" exactly once. A follow-up build (without cleaning anything) then doesn't rebuild flutter-sdk, even if it failed before. And then, flutter apps depending on flutter sdk start to fail because of missing files. Could it be that the failures in do_unpack are not properly handled and thus are not recognized as build failures by the yocto environment?

Sorry for the long delays when reporting back by the way: I really did a lot of builds until I finally was able to reproduce something here, so this did cost me a lot of waiting time, waiting for builds, again and again, thus the delay.

perezmeyer, 2024-02-05:
> I now also tested a fresh build with empty DL_DIR / SSTATE_CACHE and without INHERIT += "rm_work", and in that case I also get the Missing "curl" tool error (!).

I saw that doing a build without cleaning the sstate cache, which makes sense. rm_work should get stuff from the cache if things need to be replay.

>  Could it be that the failures in do_unpack are not properly handled and thus are not recognized as build failures by the yocto environment?

Now hat's actually a nice idea to check.

jwinarske, 2024-02-05:
@Markus43 @perezmeyer Please see if https://github.com/meta-flutter/meta-flutter/pull/402 resolves the issue.  

Quickest way to repro/test:
```
bitbake ca-certificates-native -cdo_cleanall
bitbake curl-native -cdo_cleanall
bitbake ninja-native -cdo_cleanall
bitbake unzip-native -cdo_cleanall
bitbake flutter-sdk-native -cdo_cleanall
bitbake flutter-sdk-native
```
I tested in two scenarios, but I am suspecting a race condition with a smaller core count machine.  My build machines are 32+ HW threads.  When testing uni-proc everything is serialized correctly.  
conf/local.conf addition:
```
PARALLEL_MAKE = "-j 1"
BB_NUMBER_THREADS = "1"
```
With the non-uni-proc build I did see this which would appear to be a bug.  I sent out an email to the OE-Core mailing list to clarify:
[image]

jwinarske, 2024-02-05:
merged to kirkstone for clarity of my OE-Core message.

jwinarske, 2024-02-05:
@Markus43 
> Sorry for the long delays when reporting back by the way: I really did a lot of builds until I finally was able to reproduce something here, so this did cost me a lot of waiting time, waiting for builds, again and again, thus the delay.

No worries.  It's easy enough to work around (`bitbake curl-native` before `bitbake flutter-sdk-native`).  The good news is that you are participating and making meta-flutter better!  There are many people who don't participate.  So thank you for that!

What is the HW thread cound on your build machine?

perezmeyer, 2024-02-05:
Awesome, I'll test it as soon as I finish some other builds I have!

My machine has 8 cores with two threads each.

jwinarske, 2024-02-05:
Tip of tree resolves it: 591ea0eefa482ab45e9b09c4173108ef0152551a
This issue should be good to go.

perezmeyer, 2024-02-05:
@jwinarske this is so far looking good. I will not declare victory until my from-scratch build ends, but I think I saw the errors earlier.

jwinarske, 2024-02-05:
This is the minimum viable repro case, and should be used to validate fix on your machine:
```
bitbake ca-certificates-native -cdo_cleanall
bitbake curl-native -cdo_cleanall
bitbake ninja-native -cdo_cleanall
bitbake unzip-native -cdo_cleanall
bitbake flutter-sdk-native -cdo_cleanall
bitbake flutter-sdk-native
```
flutter-sdk-native should not start until curl-native is done.  I was seeing them run in parallel before the fix.  Which on a higher core count machine would likely not hit the issue (my case).  A lower core count machine would be more likely to hit it.

perezmeyer, 2024-02-05:
I confirmed with a clean, from-scratch build, having rm_work enabled. I still need to build an app yet

perezmeyer, 2024-02-06:
And I could build an app too, always with `rm_work` enabled. Thanks a lot @Markus43 and @jwinarske  !

Markus43, 2024-02-06:
Both building from scratch and incremental builds work again for me, also with rm_work. Many thanks @jwinarske for your support!

closed 2024-02-06 by Markus43