meta-flutter/meta-flutter #387: Package build stalls and crashes for debug releases

matsch82, 2024-01-22:
Hi,

i am currently working on a project that uses flutter on a Xilinx device, thus we have to use petalinux and do some patches to get it working. 
Unfortunately. the app debug bulds are not working. It shows a strange behaviour that is probably not related to the patches, but more to the  flutter sdk tooling.

In this example the `flutter-packages-example-video-player`, but it the same for all packages.  Release builds are working fine, but if ` FLUTTER_APP_RUNTIME_MODES = "debug"`  is set, 
flutter app build process crashes after a very long time, and consuming 30+ GB RAM.

`ERROR: flutter-packages-example-video-player-git-r0 do_compile: failed -6 (cmd was /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot --deterministic --snapshot_kind=app-aot-elf --elf=libapp.so --strip --obfuscate .dart_tool/flutter_build/*/app.dill):                                                                   
../../third_party/dart/runtime/vm/kernel_loader.cc: 221: error: Trying to load a concatenated dill file at a time where that is not allowed 
version=3.1.5 (stable) (Tue Oct 24 04:57:17 2023 +0000) on "linux_simarm64"
  pid=768640, thread=768640, isolate_group=isolate(0x55f1c35e8390), isolate=(nil)((nil))                                                                                                                                     os=linux, arch=arm64, comp=no, sim=yes                                                                                                                                                                                     isolate_instructions=0, vm_instructions=0                                                                                                                                                                                  fp=7ffeeae0df60, sp=7ffeeae0de28, pc=55f1c30cedec
  pc 0x000055f1c30cedec fp 0x00007ffeeae0df60 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x446dec
  pc 0x000055f1c2f3fae4 fp 0x00007ffeeae0e040 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x2b7ae4
  pc 0x000055f1c3020e80 fp 0x00007ffeeae0e0b0 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x398e80
  pc 0x000055f1c2f63a98 fp 0x00007ffeeae0e570 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x2dba98
  pc 0x000055f1c2f6388e fp 0x00007ffeeae0e610 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x2db88e
  pc 0x000055f1c304c80a fp 0x00007ffeeae0e770 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x3c480a
  pc 0x000055f1c2fd0b4d fp 0x00007ffeeae0e840 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x348b4d
  pc 0x000055f1c2fd0d90 fp 0x00007ffeeae0e8b0 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x348d90
  pc 0x000055f1c339028b fp 0x00007ffeeae0ee50 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x70828b
  pc 0x000055f1c339081e fp 0x00007ffeeae0ef20 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x70881e
  pc 0x000055f1c2f38fb2 fp 0x00007ffeeae0f0c0 /home/peta/projects/share/petalinux/build/tmp/work/cortexa72-cortexa53-xilinx-linux/flutter-packages-example-video-player/git-r0/git/engine_sdk/sdk/clang_x64/gen_snapshot+0x2b0fb2
-- End of DumpStackTrace
WARNING: flutter-packages-example-video-player-git-r0 do_package_qa: QA Issue: File /usr/share/flutter/video_player_example/3.13.9/release/lib/libapp.so in package flutter-packages-example-video-player contains reference to TMPDIR [buildpaths]`

is this a known issue?
I try to strip it down to a minmal example, but I assume it will happen with other versions/environments as well.

TeknoVenus, 2024-01-22:
I just hit the same issue after updating to the latest commit - seems there's an invalid if statement that triggers AOT build for debug apps, which is not a valid condition.

Opened a PR to fix

matsch82, 2024-01-22:
Thank you very much! This has actually solved the problem!

closed 2024-01-22 by matsch82

jwinarske, 2024-01-22:
@matsch82 Hi Matthias,

The recommended development work flow for runtime=engine is to use custom-devices from host side.  Installing a debug image should be avoided (footprint large, low performance).  Rather you want to use custom-devices from the host debug machine.  This is 100% supported by the flutter tooling and works well.  Example would be to navigate to your flutter app, then `flutter run -d <my custom device>`.  The custom-device config is what creates bundle folder on target and starts embedder.  If you need a clear example of how to do this on your target I am happy to help.

TeknoVenus, 2024-01-22:
I'm aware of that, but the current recipe has a malformed if statement that means it attempts to perform AOT for debug images. That's what the PR fixes, so it behaves as you describe.

jwinarske, 2024-01-22:
Cool thanks.  Yeah we only use host tools to run debug on target using custom-devices.  Working tools here:
https://github.com/meta-flutter/workspace_automation

If I were to install the debug images from all the example apps it would be a very large OS image.  Not to mention I can ad-hoc push the debug bits from the host at any point, as all engine variants are installed by default.