meta-flutter/meta-flutter #395: gn-fetcher fails to find python

Simone-RM, 2024-02-02:
Hello,

I've added the meta-flutter layer to my yocto configuration (running on Ubuntu 22.04.3). When I try "bitbake flutter-engine" (or any other app, eg. flutter-animated-background), I get the following error:

```
WARNING: flutter-engine-git-r0 do_fetch: Failed to fetch URL gn://github.com/flutter/engine.git;name=flutter, attempting MIRRORS if available
ERROR: flutter-engine-git-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export PATH="/home/simone/yocto/engicam-kirkstone/build/tmp/sysroots-uninative/aarch64-linux/usr/bin:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/bin/python3-native:/home/simone/yocto/engicam-kirkstone/sources/poky/scripts:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot/usr/bin/crossscripts:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/sbin:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/bin:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/sbin:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/bin:/home/simone/yocto/engicam-kirkstone/sources/poky/bitbake/bin:/home/simone/yocto/engicam-kirkstone/build/tmp/hosttools"; export HOME="/home/simone"; export PATH="/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/share/depot_tools:/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/share/depot_tools/bootstrap-2@3.8.9.chromium.14_bin/python/bin:${PATH}"; export DEPOT_TOOLS_UPDATE=0; export GCLIENT_PY3=0;             export CURL_CA_BUNDLE=/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/etc/ssl/certs/ca-certificates.crt;             gclient.py config --spec 'solutions = [
    {
        "managed": False,
        "name": "src/flutter",
        "url": "https://github.com/flutter/engine.git",
        "custom_vars": {    "download_android_deps": False,     "download_windows_deps": False,     "download_linux_deps": False,   },
        "custom_deps": {}
    }
]' &&             gclient.py sync --shallow --no-history -R -D --revision f40e976bedff57e69e1b3d89a7c2a3c617a03dad -j 8 -v failed with exit code 127, output:
/usr/bin/env: ‘python’: No such file or directory
```

any idea how to fix it?

Python3 is installed on my host (via the python-is-python3 package):

```
$ /usr/bin/env python
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
```

I'm using the latest kirkstone branch (commit f4f58f952482793a3ddb1ea3685733e48da45b8a)

Simone-RM, 2024-02-02:
Replacing `/usr/bin/env python` with `/usr/bin/python` in `tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot-native/usr/share/depot_tools/gclient.py` seems to help (it's downloading a lot of stuff now). It's not ideal though.

jwinarske, 2024-02-02:
@Simone-RM Hi.  This layer is in production in a number of places without issue.  So the likelihood is that you have a host configuration issue.  Please check that your Yocto system per-requisites are met:
https://docs.yoctoproject.org/4.0.15/ref-manual/system-requirements.html

Simone-RM, 2024-02-05:
I think I have all the prerequisites, and I've been building images (without Flutter) on this setup for a while now.

One other bit of information: I'm running Ubuntu on aarch64 (Apple Silicon with Parallels). Has anyone every tried that?

I'm starting to think that is the real problem... changing gclient's shebang allows it to download everything, but then compilation fails:

```
Generating GN files in: out/linux_debug_arm64
NOTE: is_debug = false use_goma = false goma_dir = "None" full_dart_sdk = false enable_unittests = false skia_enable_flutter_defines = true skia_use_dng_sdk = false skia_use_sfntly = false skia_enable_pdf = false skia_use_x11 = false skia_use_wuffs = true skia_use_expat = true skia_use_fontconfig = true skia_use_icu = true is_official_build = true android_full_debug = false is_clang = true skia_gl_standard = "" use_clang_static_analyzer = false enable_coverage = false enable_lto = true host_os = "linux" host_cpu = "arm64" target_os = "linux" target_cpu = "arm64" dart_target_arch = "arm64" flutter_use_fontconfig = true dart_component_kind = "static_library" embedder_for_target = true dart_lib_export_symbols = false flutter_runtime_mode = "debug" dart_version_git_info = true dart_runtime_mode = "develop" enable_desktop_embeddings = true enable_backtrace = true bssl_use_clang_integrated_as = true target_sysroot = "/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot" custom_sysroot = "/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/recipe-sysroot" custom_toolchain = "/home/simone/yocto/engicam-kirkstone/build/tmp/work/armv8a-poky-linux/flutter-engine/git-r0/src/buildtools/linux-x64/clang" custom_target_triple = "aarch64-unknown-linux-gnu" skia_use_gl = true skia_use_vulkan = true skia_use_vma = false shell_enable_vulkan = true disable_hidden_visibility = false dart_include_wasm_opt = false dart_platform_sdk = true build_glfw_shell = false build_embedder_examples = false stripped_symbols = false dart_use_mallinfo2 = true impeller_enable_3d = true impeller_enable_vulkan = true impeller_enable_opengles = true angle_build_all = false angle_has_astc_encoder = false angle_use_x11 = false angle_use_wayland = false angle_enable_vulkan_validation_layers = false angle_vulkan_headers_dir = "//third_party/vulkan-deps/vulkan-headers/src" angle_vulkan_loader_dir = "//third_party/vulkan-deps/vulkan-loader/src" angle_vulkan_tools_dir = "//third_party/vulkan-deps/vulkan-tools/src" arm_tune = "armv8a+crc+crypto"
Unknown architecture (aarch64) -- unable to run ninja.
No prebuilt ninja binary was found for this system.
Try building your own binary by doing:
  cd ~
  git clone https://github.com/ninja-build/ninja.git -b v1.8.2
  cd ninja && ./configure.py --bootstrap
Then add ~/ninja/ to your PATH.
WARNING: exit code 1 from a shell command.
```

which is a pity (also considering I do have a prebuilt ninja installed as /usr/bin/ninja)

jwinarske, 2024-02-05:
@Simone-RM Good call on adding the details and renaming the issue.

I have never built a Poky Distro on an Aarch64 machine.  Largely because all the machines are so largely under-powered, and I would prefer to watch grass grow :)  I build on an AMD Thread Ripper 32+ HW threads, Intel 32 HW threads, AGL builds run on self hosted High Availability servers, and others use Extra Large Virtual CPU instances (AWS).  AWS is now offering Yocto image builds as a service.

Are you building a full OS image, or just trying to build Flutter engine artifacts?
* If full OS image can you build it short of the meta-flutter recipes?
* If just flutter engine artifacts I would suggest a standalone command line build with my flutter-engine patches.

jwinarske, 2024-02-05:
You might also try updating depot-tools to tip of tree, and switching it to python3.  It's using an older commit with python2, as this was a requirement for flutter-engine; may have changed in the last six months.  You can also try adding ninja-native to DEPENDS, keep in mind ninja is part of depot-tools; which might be an issue with your newer combo (Ubuntu running on apple silicon).

jwinarske, 2024-02-05:
I tried rolling depot-tools to tip of tree main, and it's complaining about missing `python3-httplib2`.  I don't have time to dig into this at the moment.

Simone-RM, 2024-02-06:
> I have never built a Poky Distro on an Aarch64 machine. Largely because all the machines are so largely under-powered, and I would prefer to watch grass grow :) I build on an AMD Thread Ripper 32+ [...]

I was using an AWS machine for my builds, but having everything local on my machine is more practical. And building on a new Mac is not that much slower than on the 32-cores AWS server. ;-)

> Are you building a full OS image, or just trying to build Flutter engine artifacts?

full image. I'm trying to add flutter to a working image, just to see if it works/how it performs on the boards we're using.

>     * If full OS image can you build it short of the meta-flutter recipes?

yes

> You might also try updating depot-tools to tip of tree, and switching it to python3. It's using an older commit with python2, as this was a requirement for flutter-engine; may have changed in the [...]

I'll try something like that. I've seen that there are several scripts that only know about x64 arch as far as Linux is concerned.

I've also tried building an image on Ubuntu 20 on an AWS server, including meta-flutter-animated-background-example: it built correctly, but I suspect that it didn't include everything needed, as I couldn't find a way to launch the app...

I tried building flutter-auto, but it causes the compiler to crash :(
```
| FAILED: shell/CMakeFiles/flutter-auto.dir/backend/wayland_vulkan.cc.o
| /yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ --sysroot=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/recipe-sysroot/usr -DASIO_DISABLE_HANDLER_TYPE_REQUIREMENTS -DASIO_HEADER_ONLY -DASIO_NO_DEPRECATED -DASIO_NO_TYPEID -DASIO_STANDALONE -DBUILD_BACKEND_WAYLAND_VULKAN -DCRASH_HANDLER_DSN=\"\" -DCRASH_HANDLER_RELEASE=\"\" -DEGL_NO_X11 -DENABLE_AGL_CLIENT -DENABLE_DART_VM_LOGGING -DENABLE_PLUGIN_ACCESSIBILITY -DENABLE_PLUGIN_COMP_REGION -DENABLE_PLUGIN_COMP_SURF -DENABLE_PLUGIN_DESKTOP_WINDOW -DENABLE_PLUGIN_GOOGLE_SIGN_IN -DENABLE_PLUGIN_GO_ROUTER -DENABLE_PLUGIN_ISOLATE -DENABLE_PLUGIN_KEYBOARD_MANAGER -DENABLE_PLUGIN_KEY_EVENT -DENABLE_PLUGIN_LOGGING -DENABLE_PLUGIN_MOUSE_CURSOR -DENABLE_PLUGIN_PACKAGE_INFO -DENABLE_PLUGIN_PLATFORM -DENABLE_PLUGIN_RESTORATION -DENABLE_PLUGIN_TEXT_INPUT -DENABLE_PLUGIN_URL_LAUNCHER -DENABLE_XDG_CLIENT -DENV64BIT -DFILAMENT_SUPPORTS_WAYLAND -DGIT_BRANCH=\"agl\" -DGIT_HASH=\"3114782\" -DHAVE_MEMFD_CREATE -DHAVE_STRCHRNUL -DLINUX -DMESA_EGL_NO_X11_HEADERS -DNDEBUG -DPATH_PREFIX=\"/usr\" -DWL_EGL_PLATFORM -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/build/shell -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/shell -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/shell/.. -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/shell/logging -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/shell/../third_party/spdlog-1.12.0/include -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/shell/../third_party/asio-1-28-1/asio/include -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/filament/libs/bluevk/include -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/. -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/flutter -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/flutter/third_party -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/flutter/third_party/rapidjson/include -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/flutter/shell/platform/common/public -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/flutter/shell/platform/common/client_wrapper/include -I/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/git/third_party/spdlog-1.12.0/include -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0=/usr/src/debug/flutter-auto/git-r0                      -fdebug-prefix-map=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0=/usr/src/debug/flutter-auto/git-r0                      -fdebug-prefix-map=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/recipe-sysroot=                      -fdebug-prefix-map=/yocto/engicam/build_robby/tmp/work/armv8a-mx8mp-poky-linux/flutter-auto/git-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -fno-rtti -DNDEBUG -flto -fno-fat-lto-objects -fPIE -fvisibility-inlines-hidden -Wtrigraphs -Wchar-subscripts -Wcomment -Wreturn-type -Wsequence-point -Wswitch -Wuninitialized -Wunused -Wswitch-bool -Wformat -Wformat-security -Wconversion -Wcast-align -Wunused-parameter -Winvalid-offsetof -fsigned-char -O2 -std=c++17 -MD -MT shell/CMakeFiles/flutter-auto.dir/backend/wayland_vulka [... cut ...]
| 	???:0
| 0x7eb604 c_parse_file()
| 	???:0
| 0x8be425 c_common_parse_file()
| 	???:0
| Please submit a full bug report,
| with preprocessed source if appropriate.
| Please include the complete backtrace with any bug report.
| See <https://gcc.gnu.org/bugs/> for instructions.
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/yocto/engicam/sources/meta-flutter/recipes-graphics/toyota/flutter-auto_git.bb:do_compile) failed with exit code '1'
```

jwinarske, 2024-02-06:
I have an M2 I rarely use.  I've ran Ubuntu on it, and perf was not so great.  I will try Ubuntu as a QEMU image with my QEMU launcher app to see if perf is any better:  https://github.com/meta-flutter/virt_qemu

That said I also exclusively use a raid10 SSD card on my workstations.  Aside from high core count CPU, fast disk IO is the next bottleneck on Yocto builds.  You can build a nice Yocto build workstation for ~$5k US.  It pays for itself in a few months of regular use.

I would start with standalone building the flutter engine directly on Ubuntu on m2.  With that working and if there are a clear set of patches (or none), then doing the Yocto build will be much clearer.

jwinarske, 2024-02-07:
I moved depot-tools to tip of tree.  See if that helps

Simone-RM, 2024-02-08:
thanks, it does help. Now flutter_engine compiles with a simple bbappend to fix clang's path:
```
CLANG_PATH = "${WORKDIR}/src/buildtools/linux-arm64/clang"
```

I'll have to fix the install job too somehow (all the 'clang_x64' instances should be replaced with 'clang_arm64') to work.

flutter-auto compilation crashes just like on the Intel server though.

jwinarske, 2024-02-09:
What value is assigned to HOST_ARCH?  I'll remove the DISTRO_FEATURES filter for vulkan backend selection.

Simone-RM, 2024-02-09:
HOST_ARCH is aarch64

jwinarske, 2024-02-10:
https://github.com/meta-flutter/meta-flutter/pull/409/files
Passes building flutter-gallery on Mac M2 -> UTM virtualization -> Ubuntu 22

jwinarske, 2024-02-10:
merged

jwinarske, 2024-02-10:
let me know if that unblocks you

Simone-RM, 2024-02-12:
I can confirm that this issue can be closed, thanks!

closed 2024-02-12 by Simone-RM