meta-flutter/meta-flutter #411: pdfium does not build with arm64 host

jwinarske, 2024-02-12:
This is currently a limitation with the PDFium project itself.  I only references linux-x64 in the DEPS file.

The fetch fails looking for linuxarm64 cip asset.

leojrfs, 2025-06-07:
I can confirm it fails with:

```
failed to resolve infra/rbe/client/linux-arm64@re_client_version:0.148.0.41b09b51-gomaip (line 8): no such package: infra/rbe/client/linux-arm64
```

rbe binaries don't seem to be available for `linux-arm64`:
- https://issues.chromium.org/issues/40277110

leojrfs, 2025-06-09:
After investigating further, i found a temporary solution.

#### There are currently two issues with `pdfium`:

1. `'buildtools/reclient'` in `DEPS` sets its package name as `'package': Var('reclient_package') + '${{platform}}'`.
This means that when building on `arm64` hosts the package name will solve to `infra/rbe/client/linux-arm64`, which does not exist:
    - https://chrome-infra-packages.appspot.com/p/infra_internal/rbe/client
The same issue was found in chromium and the solution was to enable this package conditionally for non linux-arm64 hosts:
    - https://issues.chromium.org/issues/40277110
    - https://chromium.googlesource.com/chromium/src/+/d2c1621d8e02c804b67cf1722595754114cb9aa%5E%21/#F1

2. `'buildtools/linux64'` in `DEPS` is hardcoded to `gn/gn/linux-amd64`. This is because the `pdfium` version in the recipe available in this repo (at the time of writting this comment) predates the change that fixes this issue:
    - this repo:
        - https://github.com/meta-flutter/meta-flutter/blob/2e600c6a3fd362de6961fea89af85b22d499f4bd/recipes-graphics/pdfium/pdfium_130.0.6721.0.bb#L25
        - https://pdfium.googlesource.com/pdfium/+/2b675cf15ab4b68bf1ed4e0511ba2479e11f1605/DEPS#248

    - upstream fix:
        - https://pdfium.googlesource.com/pdfium/+/cc0f452e0b93e8ad80c096b8b40b3a736b5e7435%5E%21/DEPS

#### Fixes and workarounds

1. Ideally, upstream would implement a similar fix to the one done in chromium. Until then we can either:
    - fork the repo and just comment the `buildtools/reclient` package declaration in `DEPS`
    - use the `DEPS` file vars and the `gclient` `custom_vars` feature to override the `reclient_package` and `reclient_version` giving it something to download in the `gclient sync` process. Heres what I am currently doing using the `GN_CUSTOM_VARS` in my `pdfium_%.bbappend`:

```
GN_CUSTOM_VARS = '\
{\
    "reclient_package": "gn/gn/", \
    "reclient_version": "Bh4zTcV8OhLnxaKF5mZ0yoyJdcBeyyQcxGxFi2DKOp0C", \
    "checkout_configuration": "small" \
}'
```

2. Im currently overriding the `SRCREV` in my `pdfium_%.bbappend`, but an update of `pdfium` in this repo would be great.

leojrfs, 2025-06-11:
Fix for problem `2.` is available at:  https://github.com/meta-flutter/meta-flutter/pull/665

leojrfs, 2025-06-12:
Potential fix for problem `1.`: https://issues.chromium.org/issues/424205782

leojrfs, 2025-06-13:
> merged: [#665](https://github.com/meta-flutter/meta-flutter/pull/665)
> 
> Have not yet confirmed issue is resolved on aarch64 host

Thanks, that solves problem `2.`. Problem `1.` needs either fix upstream and another bump of the version here, or temporarily, the workaround mentioned before:

```
GN_CUSTOM_VARS = '\
{\
    "reclient_package": "gn/gn/", \
    "reclient_version": "git_revision:b99a82ca8ee957da829d6313b818b99df8e7ccb8", \
    "checkout_configuration": "small" \
}'
```

leojrfs, 2025-06-26:
Can this issue now be closed?

jwinarske, 2025-06-30:
Just need a fix confirmation from stakeholder(s)

jwinarske, 2026-09-01:
This needs to move to an FFI module.

jwinarske, 2026-09-18:
@leojrfs the workaround landed and is still in the recipe:

```
GN_CUSTOM_VARS ?= '\
{\
    "reclient_package": "gn/gn/", \
    "reclient_version": "git_revision:b99a82ca8ee957da829d6313b818b99df8e7ccb8", \
    "checkout_configuration": "small" \
}'
```

plus #665 for the gn binary.  Can you confirm pdfium builds on an aarch64 host against current master?  I have no arm64 builder, so this has been sitting on that one data point since June.

If it builds, I'll close this and open a separate one for moving pdfium to an FFI module -- different problem, and it shouldn't keep this open.

Unrelated fix that may matter to you if you build for x86_64 targets: #990.  pdfium was building its own nasm with the target toolchain.