meta-flutter/meta-flutter #415: Missing providers for nativesdk-flutter-sdk (?)

kwavnet, 2024-02-20:
After upgrading from 3.13.x to latest 3.19 we faced

```
2024-02-19T15:01:33.2462337Z User-Agent: constructed: 'libdnf (Ubuntu 20.04; generic; Linux.x86_64)'
2024-02-19T15:01:33.2463556Z repo: using cache for: oe-repo
2024-02-19T15:01:33.2464598Z oe-repo: using metadata from Mon 19 Feb 2024 02:57:01 PM UTC.
2024-02-19T15:01:33.2466093Z Last metadata expiration check: 0:00:01 ago on Mon 19 Feb 2024 03:01:32 PM UTC.
2024-02-19T15:01:33.2468132Z --> Starting dependency resolution
2024-02-19T15:01:33.2469062Z --> Finished dependency resolution
2024-02-19T15:01:33.2469894Z Error: 
2024-02-19T15:01:33.2470659Z  Problem: conflicting requests
2024-02-19T15:01:33.2472053Z   - nothing provides libfontconfig.so.1()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2473913Z   - nothing provides libpango-1.0.so.0()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2475795Z   - nothing provides libpangocairo-1.0.so.0()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2477650Z   - nothing provides libatk-1.0.so.0()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2479441Z   - nothing provides libgdk-3.so.0()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2481208Z   - nothing provides libgtk-3.so.0()(64bit) needed by nativesdk-flutter-sdk-git-r0.x86_64_nativesdk
2024-02-19T15:01:33.2482671Z (try to add '--skip-broken' to skip uninstallable packages)
```

Adding

```
RDEPENDS:${PN} += "\
    atk \
    fontconfig \
    gtk+3 \
    pango \
"
```

to flutter-sdk_git.bb (through a bbappend) fixed it for us - just wanted to check if can be reproduced by someone else before posting a PR

jwinarske, 2024-02-20:
@kwavnet What is the output from?  Attempting to build under Yocto SDK environment?

jwinarske, 2024-02-21:
I ran multiple tests on my end with AGL master (with ToT meta-flutter) without hitting issue.  That said I implemented the additional runtime packages, and it did not break so I went ahead and merged the change:
https://github.com/meta-flutter/meta-flutter/commit/ae8df8fe0cf3599dcfe73a94f47ab22166d4000e

closed 2024-02-21 by jwinarske