meta-flutter/meta-flutter #428: dart-app class

jwinarske, 2024-02-23:
Create a class to support building executable Dart apps.  Depends on dart-sdk-native.

r3flow, 2024-02-24:
Does this mean a standalone cross-compiled executable or a portable kernel module that requires the Dart SDK (on the target) to run?

jwinarske, 2024-02-27:
Standalone cross compiled executable.  

The dart-sdk scenario already exists.  You just need to install the dart code.

r3flow, 2024-02-27:
That would be great! Based on your work, I use cross-compiled Dart apps, but the method I use doesn't work since Dart >=3.2. :-( I haven't had time to fix the problem since then, it would be nice if this feature could be included in this repo. :)

https://github.com/r3flow/meta-dart

jwinarske, 2026-09-22:
@r3flow still open, still wanted.  Nothing has been written, but the piece that made it awkward is in place now, so here is the shape I'd build.

## What changed

A standalone cross-compiled Dart executable needs two things: a gen_snapshot that runs on the builder and emits code for the target, and the target's dartaotruntime to pair the snapshot with.  The second has always been there -- the dart-sdk recipe installs a full SDK to ${datadir}/dart-sdk.  The first was the awkward one.

#1009 solved that exact problem for flutter-engine: its host tools are deployed per MACHINE and per runtime mode, and a small recipe (flutter-engine-sdk, native + nativesdk) stages them into recipe-sysroot-native where uninative relocates them.  gen_snapshot there is a build-machine binary emitting target code, which is precisely the shape a dart-app class needs.

The dart-sdk recipe already builds the same way -- `-a arm64_${GN_HOST_ARCH}`, the cross-canadian form -- so its own host-side gen_snapshot is produced today and then thrown away.

## Plan

1. dart-sdk grows a do_deploy for its host-side tools, per MACHINE, mirroring what flutter-engine now does.
2. A small native recipe stages them, same pattern as flutter-engine-sdk.  This is the missing dart-sdk-native piece this issue asked for -- the SDK variant exists, the cross tooling does not.
3. dart-app.bbclass: `dart compile aot-snapshot` (or gen_snapshot directly) against that, then package the snapshot with the target dartaotruntime.  RDEPENDS on dart-sdk for the runtime, or concatenate into a single executable the way `dart compile exe` does locally.

## Two things to get right

**Version pairing.**  Use the dart-sdk recipe's gen_snapshot, not flutter-engine's.  The engine carries Flutter's fork of Dart pinned to the engine revision; a snapshot from that paired with a plain dart-sdk dartaotruntime is a version mismatch waiting to happen.

**`dart compile exe` is not the mechanism.**  It does not cross-compile, which is why the pre-3.2 method in meta-dart stopped working rather than needing a small fix.  The two-step snapshot + runtime route is the one that survives SDK churn.

Not scheduled yet.  Posting the plan so it is not re-derived, and so meta-dart's approach can be compared against it.

jwinarske, 2026-09-23:
Correction to step 1 above.  I said the dart-sdk recipe already produces a host-side gen_snapshot that targets the target, and that it is thrown away.  It does not.  Built it to check.

dart-sdk for qemuarm64, 3.13.4, out dir `ProductXARM64_X64`:

```
gen_snapshot                        ELF 64-bit LSB pie executable, ARM aarch64
gen_snapshot_product                ELF 64-bit LSB pie executable, ARM aarch64
gen_snapshot_product_linux_x64      ELF 64-bit LSB pie executable, ARM aarch64
dartaotruntime                      ELF 64-bit LSB pie executable, ARM aarch64
```

Everything in there is aarch64, including the variants named for other targets -- those are target-hosted generators emitting x64 or riscv64 code, which is the opposite of what a cross build on the host needs.  The cause is in the recipe: do_configure exports

```
DART_USE_SYSROOT="${TARGET_SYSROOT}"
DART_USE_TOOLCHAIN="${STAGING_DIR_NATIVE}/usr/bin"
```

so gn builds everything with the OE cross toolchain, host-intended targets included.  The `-a arm64_${GN_HOST_ARCH}` argument names the pair but does not, under that environment, get us an x86-64 binary.

For reference the x86-64 target build (`ProductXX64_X64`) looks fine, but that case is degenerate -- host and target are the same arch, so it proves nothing.

The installed SDK is as expected: `image/usr/share/dart-sdk/bin/dartaotruntime` is aarch64, so the runtime half of a dart-app is available on target today.

## What this does to the plan

Step 1 is not "deploy what is already built".  It is "arrange for a host gen_snapshot that targets the target", which is a different piece of work:

- a native variant of dart-sdk built with the native toolchain and `dart_target_arch` set to the target, rather than the cross toolchain, or
- flutter-engine's gen_snapshot, which already is a host binary emitting target code -- with the version-pairing caveat I raised above, and it is Flutter's fork rather than the dart-sdk this class would pair with.

Steps 2 and 3 stand.  Posting this before anyone builds on the wrong premise.

jwinarske, 2026-09-23:
Retracting the correction above -- it was wrong, and the original plan stands.  I looked in the mode root and missed the host toolchain subdirectory.

qemuarm64, dart-sdk 3.13.4, `out/ProductXARM64_X64/clang_x64/`:

```
gen_snapshot           ELF 64-bit LSB pie executable, x86-64
gen_snapshot_product   ELF 64-bit LSB pie executable, x86-64
```

and it identifies its target itself:

```
$ out/ProductXARM64_X64/clang_x64/gen_snapshot --version
Dart SDK version: 3.13.4 (stable) (Tue Sep 15 01:01:15 2026 -0700) on "linux_simarm64"
```

x86-64 binary, arm64 codegen.  Same `clang_<host>/` layout flutter-engine uses, and it is discarded at the end of the build today.

What I reported before -- everything aarch64 -- was the mode root, which holds the target-side binaries.  Both are there; I checked one and generalised.

The x86-64 target build has no `clang_x64/` subdirectory at all, because host and target are the same arch and nothing separate is needed.  That is why it looked fine.

So step 1 is what I first said: deploy `out/<mode>/clang_<host>/` from dart-sdk, per MACHINE, the same way flutter-engine now deploys its host tools.  Steps 2 and 3 unchanged.  The runtime half is already in place -- the installed SDK's dartaotruntime is aarch64.

closed 2026-09-24 by jwinarske