meta-flutter/meta-flutter #429: Generate version.json when building flutter apps

Markus43, 2024-02-23:
Packages like [package_info_plus](https://pub.dev/packages/package_info_plus) provide ways to read the name and version of an app programmatically within flutter apps.
This doesn't work with custom embedders unfortunately: flutter "bundle" builds simply do not contain these infos in a parsable way.
On linux gtk builds, this info is available in a file version.json directly in the flutter_assets folder.

Typical content:
> {"app_name":"my_app","version":"1.2.3","build_number":"42","package_name":"my_app"}

Would it maybe make sense to generate the same file with flutter app builds in this recipe?

According to the reply I got on custom embedders discord, this is currently expected to be generated by the tooling that belongs to the embedder ...

jwinarske, 2024-02-23:
I was looking at this yesterday.  package_info_plus requires this file.  In the case of the linux it doesn't use the platform interface, and it expects the file to be found relative to the embedder; which is not the case.  This is kind of quirky in the first place.

Markus43, 2024-02-23:
> ... This is kind of quirky in the first place.

Yes it is. I don't understand why they don't just put this information in a build environment variable and wrap it with a String.fromEnvironment in some k*** constant so it can be accessed where necessary. On Windows. they put it in the .exe, and package_info_plus extracts it back from the exe.

Probably for meta-flutter it would be best to just put such things in a dart-define, e.g. `meta_flutter.appVersion`, so they can just be parsed like that.

jwinarske, 2024-02-23:
Ultimately it would be best if it were a federated plugin.  Then we could easily override the behavior.  This is what I like about the first party stuff.  I supposed two options:
1. Fork the plugins (too much overhead to manage)
2. Upstream a better design

I'm finding quite a few plugins that could use a re-work.  Conversion to federated, and pigeon where it makes sense.

There really should be a standard way of returning the flutter_asset directory to a plugin.  Perhaps it's seen as a security risk, not sure why it was omitted in the registrar.

In the case of the linux variant of package_info_plus it avoids the platform channel call, so it's a cheap call.

jwinarske, 2024-02-23:
I just dialoged with Chinmay.  He is indicating that if flutter_assets is mapped as an asset_manager (engine side) you can just use relative paths in Dart, and it will iterate all asset_managers until it finds a match.  So the approach in the Linux Dart code for package_info_plus could simply load `version.json`, opposed to "finding the executable approach".  Not sure if that's in the Linux embedder tree, but mapping the `flutter_assets` folder to an asset_manager really should be added.  If it's already there it would be an easy Dart change.  I know I've had to work around this in other cases, and this would resolve my work arounds.  This would be a good patch to carry for flutter-engine.

jwinarske, 2024-02-27:
So two parts to address this:
1. https://github.com/fluttercommunity/plus_plugins/pull/2617
2. https://github.com/meta-flutter/meta-flutter/issues/431

jwinarske, 2024-02-27:
Turns out I was premature on the PR (1).  After cleaning the build and re-running it appears there is not a default asset manager for Linux GTK.  So I need to submit an Engine PR.

jwinarske, 2026-08-28:
Closing with what this turned out to be, since the three-part plan above resolves to one part — and not the part expected.

## 1. The engine patch is not needed (#431)

`RunConfiguration::InferFromSettings` already registers `settings.assets_path` as a `DirectoryAssetBundle`, unconditionally:

```cpp
// engine/src/flutter/shell/common/run_configuration.cc
asset_manager->PushBack(std::make_unique<DirectoryAssetBundle>(
    fml::OpenDirectory(settings.assets_path.c_str(), false,
                       fml::FilePermission::kRead), true));
```

and `embedder.cc` sets `settings.assets_path = args->assets_path`, which every embedder supplies. So `flutter_assets` has always been an asset bundle for embedder-API embedders, and `rootBundle.load('version.json')` resolves there. The 2024 observation that closed out the discussion — no default asset manager for Linux GTK — was true of the GTK desktop embedder, which is not the one this layer ships. #431 is closed.

## 2. meta-flutter does not need to generate anything

`do_install` in `conf/include/flutter-app.inc` already copies `build/flutter_assets/*` wholesale into the target bundle. Anything the app puts in `flutter_assets` is installed with no recipe change. The gap was never installation.

A `do_install[postfuncs]` that parsed pubspec.yaml and wrote the file was tried and **discarded**: it put application-specific generation in a build layer, applied to every recipe, and became a fourth implementation of a file flutter already writes twice. It also silently diverged, ignoring `--build-name`/`--build-number`, which flutter honours.

## 3. The actual gap is upstream

`flutter build bundle` does not write `version.json`. flutter_tools generates it only from the desktop and web targets:

| platform | how the app's version reaches it |
|---|---|
| Linux GTK | `linux.dart` → `version.json` in `flutter_assets` |
| Web | `web.dart` → `version.json` at the web root |
| Windows | read from the executable's resources |
| macOS/iOS | `Info.plist` |
| Android | `PackageManager` |
| **`flutter build bundle`** | **nothing** |

Every platform that ships a bundle gets this somehow; the generic bundle path — the one every custom embedder uses — is the case nobody wired up. `CopyFlutterBundle` already passes `additionalContent` for `NativeAssetsManifest.json`, so `version.json` belongs beside it. The payload construction was duplicated verbatim between `linux.dart` and `web.dart`, so it lifts to one helper and gains the missing call site rather than becoming a third copy.

## 4. package_info_plus also assumes a layout

Separately, its Linux implementation resolves `/proc/self/exe` and looks for `dirname(exe)/data/flutter_assets/version.json` — the `flutter build linux` layout. Under Yocto the embedder and the bundle land in unrelated prefixes and one embedder serves several bundles, so no path derived from the executable can be right. It should consult `rootBundle` first, falling back to the existing lookup.

## Where that leaves it

Both changes are written and tested, on branches, not yet proposed upstream. Nothing is needed in meta-flutter: once an SDK carrying the flutter_tools change is in use, `version.json` appears in `flutter_assets` and the existing `do_install` ships it like any other asset.

closed 2026-08-28 by jwinarske