meta-flutter/meta-flutter #430: rust example apps?

Markus43, 2024-02-23:
Since you're currently building a lot of example apps, I'd like to suggest one app type that is still missing in the recipes: I couldn't find a working app example that uses rust bindings. There is something with flutter_rust_bridge, membrane and flutter-test-frb in the github workflows, but that seems to be pretty outdated, and I couldn't really understand what I would need to do to build an app with these.

Do you have an example using flutter_rust_bridge and/or membrane? Both look like a very nice alternative to ffi when it comes to performance-critical things, and embedded development tends to need such things quite often :-)

jwinarske, 2024-02-23:
Yeah I have a process for updating these.  Rust in Yocto is a pain.  I almost update the Rust recipes earlier today.

The process is this:
* update primary recipe SRC_REV
* run bitbake on the recipe, it will fail
* open terminal to path of src
* delete Cargo.lock
* delete toolchain file if present
* add the toolchain version used in yocto to your path
* build the crate using this toolchain.  This is to update Cargo.lock to the correct versions (this is a critical step).
* run my [tool](https://github.com/jwinarske/root-pkg-ws) passing --manifest-path=<repo path>/Cargo.toml
* Copy the output from this tool to the recipe file.
* rebuild and confirm it passes

If you need a newer Rust toolchain for Kirkstone, you can use:
https://git.yoctoproject.org/git/meta-lts-mixins

Feel free to update the flutter-rust-bridge-example recipe post a PR.

I will be moving to the upcoming Yocto LTS branch next week, as AGL is moving.

jwinarske, 2024-02-24:
https://github.com/meta-flutter/meta-flutter/pull/432

jwinarske, 2024-02-24:
This one is under dynamic as it's dependent on clang:
https://github.com/meta-flutter/meta-flutter/blob/kirkstone/dynamic-layers/clang-layer/recipes-devtools/membrane/membrane-example_git.bb

jwinarske, 2024-02-27:
So for membrane this is the example flutter app:
https://github.com/jerel/membrane_template/blob/main/flutter_example/lib/main.dart
membrane auto-generates an API used on the Dart side.  This makes the build step more complicated, as it involves llvm host/target.
* Toyota uses membrane in production.  There is no update to the example app per-say.

Apps/projects that support building multiple components in the same build pass are not Yocto friendly and require adapting.  Both of the above require two recipes.  One for the Rust shared module, one for the Flutter app.

Embedding the building of a native module in your Flutter app is cool and all, but not so great for production environments, and makes creating a Yocto recipe far more time consuming.  Flutter Packages Rive, Pdf, etc do this.  I did some work on Rive last week; it's pretty convoluted and doesn't have a repo for all their source.

Does this help address this issue?  If so please close.

jwinarske, 2024-02-27:
Tip of tree flutter_rust_bridge requires Rust toolchain 1.70.0.  With the AGL toolchain update that brings it to 1.68.2.  So when developing Rust for Flutter and Embedded you have to be sensitive to versions.

Markus43, 2024-02-28:
> Does this help address this issue? If so please close.

All very helpful, many thanks for this! I'll try this out soon and close the issue once I have it up and running.

Markus43, 2024-03-22:
Thanks for these detailed steps! I'm a bit curious as of why the many steps are necessary.

Some questions/feedback regarding your initial steps, in order to get a better understanding:

> update primary recipe SRC_REV

So the "primary recipe" is the rust recipe ... Took me a while to understand that there are two recipes involved because they live in totally different folders. Since the rust recipe is *only* for that one app, I'd suggest moving it to the app for more clarity.

> run bitbake on the recipe, it will fail
> open terminal to path of src
> delete Cargo.lock
> delete toolchain file if present
> add the toolchain version used in yocto to your path
> build the crate using this toolchain. This is to update Cargo.lock to the correct versions (this is a critical step).

Did I understand correctly that I should build the rust project for the host in this part? If so, couldn't I simply do these steps on a linux desktop build where I just change the rust version to the one used in yocto?

> run my [tool](https://github.com/jwinarske/root-pkg-ws) passing --manifest-path=/Cargo.toml
> Copy the output from this tool to the recipe file.

I could not find any explanation why `cargo bitbake` is not sufficient here. I see there's a lot of different output from your tool for the SRC_URI checksums. Why is that part necessary, what kind of problem does it solve?

> If you need a newer Rust toolchain for Kirkstone, you can use:
> https://git.yoctoproject.org/git/meta-lts-mixins

FWIW, updated link: https://git.yoctoproject.org/meta-lts-mixins/log/?h=kirkstone/rust . Many thanks for the hint, I'm aiming for frb version2, where I'll need that for sure.

By the way: Did you ever try https://github.com/rust-embedded/meta-rust-bin ? Currently requires a network connection for the build step (which is of course bad, they have an open task to fix that) but it doesn't seem to require configuring all the dependent crates, which would be a huge maintenance timesaver AFAICS ...

jwinarske, 2024-04-13:
@Markus43 The whole premise of Yocto is reproducible builds.  So any option or choice related to Rust here is guided by this principle.

If your goal is to make artifacts quickly and easily that run on target without concern for reproducibility then idiomatic Yocto is not what you want.

My steps are a result of dealing with some very large rust projects (800+ creates).

Markus43, 2024-04-15:
> @Markus43 The whole premise of Yocto is reproducible builds. So any option or choice related to Rust here is guided by this principle.

I fully agree with this principle, and I'm sorry for any confusion as I failed to make this clear. I'm probably still too much of a beginner on the yocto side to understand the implications of certain statements. I'm trying to improve on that, though ;-)

My basic problem of understanding (that I try to resolve here) is:

* I can build flutter apps that come with a pubspec.lock file, where the lock file specifies what exactly to build. I don't have to provide any checksums for the packages - resolving the versions based on pub.dev is deemed sufficient to create reproducible builds.
* When adding rust parts to the app, there is a cargo.lock file. This works pretty much the same as the flutter lock file, but it looks as if that's not sufficient to create reproducible builds: I need to add checksums for all used crates via the steps you provided.

Where does that difference come from?

jwinarske, 2024-04-15:
Once I implement https://github.com/meta-flutter/meta-flutter/issues/425 then flutter apps will follow the rust pattern of explicitly defining packages and versions.

Currently reproducible builds for flutter apps happen by archiving the entire pubspec folder per app.  While this technically enables reproducible builds, it is wasteful of space when you have many apps.

Markus43, 2024-07-15:
> Yeah I have a process for updating these. Rust in Yocto is a pain. I almost updated the Rust recipes earlier today.
> 
> The process is this:
> 
> * update primary recipe SRC_REV
> * run bitbake on the recipe, it will fail
> * open terminal to path of src
> * delete Cargo.lock
> * delete toolchain file if present
> * add the toolchain version used in yocto to your path
> * build the crate using this toolchain.  This is to update Cargo.lock to the correct versions (this is a critical step).
> * run my [tool](https://github.com/jwinarske/root-pkg-ws) passing --manifest-path=/Cargo.toml
> * Copy the output from this tool to the recipe file.
> * rebuild and confirm it passes
> 
> If you need a newer Rust toolchain for Kirkstone, you can use: https://git.yoctoproject.org/git/meta-lts-mixins
> 
> Feel free to update the flutter-rust-bridge-example recipe post a PR.
> 
> I will be moving to the upcoming Yocto LTS branch next week, as AGL is moving.

Long time, but now finally I can confirm that these steps are working fine for me. I tried them with an own, minimalistic sample project using flutter_rust_bridge 2.0 on scarthgap.

I'd like to add a PR for frb 2.0, and I have some questions regarding that:
* fzyzcjy-flutter-rust-bridge-frb-example-gallery-rust_0.1.0.bb was removed in 61ff1a5e06f3695a6e7e2fcdfd89b06309b4290f . Was this intentional? If not: how to re-add it in a way that auto-roll won't purge it again?
* The same happened to the membrane recipe by the way ...
* the rust lib currently ends up in /usr/lib/rustlib/aarch64-mydistro-linux-gnu/lib/my_frb_lib.so - which is not found by the flutter part of frb. This is the default behaviour when using "inherit cargo". How would you like that to be solved in the recipse? One solution is to copy it to /usr/lib, but that requires some hacks to get past the QA checks which doesn't feel right for me...

jwinarske, 2025-06-05:
There is support in Yocto starting at (?) that will update crates for a recipe.  Haven’t tried it out yet.

    bitbake -c update_crates recipe-name

Implementation:
https://git.yoctoproject.org/poky/tree/meta/classes-recipe/cargo-update-recipe-crates.bbclass

jwinarske, 2026-09-26:
Closing as not planned.

Nothing rust has landed here in the 19 months this has been open -- no cargo plumbing, no `flutter_rust_bridge` recipe, nothing. That is not an oversight. A rust example app is not one recipe, it is a crate set that has to be re-vendored and re-locked on every engine roll, on top of the rust/Yocto version dance, and the app examples exist to show how to package a Flutter app, not to carry a second language's toolchain maintenance.

The pieces for anyone who wants to do it in their own layer have improved since this was filed. Yocto can regenerate a recipe's crate list:

```
bitbake -c update_crates <recipe>
```

(`cargo-update-recipe-crates.bbclass` in poky). Pair that with `dart-app-pub`, which landed this week and vendors a Dart app's pub dependencies offline, and the shape of a `flutter_rust_bridge` app recipe is not hard to see -- it is the upkeep that is the problem.

If someone shows up with a working recipe and an interest in keeping it green through rolls, I will take it. Filing that as a PR beats reopening this.

closed 2026-09-26 by jwinarske