meta-flutter/meta-flutter #438: [Linux GTK] incoming GTK4

jwinarske, 2024-03-01:
Engine main will soon be merging GTK4 support.

For branches that do not have GTK4 we need to disable desktop package config for flutter-engine.

doesnotexist, 2024-03-01:
I take that means future desktop package will only be enabled in a future scarthgap and subsequent branches, is that right?  Though I suspect scarthgap branch of meta-flutter will beat GTK4 merging into flutter desktop.   
Can kirkstone and earlier branches just freeze their a flutter desktop version prior to GTK4 support?

jwinarske, 2024-03-01:
Yes it would be limited to scarthgap.  For kirkstone and dunfell it will not be available, unless you lock to a specific SDK range; pre GTK4 commit.  It's going into main now, so maybe the weeks.

Once it rolls into stable I will disable building it by default.  The alternative is to augment the build with GTK4 recipe and dependencies.  This in itself is not trivial.

I will be adding a scarthgap branch next week.

Markus43, 2024-03-01:
> For branches that do not have GTK4 we need to disable desktop package config for flutter-engine.

Which brings me to another question: Is it possible to use the gtk embedder together with meta-flutter? I would really be interested in the performance differences between that and the custom embedders.

> I will be adding a scarthgap branch next week.

Scarthgap isn't released yet (April 2024), is it? Just asking because I don't see it in the BSPs that I use yet ...

jwinarske, 2024-03-01:
The GTK embedder is built by the flutter-engine recipe now by default.  The only remaining step is to cross compile the plugins, which needs a flutter tooling patch.

Scarthgap branching from master I'm told will happen this or next week.  AGL has asked me to support it; which will happen next week.

You can do performance analysis on desktop for your use case today, no?

Markus43, 2024-03-01:
> Scarthgap branching from master I'm told will happen this or next week. AGL has asked me to support it; which will happen next week.

I'm all in with supporting most recent versions of the software I use, and I'll switch to scarthgap as soon as I can. I just wanted to point out that it'll still be a few months until it's reasonable to expect BSPs to support scarthgap, leaving users of this layer with no possibility to use your latest updates.

> The GTK embedder is built by the flutter-engine recipe now by default. The only remaining step is to cross compile the plugins, which needs a flutter tooling patch.
> 
> You can do performance analysis on desktop for your use case today, no?

My app basically runs on Linux and Windows Desktop, so basically I can test it there, yes. I am actually thinking about using the GTK embedder in a product, at least as a fallback in case of hw-specific issues with custom embedders: Performance-wise it might be just good enough, too. Even if homescreen is currently the one that performs best in my case ;-)

jwinarske, 2024-03-01:
@Markus43 😎 I'm in the process of finalizing a few plugins before the next major release of Homescreen.  Update includes Plug-in registrar,  Texture registrar, "Platform view" support, headless, etc.  Plugins closely match Windows style, as pigeon supports CPP.  Coming  soon.  WebView (CEF) support post release.  DRM after this.

jwinarske, 2026-08-24:
stale

closed 2026-08-24 by jwinarske