meta-flutter/meta-flutter #448: Python module not found

gianlucacornacchia, 2024-03-08:
Disclaimer:
I'm trying to build yocto 4.2 Mickledore, I know there is no support for this version, but I wonder if someone can help with this error:

File: '/****/sources/meta-flutter/classes/gn-fetcher.bbclass', lineno: 7, function: __anon_9__workspace_*******_sources_meta_flutter_classes_gn_fetcher_bbclass
     0003:# SPDX-License-Identifier: MIT
     0004:#
     0005:
     0006:python () {
 *** 0007:    import gn
     0008:    bb.fetch2.methods.append(gn.GN())
     0009:}
     0010:
     0011:DEPENDS += " \
Exception: ModuleNotFoundError: No module named 'gn'

Building yocto 4.0 kirkstone in the same machine works fine.

Are there any plans to support newer yocto releases ?

Thank you

jwinarske, 2024-03-08:
Yes.  This will happen in the next couple of weeks.  It's the next LTS.

https://github.com/meta-flutter/meta-flutter/issues/439

gianlucacornacchia, 2024-03-08:
Good to know,
thank you

jwinarske, 2024-03-09:
https://github.com/meta-flutter/meta-flutter/tree/master

This should work for Mickledore:
https://github.com/meta-flutter/meta-flutter/blob/master/classes/gn-fetcher.bbclass#L8-L13

jwinarske, 2024-03-13:
proposal is this:
```
python () {
    import os
    # add lib folder to path
    for layer in d.getVar('BBLAYERS').split():
        path = os.path.join(layer, 'lib')
        if 'meta-flutter' in path and os.path.isdir(path):
            sys.path.append(path)
    import gn
    bb.fetch2.methods.append(gn.GN())
}
```
change in gn.py

can you confirm?

jwinarske, 2024-03-30:
@gianlucacornacchia where are we at with this issue?

gianlucacornacchia, 2024-04-02:
sorry for the delay, I was on vacation.

The patch works, but I've applied it to the gn-fetcher.bbclass. 
In this way I've been able to compile flutter for mickledore using the kirkstone branch.
Now I have some issues running the application, but maybe it's not related to flutter.

Do you think it's better to use the master branch for mickledore ?

thank you

jwinarske, 2024-04-13:
@gianlucacornacchia no worries.

The only unknown to using master is if it supports the recently added lines in layer.conf.  Try it out by adding `mickledore` to `LAYERSERIES_COMPAT_flutter-layer`,  if that works I'll merge it to master

closed 2025-06-05 by jwinarske