meta-flutter/meta-flutter #452: Build without `desktop-embeddings` still depends on GTK+

ardera, 2024-03-13:
When building with
`PACKAGECONFIG:pn-flutter-engine:remove = "desktop-embeddings"`, the build still depends on GTK+. Seems like it's because `flutter/testing` is included somehow:

```
ERROR: flutter-engine-git-r0 do_configure: ExecutionError('/yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/temp/run.do_configure.19619', 1, None, None)
ERROR: Logfile of failure stored in: /yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/temp/log.do_configure.19619
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/yocto/spaceship/build/../layers/meta-flutter/recipes-devtools/depot-tools/depot-tools-native_git.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/gcc/gcc-cross_11.4.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_11.4.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-graphics/fontconfig/fontconfig_2.13.1.bb:do_populate_sysroot', '/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-extended/zip/zip_3.0.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-support/ca-certificates/ca-certificates_20211016.bb:do_populate_sysroot', 'virtual:native:/yocto/spaceship/build/../layers/openembedded-core/meta/recipes-support/curl/curl_7.82.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['depot-tools-native', 'glibc', 'gcc-cross-aarch64', 'gcc-runtime', 'quilt-native', 'fontconfig', 'wayland', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'pbzip2-native', 'zip-native', 'ca-certificates-native', 'curl-native', 'libtool-native', 'openssl-native', 'linux-libc-headers', 'zlib-native', 'texinfo-dummy-native', 'mpfr-native', 'gnu-config-native', 'gmp-native', 'xz-native', 'flex-native', 'zstd-native', 'binutils-cross-aarch64', 'libmpc-native', 'attr-native', 'freetype', 'zlib', 'util-linux', 'expat', 'libxml2', 'libffi', 'libgcc', 'bzip2-native', 'perl-native', 'gettext-minimal-native', 'm4-native', 'libpng', 'opkg-utils', 'libxcrypt', 'ncurses', 'libcap-ng', 'util-linux-libuuid', 'libpam', 'bash-completion', 'python3', 'make-native', 'gdbm-native', 'cracklib', 'flex', 'libtirpc', 'libnsl2', 'openssl', 'readline', 'sqlite3', 'bzip2', 'gdbm', 'xz']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| NOTE: FLUTTER_RUNTIME_MODES=debug profile release
| NOTE: CLANG_BUILD_ARCH=x64
| ERROR at //build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
|     pkgresult = exec_script(pkg_config_script, args, "value")
|                 ^----------
| Current dir: /yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/src/out/linux_debug_arm64/
| Command: vpython3 /yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/src/build/config/linux/pkg-config.py -s /yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/recipe-sysroot -a arm64 --system_libdir lib gtk+-3.0
| Returned 1.
| stderr:
| 
| Package gtk+-3.0 was not found in the pkg-config search path.
| Perhaps you should add the directory containing `gtk+-3.0.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'gtk+-3.0' found
| Traceback (most recent call last):
|   File "/yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/src/build/config/linux/pkg-config.py", line 249, in <module>
|     sys.exit(main())
|   File "/yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/src/build/config/linux/pkg-config.py", line 144, in main
|     prefix = GetPkgConfigPrefixToStrip(options, args)
|   File "/yocto/spaceship/build/tmp/work/aarch64-tdx-linux/flutter-engine/git-r0/src/build/config/linux/pkg-config.py", line 82, in GetPkgConfigPrefixToStrip
|     prefix = subprocess.check_output([options.pkg_config,
|   File "/home/builder/.cache/.vpython-root/store/cpython+gc31v00inp54tj1448itdbhvds/contents/lib/python3.8/subprocess.py", line 415, in check_output
|     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
|   File "/home/builder/.cache/.vpython-root/store/cpython+gc31v00inp54tj1448itdbhvds/contents/lib/python3.8/subprocess.py", line 516, in run
|     raise CalledProcessError(retcode, process.args,
| subprocess.CalledProcessError: Command '['pkg-config', '--variable=prefix', 'gtk+-3.0']' returned non-zero exit status 1.
| 
| See //flutter/shell/platform/linux/config/BUILD.gn:8:1: whence it was called.
| pkg_config("gtk") {
| ^------------------
| See //flutter/testing/BUILD.gn:56:18: which caused the file to be included.
|     configs += [ "//flutter/shell/platform/linux/config:gtk" ]
|                  ^------------------------------------------
| Generating GN files in: out/linux_debug_arm64
| WARNING: exit code 1 from a shell command.
ERROR: Task (/yocto/spaceship/build/../layers/meta-flutter/recipes-graphics/flutter-engine/flutter-engine_git.bb:do_configure) failed with exit code '1'
```

jwinarske, 2024-03-13:
Yeah that needs a patch.  Patches based on PKGCONFIG variables need to become a thing.  What branch?

ardera, 2024-03-13:
I'm on kirkstone, though not the latest commit, engine 3.19.2.

I think you can get away with patching it unconditionally though, I have this patch which avoids including `//flutter/testing` when enable_unittests is false:

[0001-Don-t-depend-on-GTK-if-unit-tests-are-disabled.patch.txt](https://github.com/meta-flutter/meta-flutter/files/14591786/0001-Don-t-depend-on-GTK-if-unit-tests-are-disabled.patch.txt)

dunno if that works for all configurations. Also it's a patch to the engine repository, can't be applied to the base checkout directory in that form

jwinarske, 2024-03-13:
@ardera Patches are applied from ${S} independent of any sub repos.  So you can do this:

    flutter-engine -cdo_devshell

Navigate to the location.  Make your change.  Commit the change.  Then run:

    git format-patch -1

Then copy this file to the meta-flutter/recipes-graphics/flutter-engine/files folder.
Append to SRC_URI.
Then in the patch file, adjust the paths to be relative to the root source path.  In most cases this means prefixing path with `flutter/`.

Example (Before)
```
$ cat 0001-gn-riscv32-and-riscv64.patch 
From 1ebbd727d828fa09b7dd65e4f018549c76cd0e48 Mon Sep 17 00:00:00 2001
From: Joel Winarske <joel.winarske@gmail.com>
Date: Wed, 13 Mar 2024 21:09:14 +0000
Subject: [PATCH] gn riscv32 and riscv64

---
 tools/gn | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/tools/gn b/tools/gn
```
After:
```
$ cat 0001-gn-riscv32-and-riscv64.patch 
From 1ebbd727d828fa09b7dd65e4f018549c76cd0e48 Mon Sep 17 00:00:00 2001
From: Joel Winarske <joel.winarske@gmail.com>
Date: Wed, 13 Mar 2024 21:09:14 +0000
Subject: [PATCH] gn riscv32 and riscv64

---
 flutter/tools/gn | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/flutter/tools/gn b/flutter/tools/gn
```
I believe unit tests are currently still enabled despite the gn enable/disable flag.

Adam reported this somewhere a while ago.

@aduskett did you come up with a patch for correctly disabling the unit tests?

aduskett, 2024-03-16:
No, instead upstream fixed it in a much more bandaid way:

https://github.com/flutter/flutter/issues/144421
Also, Wayland is unconditionally required for 3.19.x but I haven't been able to figure out the exact commit that made that change.

jwinarske, 2024-03-18:
@ardera This issue should be fixed with 27db77b99b601e04a9a7de9681eaa7d2a8067927.  Please confirm.

ardera, 2024-03-20:
> ...
> Then in the patch file, adjust the paths to be relative to the root source path.  In most cases this means prefixing path with `flutter/`.
> ...

Ah, yeah that's how I did it before. Though would be really cool if that worked better somehow, the multiple git dirs also break `devtool modify`

Thanks for the quick fix! Can confirm it's working now.

closed 2024-03-20 by jwinarske

aduskett, 2024-03-20:
> @aduskett Likely this one: https://github.com/flutter/engine/blob/main/build_overrides/vulkan_headers.gni

Hmmm, probably not. Or at least, I am not seeing a correlation. Buildroot already runs sed on those lines to set them as true or false depending on whether or not Wayland or X11 are selected. 

See:
https://gitlab.com/buildroot.org/buildroot/-/blob/master/package/flutter-engine/flutter-engine.mk#L155
https://gitlab.com/buildroot.org/buildroot/-/blob/master/package/flutter-engine/flutter-engine.mk#L165

jwinarske, 2024-03-20:
@aduskett Did you flush out angle?
```
third_party/angle/build_overrides/vulkan_common.gni:vulkan_use_wayland = angle_use_wayland
third_party/angle/build_overrides/wayland.gni:wayland_dir = angle_wayland_dir
third_party/angle/build_overrides/wayland.gni:wayland_gn_dir = ""
third_party/angle/src/tests/BUILD.gn:    if (angle_use_wayland) {
# #third_party/angle/BUILD.gn:  import("//build_overrides/wayland.gni")
third_party/angle/gni/angle.gni:    angle_use_wayland =
third_party/angle/gni/angle.gni:        ozone_platform_wayland && !is_ggp && is_linux && !is_castos
third_party/angle/gni/angle.gni:    angle_use_wayland = is_linux && !is_ggp
```
Are you trying to build without wayland and x11/xcb deps; Direct2Display?

jwinarske, 2024-03-20:
@aduskett Looks like build parameters are not correctly being passed to swiftshader.  Let me take a look at the swiftshader tree

jwinarske, 2024-03-20:
@aduskett In swiftshader tip of tree there is support for either DirectFB or Direct2Display (D2D).  It should be the same for flutter-engine swiftshader DEP commit.

```
if(SWIFTSHADER_BUILD_WSI_DIRECTFB)
    find_library(DIRECTFB directfb)
    find_path(DIRECTFB_INCLUDE_DIR directfb/directfb.h)
endif(SWIFTSHADER_BUILD_WSI_DIRECTFB)
if(SWIFTSHADER_BUILD_WSI_D2D)
    find_library(D2D drm)
    find_path(D2D_INCLUDE_DIR libdrm/drm.h)
endif(SWIFTSHADER_BUILD_WSI_D2D)
...
if(LINUX)
    option_if_not_defined(SWIFTSHADER_BUILD_WSI_XCB "Build the XCB WSI support" TRUE)
    option_if_not_defined(SWIFTSHADER_BUILD_WSI_WAYLAND "Build the Wayland WSI support" TRUE)
    option_if_not_defined(SWIFTSHADER_BUILD_WSI_DIRECTFB "Build the DirectFB WSI support" FALSE)
    option_if_not_defined(SWIFTSHADER_BUILD_WSI_D2D "Build the Direct-to-Display WSI support" FALSE)
endif()
```
You just need to add some logic in flutter engine land to set it correctly.

jwinarske, 2024-03-20:
In here:  flutter/third_party/swiftshader/src/Vulkan/BUILD.gn

It skips the desired options all together.

jwinarske, 2024-03-20:
@aduskett Changes would need to be made upstream in the swiftshader repo, then it would eventually get picked up by the flutter engine master.  Until then you would need to patch it.  Might be easier to patch.

aduskett, 2024-03-24:
@jwinarske After some digging, I found the source of the problem:

https://github.com/flutter/buildroot/blob/092e19a7caead03d25b161f33cff6ca520630aac/build/config/BUILDCONFIG.gn#L306

I am fixing the problem by running `sed "s%ozone_platform_x11.*%ozone_platform_x11 = false%g" -i build/config/BUILDCONFIG.gn` and `sed "s%ozone_platform_wayland..*%ozone_platform_wayland. = false%g" -i build/config/BUILDCONFIG.gn` as a pre-configure hook if x11 or wayland are unselected.