meta-flutter/meta-flutter #459: layer name change

kwavnet, 2024-03-25:
In https://github.com/meta-flutter/meta-flutter/commit/07e0773a94a4804ad8efc05c6c7c1bf20268b8ea#diff-22294624f20f807485d85b0f15eac2e1dac979796d77182b94b4896eb103c243R16`(for kirkstone, but master is affected as well) the BBCOLLECTION name was changed to `flutter-layer`.
This broke all of our `BBFILES_DYNAMIC` settings in multiple layers (plus likely, I haven't checked, some `LAYERDEPENDS_`).

As we don't have an mono repo, would it be an option to add

```
BBFILE_COLLECTIONS += "meta-flutter"
BBFILE_PATTERN_meta-flutter = "${BBFILE_PATTERN_flutter-layer}"
LAYERSERIES_COMPAT_meta-flutter = "${LAYERSERIES_COMPAT_flutter-layer}"
```

for a grace period so we can update our layer settings?

jwinarske, 2024-03-27:
Done.  How much time would you like?

b06036b9ecbf45b381d3ab697773da7a195b9d47
2b3889e99f266299f7734da744527329e8f5f48c

jwinarske, 2024-03-27:
Noted

closed 2024-03-27 by jwinarske