meta-flutter/meta-flutter #472: [depot-tools] vpython attempts to break containment

StaticRocket, 2024-04-12:
The `vpython` tool in depot-tools currently reaches out of the build directories into `~/.cache` to setup a download cache among other things.

This behavior can be changed through the use of the `VPYTHON_VIRTUALENV_ROOT` environment variable.

I'm curious what this should be set to. Ideally a common cache an prefetch stage would save some time on each individual recipe, but that has the possibility of creating extra states not actively tracked by bitbake...

jwinarske, 2024-04-13:
@StaticRocket hey there.

The term is `host contamination`.

The location would depend on what recipes are using vpython.  What logs are you seeing vpython in?

StaticRocket, 2024-04-13:
Currently, depot-tools and flutter-engine are running into this. Probably more recipes, but my build is bailing out there and the dependency chain is masking those others.

I didn't think this counts as host contamination because it's not the usual linker lookup issues or weird host dependencies. It's just attempting to use nonsanctioned paths.

jwinarske, 2024-04-13:
I consider host contamination anything that incorrectly uses host state/artifacts; opposed to a specific QA error.  If the build state is not correctly sandboxed from the host, then I call it host contamination.

I suspect it's a one line addition to the gn and cipd fetchers; consumers of depot-tools.  I'll take a look in a bit

jwinarske, 2024-04-13:
@StaticRocket I just pushed change to master.  Let me know if that resolves the issue.  There is an issue with naming folders prefixed with `.` in ${WORKDIR}, in that they don't get cleaned up, otherwise I would have used the same naming (.cache).  Naming it ${WORKDIR}/vpython ensures the folder gets cleaned up on a `-cdo_cleansstate` or `-cdo_cleanall`.

I haven't checked yet if ~/.cache is created when depot-tools-native is built.  Doing that now.

jwinarske, 2024-04-13:
I just confirmed running depot-tools-native does not create `~/.cache/.vpython-root`.

jwinarske, 2024-04-13:
I have confirmed flutter-engine build no longer writes to ~/.cache.  Change has been cherry picked to kirkstone.

Thanks for the report!

closed 2024-04-13 by jwinarske