meta-flutter/meta-flutter #473: [kirkstone] Audioplayers Example Errors

lapumb-spindance, 2024-04-15:
Has anyone been able to get the [audioplayers](https://github.com/meta-flutter/meta-flutter/blob/kirkstone/meta-flutter-apps/recipes-graphics/flutter-apps/third-party/bluefireteam-audioplayers-packages-audioplayers-example_git.bb) example to work / play audio with a Yocto build? I get the following errors when running:

```text
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: <asynchronous suspension>
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: #2      hs._create (oLe:173)
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: <asynchronous suspension>
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method create on channel xyz.luan/audioplayers)
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: #0      MethodChannel._htb (package:flutter/src/services/platform_channel.dart:332)
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: <asynchronous suspension>
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: #1      Ht.KSa (xMe:18)
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: <asynchronous suspension>
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: #2      hs._create (oLe:173)
Apr 15 19:31:45 raspberrypi4-64 flutter-client[1356]: <asynchronous suspension>
```

For what it's worth, I have the following in my `CORE_IMAGE_EXTRA_INSTALL`:

```text
CORE_IMAGE_EXTRA_INSTALL += "\
    weston \
    weston-init \
    \
    wayland \
    wayland-utils \
    \
    flutter-wayland-client \
    \
    \
    dhcpcd \
    wireless-regdb-static \
    iw \
    wpa-supplicant \
    packagegroup-base-wifi \
    \
    openssh \
    ca-certificates \
    kernel-modules \
    \
    libserialport-dev \
    sqlite3 \
    libsqlite3-dev \
    \
    alsa-utils \
    fluent-bit \ 
    \
    gstreamer1.0 \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    gstreamer1.0-plugins-base \
    \
    packagegroup-bluefireteam-audioplayers \
    \
"
```

I am building for / running on the `raspberrypi4-64` ([Raspberry Pi Compute Module 4](https://www.raspberrypi.com/products/compute-module-4/?variant=raspberry-pi-cm4001000) mounted on an [IO board](https://www.raspberrypi.com/products/compute-module-4-io-board/)) and am able to play audio with gstreamer (`gst-launch-1.0 audiotestsrc ! audioconvert ! audioresample ! autoaudiosink`), but get the above exception when attempting to interact with the audioplayers example in any way. 

Any help is appreciated.

jwinarske, 2024-04-15:
That example works on v1/V2 ivi-homescreen when built with correct flag.  V2 will be released when I'm back from my knee surgery recovery

lapumb-spindance, 2024-04-15:
@jwinarske When you say "correct flag" are you referring to when `ivi-homescreen` is built or when the `packagegroup-bluefireteam-audioplayers` (`bluefireteam-audioplayers-packages-audioplayers-example`) is built? Regardless, do you happen to remember the flag?

jwinarske, 2024-04-15:
Actually I don't see it exposed on v1.  You'll have to wait a few weeks for v2

lapumb-spindance, 2024-04-16:
@jwinarske so based on a previous interaction we had, you mentioned that "[_It uses the native platform channel implementation from the embedder. So we ignore the native code in the plug-in projects, and only use what's available in the embedder_](https://github.com/meta-flutter/meta-flutter/issues/340#issuecomment-1811749208)". If I'm understanding correctly, this means that the platform-specific implementation for `audioplayers` is ignored and unless support for `audioplayers_linux` (i.e., `gstreamer`) is added to the `flutter-wayland-client` embedder, the federated plugin is not supported?

Also, I see that there is a [flutter-video-player-plugin](https://github.com/meta-flutter/meta-flutter/blob/master/recipes-graphics/sony/flutter-video-player-plugin_git.bb) for the `flutter-wayland-client`, which appears to add `gstreamer` support. Would this be a valid way of using the `audioplayers` plugin?

jwinarske, 2024-04-16:
@lapumb-spindance  As of today: 
1. There is not a dedicated executable for a given flutter app (flutter + plugin combo), but rather an embedder executable that can run any Flutter app at runtime.
2. Federated plugins are supported by including the plugin native code into the embedder.
3. Yocto Recipes are separated between Flutter code and Native code (embedder)
4. The application is installed in a bundle folder which the embedder executable points to.

Benefits of this approach:
* Scalability. Installing many applications on device will only change with the actual size of the libapp.so file.  This is done by removing all redundant components and libraries.
* Plugin code quality.  Enables locating all native plugin code into a single mono repo. MISRA compliance is much more manageable.
* Licensing is easier to track.  When a plugin is pulled into mono-repo it requires careful scrutiny to mitigate liability for commercial products.
* Cross compilation is more easily managed.  AOT images are cross compiled.  Embedders are cross compiled.
* Leverages custom-device support for debugging.  Can be local or local+target.  See https://github.com/meta-flutter/workspace_automation for more details.

Cons of this approach:
* it doesn't directly use Flutter tooling to implement a hybrid app (Flutter + native plugin code).

This approach and this layer (meta-flutter) is being used in many commercial projects.

Regarding ivi-homescreen v2:
* implements a full featured desktop plugin and texture registrar (as used by the Windows embedder)
* Native plugin code follows Pigeon CPP standard (Windows).  Very easy to port between Windows plugins and ivi-homescreen v2 plugins.  Only variation is Linux vs Windows APIs.
* Performance is still better than the Flutter GTK embedder, and supports more features and backends out of the box.

I hope this help to clarify things.

lapumb-spindance, 2024-04-16:
@jwinarske that was a very helpful breakdown - I greatly appreciate it. I look forward to seeing the ivi-homescreen v2, but in the meantime will be adding the `audioplayers_linux` platform code to the `flutter-wayland-client` embedder. Thank you!

jwinarske, 2024-05-13:
@lapumb-spindance 
ivi-homescreen v2.0 plugin functional since December is now available:  https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/audioplayers_linux
Yotco recipe for v2 will be pushed today.

closed 2024-05-13 by jwinarske