meta-flutter/meta-flutter #474: Text no longer displayed in flutter app using flutterpi embedder and nanbield or scarthgap releases

resilem, 2024-04-17:
Hello,

I have an image that uses flutterpi embedder and displays the flutter material 3 demo app on a RPI CM4. I have been using this image successfully with kirkstone branch for the last 2 months. I am experimenting with transitioning to next LTS release, scarthgap using the master branch. I am finding that all text within the flutter app is no longer being displayed. This happens with any flutter app I attempt to run, not just the material 3 demo.

Here is the flutter-pi output when ran from kirkstone system:
```
root@raspberrypi4-64:~# flutter-pi --release /usr/share/flutter/flutter-samples-material-3-demo/3.19.2/release/
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
keyboard.c: Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
flutter: {}
```

Here is the flutter-pi output when ran from scarthgap system:
```
root@raspberrypi4-64:~# flutter-pi --release /usr/share/flutter/flutter-samples-material-3-demo/3.19.3/release/
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
/modesetting.c: Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
/flutter-pi.c: Could not create drmdev from device at "/dev/dri/card1". Continuing.
/keyboard.c: Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
flutter: {}
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
[ERROR:flutter/flow/layers/transform_layer.cc(23)] TransformLayer is constructed with an invalid matrix.
```

I have not experimented with other embedders, it is not clear to me if this is an issue that is embedder specific or not. Any suggestions on how to approach this issue or debug it further would be appreciated.

jwinarske, 2024-04-18:
@resilem I can run a test tomorrow with ivi-home screen v2 + flutter-samples-material-3-demo on a RPI5.  

I do know gallery works as expected on a RPI5 running scarthgap + ivi-homescreen v2.

When you transitioned to scarthgap did you start with a clean shared state and DL_DIR?

jwinarske, 2024-04-19:
@resilem
Just ran test with scarthgap + ivi-homescreen v2 + flutter-samples-material-3-demo + Raspberry Pi 5

[image]

Start with:
```
bitbake flutter-samples-material-3-demo -cdo_cleanall
bitbake flutter-samples-material-3-demo
```

Any embedder should be able to run this sample.  It's also possible you are missing a runtime dependency which my image/recipe combo is implicitly installing

jwinarske, 2024-04-19:
cc @ardera

resilem, 2024-04-19:
I did not get a different result using `bitbake flutter-samples-material-3-demo -cdo_cleanall`. Going to do full rebuild with new sstate next.

resilem, 2024-04-20:
A full rebuild with a new sstate and targeting scarthgap results in the same behavior, no text displayed in the application. Will give a full clean nanbield build a shot next.

ardera, 2024-04-21:
I'll try reproducing it. Does the issue also happen when both ivi-homescreen and flutter-pi are present in the image?

resilem, 2024-04-21:
I am seeing same result w/ nanbield as well. Only flutterpi is present in the image, ivi-homescreen is not.

resilem, 2024-04-21:
Here is the image recipe.
```
inherit core-image

IMAGE_FEATURES:append = " hwcodecs kernel-modules flutter-pi flutter-samples-material-3-demo"
```
I am using `poky` distro and making a few tweaks to `DISTRO_FEATURES`.
```
DISTRO_FEATURES:append = " pam kms"
DISTRO_FEATURES:remove = "x11"
```

I am also seeing this behavior with a rpi3.

ardera, 2024-04-22:
@resilem This is just a bare flutter-pi build right? So no wayland either?

I think there just might be no fonts present on the image. With ivi-homescreen, they're present because weston depends on them.

@jwinarske I'll make a PR to that adds a runtime recommendation for `liberation-fonts` to the flutter-pi recipe.

resilem, 2024-04-22:
Wayland is present in `DISTRO_FEATURES`.

resilem, 2024-04-22:
I have confirmed that adding:
```
RRECOMMENDS:${PN} += "\
    liberation-fonts \
    "
```
to `flutter-pi_git.bb` resolves the issue, text is displayed again.

Thanks @ardera & @jwinarske

jwinarske, 2024-04-22:
merged to master
https://github.com/meta-flutter/meta-flutter/commit/9c20d1a9502c0eb4bb8e93436e73b6ffe9f4ebd2

cherry picked to scarthgap->kirkstone

closed 2024-04-22 by jwinarske