meta-flutter/meta-flutter #475: Federated Plugin not loaded in Release buids, works well with Debug

matsch82, 2024-04-18:
It seems like the bug https://github.com/meta-flutter/meta-flutter/issues/115 is again present in kirkstone branch with 3.19.3 flutter.

I sucessfully integrated the sony elinux camera plugin, it is loaded properly when running with debug, but is not used when running release version:
`LD_LIBRARY_PATH=/usr/share/flutter/3.19.3/debug/lib/ flutter-drm-gbm-backend -b /flutter/cam_example/3.19.3/release/` works while 
`LD_LIBRARY_PATH=/usr/share/flutter/3.19.3/release/lib/ flutter-drm-gbm-backend -b /flutter/cam_example/3.19.3/release/` fails to load the dart platform implementation and does some fallback.

The Plugin uses the following pubspec declaration:
```yaml
dependencies:
  camera_platform_interface: ^2.5.0
  flutter:
    sdk: flutter

flutter:
  plugin:
    platforms:
      linux:
        pluginClass: CameraElinuxPlugin
        dartPluginClass: ELinuxCamera
```

jwinarske, 2024-04-18:
@matsch82 it certainly works in master (scarthgap), as that has been the recent focus.  I've been cherry picking into kirkstone.  

That said two options: 
1.  Provide the compile log for your app.  You can change path and app names if you like
2.  Attempt repro with one of the 1st party examples, without any changes to the app.  File selector has been my latest go-to test for validating this.

I'll test ivi-home screen v2 with file selector on a kirkstone image tomorrow.

One thing you can try pretty quickly is add back the obfuscate flag in gen_snapshot step.  Pretty certain I ruled this out a while back, but worth a try

jwinarske, 2024-04-19:
I am not able to repro this on kirkstone tip of tree.  Also be sure to run `-cdo_cleanall` on the flutter app if you have been moving commits in the past couple of months.

matsch82, 2024-04-24:
Unfortunately i cannot use the kirkstone branch directly, since i have to patch a few things to make it work on my hardware platform (xilinx) and build environemt (petalinux).
I did a full rebuild of the whole system and it is still there for the camera_plugin, but not for the video_player_plugin. 
I added a debug-print to the constructor of the elinux camera plugin (basically here: https://github.com/sony/flutter-elinux-plugins/blob/main/packages/camera/lib/src/elinux_camera.dart#L23) to see if the constructor is called.
For debug builds it is called, but not for release builds.

I added the build logs of the embedder and the app in the zip file. Obfuscation seems to be enabled for release. 
But don't know how to debug if further, since i have not fully understood the root cause. And release builds are hard to debug.
For me a workarround would be sufficient.. it one exists.
[logs.zip](https://github.com/meta-flutter/meta-flutter/files/15090952/logs.zip)

jwinarske, 2024-05-13:
@matsch82 I'm thinking it has to do with the Sony code.  It implements a custom plugin type, no?  Opposed to a linux, windows, etc.

For instance with ivi-homescreen v2 I have a video player plugin:
https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/video_player_linux
which uses a federated linux plugin:
https://github.com/meta-flutter/video_player_linux/blob/main/example/pubspec.yaml

So I'm thinking there are some changes required in SDK to support the Sony plugins.

I also have a federated Camera plugin (libcamera) work in progress here:
https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/camera

matsch82, 2024-06-25:
@jwinarske the sony code implements a custom platform type "elinux", which is imho not needed.
But even if I remove all this specific stuff, it still shows some strange behaviour.
I recently added a third plugin (camera, video and extrenal texture so far), now the camera plugin is also broken in debugging builds. All other plugins are working. 

I try to update to your release 3.22 now. Lets see what this brings.

matsch82, 2024-07-23:
@jwinarske 
I finally found a way to deal with this issue  in our project.  But never the less, it is a bug in the flutter/dart engine somewhere.

The setup was the following:
Custom Cameara Dart Plugin --> Camera Platform Interface  --> Custom Camera Native Plugin  

It turns out, that the Custom Camera Plugin is not initialized by the dart VM in case of aarch64 release builds with libflutter_engine in the release version. libflutter_engine in debug initialized the plugin.
For x86 everything works as expected.
The workaround is to initialize it manually in the flutter app, if we detect linux platform.
Basically this:
`void main() {
  if(Platform.isLinux) {
    ELinuxCamera.registerWith();
  }
...}
  `
  so from my perspecive, this ticket can be closed, since we have workarround at hand.

jwinarske, 2024-08-05:
I see.  I did add support for that quite a while back.  All the functional test cases I've ran are release, and they work as expected on many platform variants.

closed 2024-08-05 by jwinarske