meta-flutter/meta-flutter #481: Build standalone firebase-cpp-sdk for yocto linux

chadrockey, 2024-04-27:
Is there a way to build just the firebase-cpp-sdk recipe in isolation?

My builds fail in quite a variety of ways, first is 

`The imported target "flatbuffers::flatc" references the file
| 
|      "............/build/tmp/work/cortexa73-cortexa53-poky-linux/firebase-cpp-sdk/11.6.0-r0/recipe-sysroot/usr/bin/flatc"

but this file does not exist.  Possible reasons include:
`

chadrockey, 2024-04-27:
Ah embarrassing, one of the layers my bsp provider included already had flatbuffers, but I needed the original from kirkstone meta open embedded, so I had to include the recipe in my top level layer as well and then this built great.

Only change I needed was allow empty as I make an embedded hardware SDK tool (think docker but just silicon) and install the libraries into the targets.

```
ALLOW_EMPTY:${PN} = "1"
ALLOW_EMPTY:${PN}-dev = "1"
```

Have you considered submitting this upstream to open embedded?

chadrockey, 2024-04-27:
Made good progress and got up to authenticating user and password.

Any idea why it seems like firestore_core isn't present?  What library would need linked?

`undefined reference to `firebase::firestore::util::Status::State::Deleter`

jwinarske, 2024-05-01:
@chadrockey Hey.  Sorry I'm been buried.  The latest Firebase integration is in an internal Toyota branch (ivi-homescreen `v2`) to be pushed maybe next week.  It has full featured Firebase plugins that work.

I have no plans to upstream anything at this time.  Upstream could be improved so that I don't need such hacky patches.  My intent was to focus on that, as it would decrease effort to roll upstream here.

chadrockey, 2024-05-02:
I think that sounds like great news @jwinarske !  I'll look out for the release!

Since most of yocto trends toward using static libraries for efficiency, could you teach me a little more about the need for using the system libraries?  Even if firebase built its own dependencies, it does that on yocto anyway kind of, so statically linking them in isn't worst, especially when 99% of the time the static firebase library would then be statically linked to the executable.

I really enjoy using firebase and think it's a nice tool, but I've constantly lived in the shadows of where it's not supported well.  Unity builds, c#, c++.  I can definitely help upstream firebase-cpp-sdk to open embedded if we can make it work.  The python versions are there, so I think it would be worthwhile and put pressure on Google to see that c++ should be better supported.

jwinarske, 2024-05-13:
@chadrockey 
https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/firebase_auth
https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/firebase_core
https://github.com/toyota-connected/ivi-homescreen/tree/v2.0/plugins/firebase_storage

closed 2024-05-13 by jwinarske

chadrockey, 2024-05-13:
Thanks @jwinarske!  I'll start looking.