meta-flutter/meta-flutter #483: `do_archive_pub_cache` runs before `do_patch`

edrose, 2024-05-03:
How come the task `archive_pub_cache` runs before `do_patch`? I'm struggling to understand the reasoning here.

I've come across two senarios where this causes problems:

* Wanting to apply a patch to a recipe that changes the dependencies. Since the pub cache doesn't match the (patched) source code, the build can fail
* Trying to use `devtool` to develop the app. If dependencies change during development, the build will fail since it's using the old pub cache. If the dependencies are broken on the commit you want to checkout, devtool fails to even checkout the code so you can't even change it

Should it not run after `do_configure`? It works fine with the few examples I've tested, but does it cause issues anywhere else?

edrose, 2024-05-03:
I've just found a third senario. Running the `clean` or `cleanall` task won't invalidate the pub cache with the current configuration since it doesn't re-run `archive_pubcache`. For a true clean build you need to run `bitbake -c cleanall <recipe>` followed by `bitbake -f -c archive_pubcache <recipe>` to force bitbake to rebuild the pub cache, otherwise the old one is used.

This was causing me problems when trying to debug #484

jwinarske, 2024-05-13:
> How come the task `archive_pub_cache` runs before `do_patch`? I'm struggling to understand the reasoning here.

The reasoning was archiving the pub cache is a higher priority than patching; the goal being to archive the source content as-is.  Matching the standard Yocto pattern; fetch and patch.

Like I mentioned in other message, a pub spec fetcher is imminent.  Which will change the current work, and remove the notion of a pub spec archive.  The recipe then becomes the origin of truth (lock file; as it should), and when you change a patch that impacts deps, the recipe will need to be updated.

Running bitbake -c cleanall for me removes the archive from the DL_DIR, followed by a subsequent build `(bitbake <recipe>` fetches fresh archive.  I never find I need to use force (-f).

closed 2024-05-13 by jwinarske