meta-flutter/meta-flutter #484: Flutter apps with git repositories in `pubspec.yaml` fail to build

edrose, 2024-05-03:
Flutter apps with dependencies specified as git repositories fail to build with the current `flutter-app` class. This wasn't the case in the past. There are two main changes that have caused this regression.

**1. `pubspec.lock` is deleted during the build processs**

From what I've observed whilst trying to debug this, the command `flutter pub get --offline` behaves differently if there is a `pubspec.lock` file compared to when there isn't. If a `pubspec.lock` file exists, it simply checks the version in the cache against the version in the lockfile.

However if a lockfile does not exist, it tries to perform a `git fetch` on the repository, even when the `--offline` flag is specified. So by deleting the lockfile it forces flutter to go online, despite the `--offline` flag. This may well be an upstream bug.

**2. `do_compile[online] = "1"` was removed**

At some point the `do_compile` task was changed to block all network access. With the above issue where `--offline` is not really offline, `flutter pub get --offline` fails due to it's inability to perform `git fetch`.

---

I don't think `pubspec.lock` should be deleted as part of the build process. I personally would go the other way, and use `flutter pub gett --enforce-lockfile` when a lockfile is present. This ensures that the version being built is _exactly_ the same as it was checked into version control, and prevents flutter from upgrading packages during the build process.

Not all flutter apps will have a lockfile, so the flag can only be applied if it exists. However removing the lockfile completely causes the issues above and also makes the builds less reproducible.

jwinarske, 2024-05-13:
@edrose Hi Ed,

There are a number of corner cases that deleting the lockfile resolves.  The primary one is hitting build issues when rolling 100+ recipes.  So for the time being the lock file needs to be deleted.

That said the correct fix for this is a pub spec fetcher.  In the AGL production ready meeting we working on this problem.  The primary issue right now is how to correctly implement SPDX support; something that is not yet in Yocto Rust support.

jwinarske, 2024-05-13:
resovled with https://github.com/meta-flutter/meta-flutter/pull/485

closed 2024-05-13 by jwinarske