meta-flutter/meta-flutter #486: PUB_CACHE_ARCHIVE fails for local sources

kwavnet, 2024-05-08:
The change introduced by #482 fails in one of our corner case builds.
The recipe in question look like this

```
SRC_URI = "\
    file://foo.tar.bz2 \
"
S = "${WORKDIR}/foo"
PUBSPEC_APPNAME = "goo"
inherit flutter-app
```

so basically a snapshot of a flutter project.
In this particular case `SRCPV` is empty, as `SRCREV` is set to `INVALID` as no revision information is available.

Leading to the following error
```
2024-05-06T16:48:24.6306546Z File: '.../build/sources/meta-flutter/conf/include/common.inc', lineno: 160, function: do_archive_pub_cache
2024-05-06T16:48:24.6309788Z      0156:    bb.note("Running [%s] in %s" % (pack_cmd, pub_cache))
2024-05-06T16:48:24.6311681Z      0157:    run_command(d, pack_cmd, pub_cache, env)
2024-05-06T16:48:24.6336021Z      0158:
2024-05-06T16:48:24.6337293Z      0159:    if not os.path.exists(localpath):
2024-05-06T16:48:24.6339709Z  *** 0160:        raise FetchError("The fetch command returned success for pub cache, but %s doesn't exist?!" % (localpath), localpath)
2024-05-06T16:48:24.6341646Z      0161:
2024-05-06T16:48:24.6342719Z      0162:    if os.path.getsize(localpath) == 0:
2024-05-06T16:48:24.6344029Z      0163:        os.remove(localpath)
2024-05-06T16:48:24.6346426Z      0164:        raise FetchError("The fetch of %s resulted in a zero size file?! Deleting and failing since this isn't right." % (localpath), localpath)
2024-05-06T16:48:24.6352257Z Exception: bb.fetch2.FetchError: Fetcher failure for URL: '.../downloads/flutter-pub-cache-foo-${SRCPV}-3.19.3.tar.bz2'. The fetch command returned success for pub cache, but .../downloads/flutter-pub-cache-foo-${SRCPV}-3.19.3.tar.bz2 doesn't exist?!
```

To me it looks like `${SRCPV}` isn't expanded by the operation, hence creating an invalid path name.
In before the problem was shadowed by `SRCREV` being `INVALID`, which then created a pub cache blob without the proper version info.

A potential fix could be
```
_REV = "${@d.getVar('SRCPV') or d.getVar('PV')}"
PUB_CACHE_ARCHIVE = "flutter-pub-cache-${PN}-${_REV}-${FLUTTER_SDK_TAG}.tar.bz2"
```

jwinarske, 2024-05-13:
I believe this is a duplicate of https://github.com/meta-flutter/meta-flutter/issues/484 which will be resolved by https://github.com/meta-flutter/meta-flutter/pull/485

jwinarske, 2024-05-13:
Agreed.  Thanks for digging in this.  I'll push a change.

closed 2024-05-13 by jwinarske