meta-flutter/meta-flutter #492: flutter-auto missing DEPENDS += "curl"

pogojotz, 2024-05-16:
I have the following error during `do_configure`:
```
ERROR: flutter-auto-2.0-r0 do_configure: ExecutionError('/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/temp/run.do_configure.5506', 1, None, None)
ERROR: Logfile of failure stored in: /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/temp/log.do_configure.5506
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/work/build/../src/meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/work/build/../src/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot', '/work/build/../src/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', '/work/build/../src/meta-raspberrypi/recipes-multimedia/rpidistro-ffmpeg/rpidistro-ffmpeg_4.3.4.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-core/glib-2.0/glib-2.0_2.78.4.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-core/glibc/glibc_2.39.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-devtools/cmake/cmake-native_3.28.3.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-devtools/gcc/gcc-cross_13.2.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-devtools/gcc/gcc-runtime_13.2.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-devtools/gcc/libgcc_13.2.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-gnome/libsecret/libsecret_0.21.4.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-graphics/mesa/mesa_24.0.3.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-graphics/wayland/wayland-protocols_1.33.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-graphics/wayland/wayland_1.22.0.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-graphics/xorg-lib/libxkbcommon_1.6.0.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22.11.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-good_1.22.11.bb:do_populate_sysroot', '/work/build/../src/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0_1.22.11.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/ninja/ninja_1.11.1.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-graphics/wayland/wayland_1.22.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['clang-cross-aarch64', 'compiler-rt', 'libcxx', 'v4l-utils', 'rpidistro-ffmpeg', 'avahi', 'bluez5', 'mobile-broadband-provider-info', 'ofono', 'openssl', 'base-files', 'base-passwd', 'dbus', 'ell', 'expat', 'gettext-minimal-native', 'glib-2.0', 'glibc', 'libxcrypt', 'libxml2', 'ncurses', 'readline', 'systemd', 'util-linux-libuuid', 'util-linux', 'zlib', 'binutils-cross-aarch64', 'cmake-native', 'gcc-cross-aarch64', 'gcc-runtime', 'libgcc', 'libedit', 'libtool-native', 'libtool', 'm4-native', 'opkg-utils', 'orc', 'python3', 'quilt-native', 'vala', 'bzip2', 'libnsl2', 'libtirpc', 'shadow-sysroot', 'shadow', 'texinfo-dummy-native', 'xz', 'zstd', 'gdk-pixbuf', 'gobject-introspection', 'libgudev', 'libsecret', 'cairo', 'libdrm', 'fontconfig', 'freetype', 'harfbuzz', 'libjpeg-turbo', 'mesa', 'pango', 'vulkan-headers', 'vulkan-loader', 'wayland-protocols', 'wayland', 'libpciaccess', 'libpthread-stubs', 'libxkbcommon', 'pixman', 'util-macros', 'kmod', 'linux-libc-headers', 'alsa-lib', 'flac', 'gstreamer1.0-plugins-base', 'gstreamer1.0-plugins-good', 'gstreamer1.0', 'lame', 'libogg', 'libpng', 'libsndfile1', 'libtheora', 'libvorbis', 'mpg123', 'pulseaudio', 'sbc', 'speex', 'speexdsp', 'x264', 'acl', 'attr', 'bash-completion', 'fribidi', 'gdbm', 'icu', 'iso-codes', 'libatomic-ops', 'libcap-ng', 'libcap', 'libdaemon', 'libffi', 'libgcrypt', 'libgpg-error', 'libical', 'libpcre2', 'libpsl', 'libseccomp', 'libsoup', 'nghttp2', 'shared-mime-info', 'sqlite3', 'taglib', 'utfcpp', 'clang-native', 'openssl-native', 'expat-native', 'libxml2-native', 'ncurses-native', 'util-linux-libuuid-native', 'zlib-native', 'binutils-native', 'elfutils-native', 'flex-native', 'gnu-config-native', 'libedit-native', 'make-native', 'ninja-native', 'patch-native', 'perl-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 'swig-native', 'bzip2-native', 'libarchive-native', 'libidn2-native', 'libnsl2-native', 'libtirpc-native', 'shadow-native', 'xz-native', 'zstd-native', 'wayland-native', 'acl-native', 'attr-native', 'curl-native', 'gdbm-native', 'gmp-native', 'gnutls-native', 'libtasn1-native', 'libbsd-native', 'libcap-native', 'libffi-native', 'libgcrypt-native', 'libgpg-error-native', 'libmd-native', 'libmicrohttpd-native', 'libmpc-native', 'libpcre2-native', 'libunistring-native', 'mpfr-native', 'nettle-native', 're2c-native', 'sqlite3-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- CMAKE_BUILD_TYPE not set, defaulting to Release.
| -- The CXX compiler identification is Clang 18.1.5
| -- The C compiler identification is Clang 18.1.5
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/bin/aarch64-taclos-linux/aarch64-taclos-linux-clang++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/bin/aarch64-taclos-linux/aarch64-taclos-linux-clang - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- GIT Branch ............. v2.0
| -- GIT Hash ............... af29a9b
| -- LLVM Version ........... 18.1.5
| -- LLVM Root .............. /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr
| -- C++ header path ........ /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/include/c++/v1/
| -- Clang CMake Path ....... /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/lib/cmake/llvm
| -- IPO .................... supported
| -- Build Type ............. LOCAL
| -- Build Documentation .... OFF
| -- Build Unit Tests ....... OFF
| -- Generate Golden Images.. OFF
| -- Project ................ ivi-homescreen
| -- Version ................ 1.0.0
| -- Generator .............. Ninja
| -- Build Type ............. Release
| -- APPEND RELEASE FLAGS ...  -fstack-protector-all
| -- APPEND RELEASE FLAGS ...  -fno-omit-frame-pointer
| -- APPEND RELEASE FLAGS ...  -Wformat=2
| -- APPEND RELEASE FLAGS ...  -D_FORTIFY_SOURCE=2
| -- CC ..................... /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/bin/aarch64-taclos-linux/aarch64-taclos-linux-clang
| -- CXX .................... /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot-native/usr/bin/aarch64-taclos-linux/aarch64-taclos-linux-clang
| -- CFLAGS ................. -target aarch64-taclos-linux  -mcpu=cortex-a53+crc+nocrypto -mbranch-protection=standard -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types   -fmacro-prefix-map=/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/git=/usr/src/debug/flutter-auto/2.0  -fdebug-prefix-map=/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/git=/usr/src/debug/flutter-auto/2.0  -fmacro-prefix-map=/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/build=/usr/src/debug/flutter-auto/2.0  -fdebug-prefix-map=/work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/build=/usr/src/debug/flutter-auto/2.0  -fdebug-p [... cut ...]
| 	#2: do_configure, /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/temp/run.do_configure.5506, line 141
| 	#3: main, /work/build/tmp/work/cortexa53-taclos-linux/flutter-auto/2.0/temp/run.do_configure.5506, line 180
ERROR: Task (/work/build/../src/meta-flutter/recipes-graphics/toyota/flutter-auto_2.0.bb:do_configure) failed with exit code '1'
```

Adding `DEPENDS += "curl"` in an .bbappend fixed it.

In my setup, I have deactivated "file_selector" PACKAGECONFIG. So I assume curl was pulled in as a dependency of Zenity and that is why this slipped through the cracks.

jwinarske, 2024-05-17:
@pogojotz Hi there.  Expect some churn over the next week.  See if https://github.com/meta-flutter/meta-flutter/pull/493 resolves your scenario.

jwinarske, 2024-05-19:
I refactored the plugin_common which is pulled in with 17d166527c4c02f5b6188d60f3ab56ec5b25d598

closed 2024-05-19 by jwinarske