meta-flutter/meta-flutter #498: PACKAGECONFIG requirements of ivi-homescreen 2.0

WAKazuyoshiAkiyama, 2024-05-30:
What are the available combination of PACAKGECONFIG of ivi-homescreen?
When I changed PACKAGECONFIG from the default, some errors happen.
So, I would like to get know the available combination of PACAKGECONFIG.

---

# ERROR1

ivi-homescreen seems not to be built without `pdf` PACKAGECONFIG.
Without `pdf`, the following error will be caused.

```
| CMake Error: The source directory "/workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/build/pdfium" does not exist.
```

```
ERROR: ivi-homescreen-2.0-r0 do_configure: ExecutionError('/workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/run.do_configure.1234254', 1, None, None)
ERROR: Logfile of failure stored in: /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/log.do_configure.1234254
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/workdir/poky/meta/recipes-core/glib-2.0/glib-2.0_2.72.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/gcc/gcc-cross_11.4.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/gcc/gcc-runtime_11.4.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-graphics/mesa/mesa_22.0.3.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-graphics/wayland/wayland-protocols_1.25.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot', '/workdir/poky/meta/recipes-graphics/xorg-lib/libxkbcommon_1.4.1.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/workdir/poky/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['glib-2.0', 'glibc', 'cmake-native', 'gcc-cross-x86_64', 'gcc-runtime', 'quilt-native', 'mesa', 'wayland-protocols', 'wayland', 'libxkbcommon', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'wayland-native', 'expat', 'libxext', 'libdrm', 'libx11', 'xorgproto', 'libxdamage', 'zlib', 'xrandr', 'libxfixes', 'libxshmfence', 'libxxf86vm', 'libxcb', 'xkeyboard-config', 'libxml2', 'libgcc', 'linux-libc-headers', 'libffi', 're2c-native', 'zstd-native', 'mpfr-native', 'flex-native', 'gnu-config-native', 'zlib-native', 'libtool-native', 'binutils-cross-x86_64', 'xz-native', 'texinfo-dummy-native', 'gmp-native', 'libmpc-native', 'curl-native', 'bzip2-native', 'ncurses-native', 'attr-native', 'python3-native', 'libxml2-native', 'expat-native', 'libffi-native', 'libpcre', 'bash-completion', 'util-linux', 'util-macros', 'libpciaccess', 'libpthread-stubs', 'autoconf-archive', 'xtrans', 'libxrender', 'libxrandr', 'libxau', 'libxdmcp', 'xcb-proto', 'python3', 'gettext-minimal-native', 'm4-native', 'openssl-native', 'readline-native', 'libtirpc-native', 'gdbm-native', 'libnsl2-native', 'sqlite3-native', 'util-linux-libuuid-native', 'bzip2', 'libxcrypt', 'ncurses', 'util-linux-libuuid', 'libcap-ng', 'opkg-utils', 'readline', 'gdbm', 'libnsl2', 'sqlite3', 'xz', 'openssl', 'libtirpc', 'perl-native', 'make-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| CMake Error: The source directory "/workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/build/pdfium" does not exist.
| Specify --help for usage, or press the help button on the CMake GUI.
| WARNING: /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/run.do_configure.1234254:162 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DFETCHCONTENT_FULLY_DISCONNECTED=ON -D CMAKE_SYSROOT=/workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/recipe-sysroot/usr -D ENABLE_LTO=ON -D BUILD_UNIT_TESTS=OFF -D BUILD_DOCS=OFF -DBUILD_PLUGIN_AUDIOPLAYERS_LINUX=OFF -DBUILD_BACKEND_WAYLAND_DRM=OFF -DBUILD_BACKEND_WAYLAND_EGL=ON -DBUILD_BACKEND_WAYLAND_VULKAN=OFF -DBUILD_PLUGIN_CAMERA=OFF -DENABLE_AGL_CLIENT=OFF -DENABLE_IVI_SHELL_CLIENT=OFF -DENABLE_XDG_CLIENT=ON -DBUILD_PLUGIN_CLOUD_FIRESTORE=OFF -DBUILD_PLUGIN_DESKTOP_WINDOW_LINUX=OFF -DENABLE_DLT=OFF -DBUILD_EGL_ENABLE_3D=ON -DBUILD_EGL_ENABLE_MULTISAMPLE=ON -DBUILD_EGL_TRANSPARENCY=ON -DBUILD_PLUGIN_FILAMENT_VIEW=OFF -DBUILD_PLUGIN_FILE_SELECTOR=OFF -DBUILD_PLUGIN_FIREBASE_AUTH=OFF -DBUILD_PLUGIN_FIREBASE_STORAGE=OFF -DBUILD_PLUGIN_GO_ROUTER=OFF -DBUILD_PLUGIN_GOOGLE_SIGN_IN=OFF -DBUILD_PLUGIN_LAYER_PLAYGROUND_VIEW=OFF -DBUILD_PLUGIN_NAV_RENDER_VIEW=OFF pdfium -DSANITIZE_ADDRESS=OFF -DBUILD_PLUGIN_SECURE_STORAGE=OFF -DBUILD_CRASH_HANDLER=OFF -DBUILD_PLUGIN_URL_LAUNCHER=OFF -D DEBUG_PLATFORM_MESSAGES=OFF -DBUILD_PLUGIN_VIDEO_PLAYER_LINUX=OFF -DBUILD_PLUGIN_WEBVIEW_FLUTTER_VIEW=OFF -Wno-dev'
| WARNING: Backtrace (BB generated script):
|       #1: cmake_do_configure, /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/run.do_configure.1234254, line 162
|       #2: do_configure, /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/run.do_configure.1234254, line 141
|       #3: main, /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/temp/run.do_configure.1234254, line 198
ERROR: Task (/workdir/meta-flutter/recipes-graphics/toyota/ivi-homescreen_2.0.bb:do_configure) failed with exit code '1'
```

---

# ERROR2

And also, if adding `pdf` in PACKAGECONFIG, the error of missing `curl.h` will be caused.

```
| In file included from /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/git/plugins/common/tools/../common.h:20,
|                  from /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/git/plugins/common/tools/command.cc:19:
| /workdir/build/tmp/work/core2-64-poky-linux/ivi-homescreen/2.0-r0/git/plugins/common/tools/../curl_client/curl_client.h:27:10: fatal error: curl/curl.h: No such file or directory
|    27 | #include <curl/curl.h>
|       |          ^~~~~~~~~~~~~
| compilation terminated.
```
(a lot of same errors happen)

This error can be avoid by adding `curl` into `DEPENDS`.

jwinarske, 2024-06-12:
what source commit are you using?

jwinarske, 2024-06-12:
Also what is the package config you are using, or want to use?
`-DDISABLE_PLUGINS=ON` will disable all plugins in the plugin folder, not required system items like keyboard and mouse cursor

WAKazuyoshiAkiyama, 2024-07-12:
@jwinarske 

I used `a130deb48a7ccaa8e4a082ad14572490d69d8f41` of meta-flutter.
ivi-homescreen revision is `c72ce63289e386bcec3c90c03ccfde00271d7d0f`

WAKazuyoshiAkiyama, 2024-07-12:
I have not modified `PACKAGECONFIG` from the default.