meta-flutter/meta-flutter #503: pub upgrade build issue in combination with `DL_DIR` when building flutter-ics-homescreen

mv0, 2024-06-21:
So I'm hitting an odd build issue (this is with Debian testing) on the AGL platform. Tried this with a few other targets, among rpi5, rpi4 and rcar3 (h3ulcb). This is with master, been hitting for few weeks now (guess I that's about 2, maybe 3 weeks when I've first noticed it) 

The build failure happens at `do_compile` stage, failing at `pub upgrade` set (from `flutter config --no-cli-animations`):

```
ERROR: flutter-ics-homescreen-git-r0 do_compile: SystemExit("failed 1 (cmd was flutter config --no-cli-animations):\nBuilding flutter tool...\nResolving dependencies...\nGot 
socket error trying to find package flutter_template_images at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Retrying in five seconds... (9 tries left)\nReso
lving dependencies...\nGot socket error trying to find package archive at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Retrying in five seconds... (8 tries 
left)\nResolving dependencies...\nGot socket error trying to find package dwds at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Retrying in five seconds... (
7 tries left)\nResolving dependencies...\nGot socket error trying to find package archive at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Retrying in five s
econds... (6 tries left)\nResolving dependencies...\nGot socket error trying to find package args at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Retrying i
n five seconds... (5 tries left)\nResolving dependencies...\nGot socket error trying to find package file at https://pub.dev.\nError: Unable to 'pub upgrade' flutter tool. Re
trying in five seconds... (4 tries left)\nResolving dependencies...\nGot socket error trying to find package args at https://pub.dev.\nError: Unable to 'pub upgrade' flutter 
tool. Retrying in five seconds... (3 tries left)\nResolving dependencies...\nGot socket error trying to find package archive at https://pub.dev.\nError: Unable to 'pub upgrad
e' flutter tool. Retrying in five seconds... (2 tries left)\nResolving dependencies...\nGot socket error trying to find package completion at https://pub.dev.\nError: Unable 
to 'pub upgrade' flutter tool. Retrying in five seconds... (1 tries left)\nCommand 'pub upgrade' still failed after 10 tries, giving up.")                                    
ERROR: Logfile of failure stored in: /home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/temp/log.do_compile.1487050                              
```

The oddness with this is that it only happens if I have a `DL_DIR` set-up  for my `conf/site.conf` file and if that `DL_DIR` is populated. 

I assumed that I had a very old `DL_DIR` and was causing that, but after cleaning that up, using a fresh one, and trying for a second time it happen again. So the first time it succeeds (that is the `DL_DIR` is basically empty) but later on, when I try again by cleaning `tmp/`  and `sstate-cache`  directories from yocto it fails wit that error. 

The connect syscalls returns ENOTREACH or something like that, like I do not have DNS on the build machine. 

Attached also what follows after that: 

```
Log data follows:                                                                                                                                            18:36:25 [23/262]
| DEBUG: Executing python function do_compile                                                                                                                                 
| DEBUG: source_dir [/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/git]                                                                     
| DEBUG: flutter_application_path []                                                                                                                                          
| DEBUG: pubspec_yaml filepath [/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/git/pubspec.yaml]                                             
| NOTE: environ({'PKG_CONFIG_SYSTEM_INCLUDE_PATH': '/usr/include', 'infodir': '/usr/share/info', 'servicedir': '/srv', 'READELF': 'aarch64-agl-linux-readelf', 'systemd_user_u
nitdir': '/usr/lib/systemd/user', 'TZ': 'UTC', 'BUILD_NM': 'nm', 'BUILD_CFLAGS': '-isystem/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/reci
pe-sysroot-native/usr/include -O2 -pipe', 'sysconfdir': '/etc', 'PSEUDO_DISABLED': '1', 'BUILD_AR': 'ar', 'LDFLAGS': '-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fcanon-pre
fix-map  -fmacro-prefix-map=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/git=/usr/src/debug/flutter-ics-homescreen/git  -fdebug-prefix-map=
/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/git=/usr/src/debug/flutter-ics-homescreen/git  -fmacro-prefix-map=/home/mvlad/agl/rpi5/rpi5/tm
p/work/aarch64-agl-linux/flutter-ics-homescreen/git/git=/usr/src/debug/flutter-ics-homescreen/git  -fdebug-prefix-map=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flu
tter-ics-homescreen/git/git=/usr/src/debug/flutter-ics-homescreen/git  -fdebug-prefix-map=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/reci
pe-sysroot=  -fmacro-prefix-map=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot=  -fdebug-prefix-map=/home/mvlad/agl/rpi5/rpi5/
tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot-native=  -Wl,-z,relro,-z,now', 'PERL_HASH_SEED': '0', 'AS': 'aarch64-agl-linux-as ', 'libdir': '/usr/lib'
, 'RANLIB': 'aarch64-agl-linux-gcc-ranlib', 'STRIP': 'aarch64-agl-linux-strip', 'PKG_CONFIG_PATH': '/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescree
n/git/recipe-sysroot/usr/lib/pkgconfig:/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot/usr/share/pkgconfig:/home/mvlad/agl/rpi5
/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot/usr/lib/pkgconfig:/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/r
ecipe-sysroot/usr/share/pkgconfig', 'BUILD_CXX': 'g++ ', 'HOME': '/home/mvlad', 'STRINGS': 'aarch64-agl-linux-strings', 'localstatedir': '/var', 'nonarch_base_libdir': '/usr/
lib', 'datadir': '/usr/share', 'OBJCOPY': 'aarch64-agl-linux-objcopy', 'exec_prefix': '/usr', 'CCLD': 'aarch64-agl-linux-gcc  -mbranch-protection=standard -fstack-protector-s
trong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/re
cipe-sysroot', 'systemd_system_unitdir': '/usr/lib/systemd/system', 'SOURCE_DATE_EPOCH': '1718000000', 'BUILD_CC': 'gcc ', 'PKG_CONFIG_DIR': '/home/mvlad/agl/rpi5/rpi5/tmp/wo
rk/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot/usr/lib/pkgconfig', 'BUILD_CXXFLAGS': '-isystem/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics
-homescreen/git/recipe-sysroot-native/usr/include -O2 -pipe', 'mandir': '/usr/share/man', 'LC_ALL': 'en_US.UTF-8', 'PKG_CONFIG_SYSTEM_LIBRARY_PATH': '/usr/lib:/usr/lib', 'BUI
LD_RANLIB': 'ranlib -D', 'LOGNAME': 'mvlad', 'PKG_CONFIG_LIBDIR': '/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot/usr/lib/pkgconfig', 'CXX': 'aarch64-agl-linux-g++  -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot', 'CPPFLAGS': '', 'libexecdir': '/usr/libexec', 'BUILD_CCLD': 'gcc ', 'includedir': '/usr/include', 'OMP_NUM_THREADS': '32', 'base_sbindir': '/usr/sbin', 'nonarch_libdir': '/usr/lib', 'BUILD_LD': 'ld ', 'CPP': 'aarch64-agl-linux-gcc -E --sysroot=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot  -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security', 'base_libdir': '/usr/lib', 'MAKE': 'make', 'docdir': '/usr/share/doc', 'FC': 'aarch64-agl-linux-gfortran  -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/mvlad/agl/rpi5/rpi5/tmp/work/aarch64-agl-linux/flutter-ics-homescreen/git/recipe-sysroot', 'CC': 'aarch64-agl-linux-gcc  -mbranch-protection=st [... cut ...]
  /home/mvlad/agl/rpi5/meta-agl-demo/recipes-demo/flutter-ics-homescreen/flutter-ics-homescreen_git.bb:do_compile
Summary: There were 7 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

What I've been also trying to work around this was:

- set `do_compile[network] = 1`, globally not just for flutter-ics-homescreen; this worked
- removed `DL_DIR` entirely; this worked aslo
- start the build, wait until it dies with that error, do a `do_cleanall` for `flutter-ics-homescreen` then retry the original build target; works as well

kwavnet, 2024-08-14:
We faced the same issue - adding `do_compile[network] = 1` to the affected recipe helper already - not sure what is causing this

jwinarske, 2024-08-14:
@mv0 The pub cache fetcher is complete and merged into scarthgap.  Flutter-ics-homescreen needs an update for 3.24.0.  Some things have changed in how lock files are handled. 

* Lock file is created if not present in do_archive_pub_cachr
* The lock file is the origin of truth
* At the end of do_restore_pub_cachr it will run dart pub get --offline --enforce-lockfile

I have a new tool called pubspec.yaml that was the reference for this implementation.  Started with this script, then ported to common.inc.

So what you can do is run this tool outside of the Yocto environment to confirm the app is ready for Yocto.  For this you would want to use the same flutter sdk as used in the Yocto build.

If running the tool passes without error, that means the offline dependencies are met.  It does not mean the build will pass; two different things.

If all that works and you still hit the issue, then we can try removing the cli animation line.

@kwavnet tip of tree scarthgap should be used for this exercise.

jwinarske, 2024-09-10:
@mv0 AGL has been updated to 3.24.2.  Unique archive name PR was picked up yesterday.  Please confirm issue has been resolved with AGL master/Ricefish 18.0.1.

kwavnet, 2024-09-11:
@jwinarske any chance that those patches will be backported to kirkstone?

jwinarske, 2024-09-11:
@kwavnet yes, dunfell as well.
https://github.com/meta-flutter/meta-flutter/issues/538

jwinarske, 2024-12-21:
What’s the status of this bug with tip of tree on master or scarthgap?

jwinarske, 2025-01-27:
@mv0 can this be closed now?

mv0, 2025-01-27:
Apologies, haven't been able to try this out, though I admit I was lazy and I saw the comments fly-by. I'll try out this week and resolve accordingly. Thanks for the nudge.

mv0, 2025-02-02:
Confirmed this no longer happens.  Thank you!

closed 2025-02-02 by mv0